3dgeo-heidelberg/helios #11: Osclillating scanner does not use acceleration

lwiniwar, 2021-01-22:
Oscillating scanners should slow down (decrease) the angular step width at the extremes of the swath. Currently, the change in angle is calculated in OscillatingMirrorBeamDeflector::currentScanLinePulse (line 56):

```[c]
int bla = std::min(currentScanLinePulse, cached_pulsesPerScanline / 2) - std::max(0, currentScanLinePulse - cached_pulsesPerScanline / 2);

state_currentBeamAngle_rad = -this->cfg_setting_scanAngle_rad + cached_angleBetweenPulses_rad * bla;
```

albertoesmp, 2021-03-02:
This issue has been addressed. Please, don't close the issue until new implementation has been exhaustively tested because this is a significative update.

han16nah, 2021-03-29:
@albertoesmp I tested the implementation and while the pattern looks realistic now, the simulated swath width is too small.
You can find the test survey and scene here: https://heibox.uni-heidelberg.de/d/599ff94a0d5f4848a5f0/
(Altitude a.g.l: 1000 m, scan angle: 20°, expected swath width: ~728 m, actual swath width: ~425 m.)

This figure shows a simulated wath with the old implementation (correct swath width) in red and the new implementation in blue
[image]

lwiniwar, 2021-05-17:
Some python code how to solve for the turning points is included here:
[oscillating.zip](https://github.com/3dgeo-heidelberg/helios/files/6494213/oscillating.zip)

Results look like this:
[image]

[image]

albertoesmp, 2022-02-06:
This issue has been solved in pull request https://github.com/3dgeo-heidelberg/helios/pull/171

It has not been merged yet because there is a test case which is not passed. But I think it is because changing the OscillatingMirrorBeamDeflector changed the size of the output point cloud. Please, confirm it and merge the pull request if it is okay for you too.

The output point cloud with new implementation looks like this:
[image]

The plots monitoring simulation look like this:
[image]