3dgeo-heidelberg/helios #12: tiff-Loader from pyhelios not working

lwiniwar, 2021-01-25:
Loading a tiff-Scenepart from pyhelios throws an error: 
[image]

Loading the same survey XML with `helios.exe` works and produces this output:
[image]

albertoesmp, 2021-02-02:
The GDAL drivers were not loaded from pyhelios, thats why ERROR 4 arised. It has been already solved. Please, close the issue once you confirm everything works as expected. Otherwise, please provide me with some demo so I can reproduce further problems.

lwiniwar, 2021-02-03:
fixed in 422f89d1efecb420bf07f43888bab4dde55845fb

closed 2021-02-03 by lwiniwar