3dgeo-heidelberg/helios #13: Issues with the release executable

maptastic, 2021-01-27:
Hello,
I am having trouble running the release in windows. When running helios.exe --test from its location in the run file the programming is crashing. Are there any additional steps we need to take before running?
Thanks

maptastic, 2021-01-27:
Some additional information, the 'helios.exe --help' runs fine. And this is the progression of the 'helios.exe --test'
[image]

han16nah, 2021-01-28:
Hi Nina,
are you also getting this warning? 
```
WARNING! You might not be inside helios-plusplus root directory.
        This might cause tests to fail.
TEST Randomness generation test                           [PASSED]
TEST Noise sources test                                   [PASSED]
```
The tests fail, because they expect the `helios.exe` to be located in the helios-plusplus root directory. This is not the case in the precompiled version (where `helios.exe` is in the "run"-subfolder). Thanks for reporting, we will work on modifying the tests to work also for this case.

as the `--help` command is working for you, could you try executing a quick demo survey?
From the root directory, try:
`run\helios data\surveys\toyblocks\als_toyblocks.xml`

If you can run this survey, you're ready to get started with HELIOS++ (and we will work on the tests in the meanwhile).

Best,

Hannah

maptastic, 2021-02-01:
Thanks so much for the quick response! I have been able to get the demo running!

closed 2021-02-01 by maptastic

thomasTUK, 2021-08-03:
Unfortutely I am not able to run the tests:

`helios-plusplus-win\run>helios data\surveys\toyblocks\als_toyblocks.xml`

HELIOS++ VERSION 1.0.8

CWD: "D:\Users\user\Downloads\helios-plusplus-win\run"
seed: AUTO
surveyPath: "data\surveys\toyblocks\als_toyblocks.xml"
assetsPath: "assets/"
outputPath: "output/"
writeWaveform: 0
calcEchowidth: 0
fullWaveNoise: 0
njobs: 0
platformNoiseDisabled: 0
legNoiseDisabled: 0
rebuildScene: 0
lasOutput: 0
las10: 0
fixedIncidenceAngle: 0

xmlDocFilename: als_toyblocks.xml
xmlDocFilePath: data\surveys\toyblocks
ERROR: loading data\surveys\toyblocks\als_toyblocks.xml failed.
ERROR: xml root not found