3dgeo-heidelberg/helios #14: Revision of tests needed

han16nah, 2021-01-28:
1) Most tests can not be run if `helios.exe` is executed from outside the root directory (e.g. from the "run"-subfolder in the precompiled release):
```
WARNING! You might not be inside helios-plusplus root directory.
        This might cause tests to fail.
TEST Randomness generation test                           [PASSED]
TEST Noise sources test                                   [PASSED]
```

- [ ] Adapt the tests to also work if `helios.exe` is called from a folder within the root directory.

2) Since a recent bug fix of platform rotations, the "Platform physics test" is failing.

- [ ] Revise the platform physics test.

albertoesmp, 2021-02-02:
Both requests have been satisfied. Please close this issue once you have confirmed everything works as expected. Otherwise, just let me know what stills failing

thomasTUK, 2021-08-03:
Hello @han16nah, I just downloaded the current version (https://github.com/3dgeo-heidelberg/helios/releases/download/v1.0.8/helios-plusplus-win.zip) ans still do get the message:

D:\Users\user\Downloads\helios-plusplus-win\run>helios --test
WARNING! You might not be inside helios-plusplus root directory.
        This might cause tests to fail.
Please, change to helios-plusplus root directory.
Alternatively, specify path to directory containing required test files with:
         --testDir <path>

Current test files directory is: "data/test/"

TEST Randomness generation test                           [PASSED]
TEST Noise sources test                                   [PASSED]

han16nah, 2021-08-03:
Hi @thomasTUK 
can you try navigating to the HELIOS++ root directory (in your case one directory up to "D:\Users\user\Downloads\helios-plusplus-win"; just type `cd ..` and then run the tests from there with:
`run\helios --test`
Please let me know if you can now run the tests.

thomasTUK, 2021-08-03:
Ah... Understood. Worked fine for me! Still got the warning:

D:\Users\user\Downloads\helios-plusplus-win> run\helios --test
WARNING! You might not be inside helios-plusplus root directory.
        This might cause tests to fail.
Please, change to helios-plusplus root directory.
Alternatively, specify path to directory containing required test files with:
         --testDir <path>

Current test files directory is: "data/test/"

TEST Randomness generation test                           [PASSED]
TEST Noise sources test                                   [PASSED]
TEST Voxel parsing test                                   [PASSED]
TEST Ray intersection test                                [PASSED]
TEST Serialization test                                   [PASSED]
TEST Survey copy test                                     [PASSED]
TEST Plane fitter test                                    [PASSED]
TEST LadLut test                                          [PASSED]
TEST Platform physics test                                [PASSED]
TEST Scene part split test                                [PASSED]

han16nah, 2021-08-03:
Okay, interesting.
So good news is that your tests passed and you can use HELIOS++ (make sure to always call it from the root directory, not from the run directory).

@albertoesmp or @miguelyermo-heidelberg Can we fix the WARNING message? It seems to be printed in the latest version(s) even though the user is indeed inside the helios-plusplus root directory. What is tested for before printing the warning? For the specific name of the helios root directory (it may be called "helios", "helios-plusplus-win", "helios-plusplus-lin" etc.?). Or rather for other files or folders that are expected to be located in the root directory? I guess the latter would be more optimal (?)

albertoesmp, 2021-08-03:
> Okay, interesting.
> So good news is that your tests passed and you can use HELIOS++ (make sure to always call it from the root directory, not from the run directory).
>
> @albertoesmp or [...]

I've already implemented a fix for this. Required test files are explicitly checked beforehand and warning is shown if and only if they cannot be accessed. Notwithstanding, this update comes bundled with a bigger one so merging will take a while.

Thanks a lot for reporting!

closed 2025-02-06 by han16nah