3dgeo-heidelberg/helios #18: Error messages

han16nah, 2021-02-24:
Suggestions for improving error messages when executing surveys from CMD.

1. If an XML file referenced in the survey is not found, we get a **warning** (or **information**?), e.g.:
```console
ERROR: loading data/platformss.xml failed.
```
This warning is only printed to stdout, if the verbosity level is set to `-v` (information, errors and warning). I would consider this message an **error** instead of a warning and also print it with the default verbosity level (only information and errors).

2. If the ID of the platform/scanner/scene is wrong (but the XML file is right), we also get a **warning**, e.g.:
```console
ERROR: platform asset definition not found: data#sr222
```
The error message does not contain the complete path and the "data#<id>" could be confusing. It might be changed to
```console
ERROR: platform asset definition not found: data/platforms.xml#sr222
```
or
```console
 ERROR: platform asset definition not found: #sr222 in data/platforms.xml
```
so that the user knows in which XML file to look for the mistake.

Again, the message should count as **error** instead of a **warning** and be displayed with the default verbosity level.

3. When the wrong *platform*-ID is provided (example above), the scene is built nevertheless and only after that, the program aborts. This is especially a problem for very large scenes, which might take long to build. Furthermore, the warning about the platform is "lost" further up in the logging and the user might struggle to find the warning. Can we abort the program immediately if no platform asset is found?

albertoesmp, 2021-03-02:
This issue has been addressed. Please, verify new behavior is as expected and close the issue. Otherwise, please provide a demo reproducing misbehaviors so they can be fixed.

han16nah, 2021-03-03:
New error messages behave as expected, thank you!

closed 2021-03-03 by han16nah