3dgeo-heidelberg/helios #25: Re-Simulation Failing on XML file

saadatshakeel, 2021-04-22:
Hello!,

When I run the simulation first time on a xml file, it runs fine but when I make changes to the xml file and try to rerun the simulation on the same xml file, I get the following error: "Cannot load scene" and simulation fails. This problem continues even if I rename the file. Moreover, if I generate a new xml file from obj. file with a name similar to one of the previous xml files, I face the same issue. I would appreciate your guidance, as I have tried hard to fix this issue but no success.

Thank you

Best Regards

han16nah, 2021-04-23:
Hi @saadatshakeel 
my first thought was, that maybe adding the flag `--rebuildScene` might help when executing HELIOS++ in the command line, so the previously built scene is overwritten. But since it also does not work if you rename the file, it is probably a different issue.

Which changes did you make in the XML file? Can you attach your XML files (survey and scene) and sceneparts (OBJ files) with which we may reproduce your problem and find a solution?

Best regards

Hannah

han16nah, 2021-04-29:
Hi @saadatshakeel 
are there any news on this issue? Were you able to solve it?

Cheers

Hannah

saadatshakeel, 2021-05-03:
Hi Annah,

Yes, I was able to resolve this issue. Thank you very much 😊

Regards

Saadat

closed 2021-05-03 by han16nah