3dgeo-heidelberg/helios #26: efilepath option for point cloud loader fails when supplying single file

lwiniwar, 2021-04-29:
As the `efilepath` option is a logical extension to `filepath`, it should also work when supplied a single file path (similar to python's `glob.glob`). Currently, loading fails when a single file (without * or ?) is given as `efilepath`.

miguelyermo-heidelberg, 2021-08-10:
@lwiniwar 
Could you confirm that this bug is still there? Is working fine for me: loading an .obj by using efilepath works fine on my side

```<param type="string" key="efilepath" value="data/sceneparts/hashing/a.obj" />```

han16nah, 2021-08-10:
My latest info that it works indeed for the objloader but not for the point cloud loader (xyzloader)

miguelyermo-heidelberg, 2021-08-10:
Indeed, the logic of the two methods is different. Thanks for pointing it out.

miguelyermo-heidelberg, 2021-08-12:
@lwiniwar, could you provide me an example where this bug appears?
I tried to load the survey ```data/scenes/voxels/sphere_xyzloader.xml``` using _efilepath_ instead of _filepath_ when loading the scene with id ```sphere_xyzloader_demo```. The debugging process shows me that the paths are being correctly managed, even if a single path is provided.

closed 2022-02-01 by han16nah