3dgeo-heidelberg/helios #27: PlatformSettings template

han16nah, 2021-04-29:
We currently have the option to define template `<scannerSettings >` at the beginning of the survey XML file, which can be reused for the legs by specifying the template ID in the leg `<scannerSettings>` (see [Wiki - Scanner Settings](https://github.com/3dgeo-heidelberg/helios/wiki/Survey#scanner-settings)). This is very handy, because the settings do not have to be defined for each leg and it reduces accidental mistakes in the XML file. We should have the same for the platform settings.

Task:
Implement the counterpart for `<platformSettings>`. 
In the `platformSettings` template, the following parameters can then be defined: 
- `movePerSec_m`
- `trajectoryTimeInterval_s`
- `onGround`
- `stopAndTurn`
- `smoothTurn`
- `slowdownEnabled`
- `yawAtDeparture_deg`. 

Also make the platform settings template accessible from **pyhelios**.

albertoesmp, 2022-01-21:
Template support for PlatformSettings has been added at https://github.com/3dgeo-heidelberg/helios/commit/9fc0b61c06da06bd9c12377a52e50aad3713a1bb

They can be defined as scannerSettings templates:
```
<document>
<platformSettings id="templateId" movePerSec_m="50" ... />
...
```

They can be used as scannerSettings templates:
```
<survey>
  <leg>
    <platformSettings template="templateId"/>
    ...
  </leg>
  ...
```

They've been implemented with the same priority than scanner settings: Leg -> Template -> Default

Please, check everything works as expected before closing this issue.

han16nah, 2022-02-02:
Just a minor thing: The platformSettings object is missing the `id` attribute.

[image]

closed 2022-02-05 by han16nah