3dgeo-heidelberg/helios #37: Negative remaining simulation times and completed percentages over 100%.

miguelyermo-heidelberg, 2021-05-18:
When executing `./helios data/surveys/toyblocks/uls_toyblocks_survey_scene_combo.xml`, if the `pulseFreq_hz` of each one of the legs is set to 10 000 Hz instead of 100 000 Hz, the first four legs are executed until the 104%, whilst the last one seems to have no end, yielding negative remaining times when the 100% is reached

[image]

han16nah, 2021-05-18:
Could this be the same issue as #22 ?
It is quite a problem.. the copter is only working properly when setting `slowdownEnabled = "false"` (which is not the default).

miguelyermo-heidelberg, 2021-05-18:
Maybe, but it seems that the behaviour is not the same in both cases (maybe because of the different pulseFreq_hz used). I'll have a look tomorrow.

han16nah, 2021-05-27:
Any news on this @miguelyermo-heidelberg ?

miguelyermo-heidelberg, 2021-05-27:
Not yet, I am focusing on other high priority tasks. I'll let you know when I start looking into it.

miguelyermo-heidelberg, 2022-01-20:
Seems to be fixed in the current dev branch.

closed 2022-01-20 by miguelyermo-heidelberg