3dgeo-heidelberg/helios #41: Caching models with identical name during scene generation

aelmiger, 2021-05-26:
Currently the way to load a particle system like plants is to export the model file of the single plant (obj in my case) and then have each transformation of that model in the scene.xml file. During scene generation this model is then imported again for each transformation leading to large delays.

The generation of the scene file is currently by far the biggest bottleneck for usecases with large amounts of instances of the same object.

I assume that it would be a large speedup if models with the same filename in the scene.xml were read only once.

lwiniwar, 2021-05-26:
Hi @aelmiger and thanks for your issue!
You are correct in that each scenepart is loaded separately, even if the same object model is used multiple times:

https://github.com/3dgeo-heidelberg/helios/blob/9e85766266726817229de4d40b23f71c725090a7/src/assetloading/XmlAssetsLoader.cpp#L755-L760

While reading the mesh could in fact be done only once, since individual transformations are applied for each instance:

https://github.com/3dgeo-heidelberg/helios/blob/9e85766266726817229de4d40b23f71c725090a7/src/assetloading/XmlAssetsLoader.cpp#L815-L822

However, what I assume takes the longest time, is the insertion of the geometry in the scene, i.e. into the kD-Tree that is thereafter used in raycasting:

https://github.com/3dgeo-heidelberg/helios/blob/9e85766266726817229de4d40b23f71c725090a7/src/assetloading/XmlAssetsLoader.cpp#L894

and (in `scene->finalizeLoading();`)

https://github.com/3dgeo-heidelberg/helios/blob/9e85766266726817229de4d40b23f71c725090a7/src/scene/Scene.cpp#L129-L139

Can you confirm that this part is what is the bottleneck in the scene building (time between `Building KD-Tree...` and ` finished in XX seconds`)? If this is the case, a more in-depth analysis of the kD-Tree would have to be carried out to identify potential speed-ups.

lwiniwar, 2021-05-26:
Another comment: as you say you are wanting to represent plants in the sense of a particle system, you could also try using the detailed voxel model (see https://github.com/3dgeo-heidelberg/helios/wiki/Scene#detailedvoxels) which allows you to achieve realism with much fewer primitives.

aelmiger, 2021-05-26:
Thank you for your response.

I did some tests with 10000 particles of simple cubes. (One small model with 10000 scene.xml entries)

Start->Building KD-Tree: 8:30 mins
Building KD-Tree - > finished in: 4 seconds
Simulation: 5 seconds

Now I joined all particle cubes to one model and repeated the process.

Start->Building KD-Tree: 6 seconds
Building KD-Tree - > finished in: 4 seconds
Simulation: 5 seconds

It appears that the loading process is the bottleneck, but it is not clear how much time the necessary transformation takes.

The voxel pipeline is currently not an option because I am using Blender.

lwiniwar, 2021-05-26:
Thanks @aelmiger for the tests. Then let me propose the following:

1. for each filter that loads a file (obj/mesh), a hash is generated (from the normalized filename should suffice, no need to go over the actual data imho) and stored in a collection.
2. this hash is used in a hashmap that provides access to the loaded geometries
3. if a filter loads a file with a hash that already exists, the geometry is cloned instead of re-reading the file. this step also happens when the file is loaded for the first time.
4. other filters are applied subsequently, independently.
5. after reading the file, the hashmap is removed from memory

So something like this:

[[image]](https://mermaid-js.github.io/mermaid-live-editor/#/edit/eyJjb2RlIjoiZ3JhcGggVERcbiAgICBBW2ZpbHRlciBpcyBmaWxlbG9hZGVyXSAtLT4gQltjYWxjdWxhdGUgaGFzaF1cbiAgICBCIC0tPiBDe2hhc2ggZXhpc3RzP31cbiAgICBDIC0tPnx5ZXN8IERbY2xvbmUgZ2VvbWV0cnldXG4gICAgQyAtLT58bm98IEVbcmVhZCBpbiBmaWxlLCBzYXZlIGhhc2ggKyBnZW9tZXRyeV1cbiAgICBFIC0tPiBEXG4gICAgRCAtLT4gRlthc3NpZ24gdG8gc2NlbmVwYXJ0XVxuICAgIEYgLS0-IHxyZXBlYXQgZm9yIGFsbCBmaWx0ZXJzfEFcbiAgICBGIC0tPiB8ZW5kfEdbcmVtb3ZlIGhhc2hlcytnZW9tZXRyaWVzXVxuXG4gICAgSFtoYXNobWFwXSAtLT4gQ1xuICAgIEggLS0-IEUiLCJtZXJtYWlkIjp7InRoZW1lIjoiZGVmYXVsdCIsInN0eWxlIjoiSCBmaWxsOiMwMDc1OGYifSwidXBkYXRlRWRpdG9yIjpmYWxzZX0)

@miguelyermo-heidelberg , does that sound viable to you?

miguelyermo-heidelberg, 2021-05-26:
Well, you almost did all the job figuring out the logic. Using hashes is a nice and simple solution. In my experience, copying memory is also a very expensive operation. I guess that cloning 10 000 cubes using will be slower than load a scene with the 10 000 cubes as a whole. Even with this, it will still be much faster than the current method. I will work on this as soon as I can. It should be easy to implement.

miguelyermo-heidelberg, 2021-06-15:
@aelmiger, may I ask you for the neccesary files in order to reproduce your tests so I can start working on it? Thanks in advance.

aelmiger, 2021-06-16:
I have attached a simple scene file with alot of cube instances. Let me know if this is sufficient for you.
Thank you for tackling this issue.
[data.zip](https://github.com/3dgeo-heidelberg/helios/files/6663973/data.zip)

miguelyermo-heidelberg, 2021-07-09:
I've just created a pull request with the implementation of the cache. I did some tests with the scene @aelmiger provided me, containing 8 000 cubes and... my load times without using the cache are of 3 seconds approximately. #aelmiger, can you confirm that the scene you provided me is loading very slow on your side? Thanks in advance.

aelmiger, 2021-07-09:
Thank you for implementing a solution. I made a mistake and didnt include the survey file in my data.
The command to reproduce loading the cubes is:
`run\helios.exe data\surveys\test_survey.xml`
[data_with_survey.zip](https://github.com/3dgeo-heidelberg/helios/files/6790885/data_with_survey.zip)

On my PC it takes multiple minutes to load this.
Do you have compiled binaries for the new implementation for me to test? (Windows)

lwiniwar, 2021-07-15:
Hi @aelmiger, thank you for providing the updated survey. I achieve the following results on a fairly performant system. I also tried running `helios` with the `--quiet` flag, which drastically improves runtime.

Version | Runtime (wall time)
--------|----------
v1.0.6 (70b018e) | 1 min 17 sec 
v1.0.6 (70b018e) (`--quiet`)| 4.18 sec 
#53  | 48.6 sec 
#53  (`--quiet`) | *2.52 sec *

I assume the speedup will be larger for more complex objects. @miguelyermo-heidelberg , is the text written to stdout buffered? It seems that there is quite a lot of text written, and that this is a limitation:

```
Loading .obj from cache
Adding primitive to Scenepart ...
# new primitives added: 12
# total primitives loaded: 12
Applying filter: rotate
Applying filter: translate
Applying filter: scale
[...]
```

Point cloud results are the same, when I set `-j 1 --seed 42`. Here's a compiled .exe/.pyd for you until we have merged this into main and published a new release:
[release_#53.zip](https://github.com/3dgeo-heidelberg/helios/files/6824452/release_.53.zip)

lwiniwar, 2021-07-15:
Also note that if you want the fast runtime but still be able to read the `INFO` output, you can run
`helios.exe path/to/survey.xml --v2 > log.txt`
which will run at speeds similar to running with `--quiet`, but will write the output to a file called `log.txt`

lwiniwar, 2021-07-30:
fixed in #53

closed 2021-07-30 by lwiniwar