3dgeo-heidelberg/helios #42: Building pyhelios from scratch in Linux.

miguelyermo-heidelberg, 2021-05-28:
I am having problems building pyhelios from scratch.
First, there is a missing dependency in the BUILDME.md. 
In "Solving Dependencies", ```libboost-python-dev``` should be added.

Once this is solved, the installed Boost library through APT does not link well at the end of compilation when ```-DPYTHON_BINDING=1``` is used to generate the Makefile. This happens both in my machine and in Docker containers.

The CMakeLists.txt has options to use a manually-compiled Boost version, that should be located in /lib directory. The problem here is that in any case, the boost installed in the system is used instead. This happens in both Ubuntu 20.04 and Ubuntu 18.04. 

In Ubuntu 16.04, since the boost version installed through apt is the 1.6x, the manually-compiled boost is used instead. The problem here is that the installed version of GDAL is not modern enough to be used in the compilation of Helios++, so everything (gdal and its dependencies) should be compiled from scratch.

I am focusing on trying to get Helios++ with python bindings enabled working in any Ubuntu version, so I can work with it, and then generalise the building process for any version. I think this is needed if we want as many people as possible to use Helios++.

As a side note, a minimum version of gcc is needed to build Helios++: gcc-7. A version older than that (like the default gcc version of Ubuntu 16.04, for instance) is not even able to compile LASlib, due to the minimum C++ standards required. This will be documented as well when everything is working properly.

miguelyermo-heidelberg, 2021-06-02:
Almost solved.

Finally, I was able to compile Helios++ enabling Python Bindings, which leads to the compilation of pyhelios.so, the shared library needed to use pyhelios module in Python. I will now detail the issues encountered and how to solve them, but **right now, following the steps given in BUILDME.md, is totally impossible to compile Helios++ enabling the possibility of use PyBindings at the same time.**

The root of all the problems that I encountered are basically two: the impossibility of overriding the Boost library location and a bug in two modules of the same Boost library: Boost.Thread and Boost.Python

On the one hand, the impossibility of overriding a version of Boost located in /lib/boost directory forced me to install Boost by using APT package manager in Ubuntu. No matter what version of the Boost library is installed from this source: the bug I encountered is reproduced at all times: the shared libraries of Boost.Thread and Boost.Python are not built because they were are not being compiled with Positional Independent Code. This means that the correct execution of those libraries depends on the absolute memory address given to them, which may vary from machine to machine. On the other hand, having two different versions of Boost installed in the system is not an option, because conflicts among versions will occur for sure. Since the /lib/boost library will be not used if there is an already installed version, the only solution I encountered here is to not have installed any Boost library in the system before trying to compile PyHelios. I tried to build from scratch a bunch o Boost versions, but only the 1.71 worked.

Installing libgdal-dev through APT is neither an option, since it directly depends of Boost library. If you first compile and install Boost from scratch, libgdal-dev will install over that a new version of Boost, which has the above mentioned bug.
https://packages.debian.org/stretch/libgdal-dev In this link, Boost is not shown in the dependencies of libgdal-dev, but it is installed along many others anyways. So, GDAL must be built from scratch as well. At this point I was paranoid enough to build everything from scratch, but there are many dependencies that can be built through APT package manager.

So, the process to install dependencies in order to compile Helios++ enabling Python Bindings in an Ubuntu 20.04 machine is described at the end of this message.

I am building a docker image in order to distribute an operating system with an already compiled Helios++ version with PyBindings enabled.

I think that this install process should be documented somewhere (maybe is too long for including it in the README.md file). What do you think? @albertoesmp @han16nah @lwiniwar 

# Helios Dependencies
## General
 - ```apt install cmake make gcc g++ libarmadillo-dev libglm-dev unzip```
## LASTools
- ```cd helios/lib```
- ```wget https://lastools.github.io/download/LAStools.zip && unzip LAStools.zip```
- ```cd LAStools && cmake .  && make```
## Boost
### Boost dependencies
 - apt install libpython3.8-dev python3
### Boost installation
 - ```wget https://boostorg.jfrog.io/artifactory/main/release/1.71.0/source/boost_1_71_0.tar.gz```
 - ```tar -xzvf boost_1_71_0.tar.gz```
 - ```cd boost_1_71_0```
 - ```./bootstrap.sh --with-python=python3.8```
 - ```./b2 cxxflags=-fPIC```
 - ```./b2 install```
## Proj
### Proj dependencies
- ```apt install pkg-config libsqlite3-dev sqlite3 libtiff5-dev libcurl4-openssl-dev```
### Proj installation
- ```wget http://download.osgeo.org/proj/proj-7.2.1.tar.gz```
- ```tar -xzvf proj-7.2.1.tar.gz```
- ```cd proj-7.2.1 && ./configure && make && make install```
## GDAL
### GDAL dependencies
- ```wget https://github.com/OSGeo/gdal/releases/download/v3.2.1/gdal-3.2.1.tar.gz --no-check-certificate ```
- ```tar -xzvf gdal-3.2.1.tar.gz```
- ```cd gdal-3.2.1 && ./configure && make && make install```

In Helios directory, simply do
```cmake -DPYTHON_BINDING=1 -DPYTHON_VERSION=38 .```
```...```
Profit.

[image]

lwiniwar, 2021-06-07:
Hi @miguelyermo-heidelberg , these are great news! I agree that this is maybe a bit too extensive for `README.md`, but would suggest to see it in `BUILDME.md`. A docker container with all dependencies and helios++ already built with python bindings sounds like a great Idea.

Could you also add some considerations on how one could compile pyhelios for other python versions? Is it as easy as replacing `apt install libpython3.8-dev` with `apt install libpython3.9-dev` (e.g.), `./bootstrap.sh --with-python=python3.8` with `./bootstrap.sh --with-python=python3.9` and ` cmake -DPYTHON_BINDING=1 -DPYTHON_VERSION=38 .` with ` cmake -DPYTHON_BINDING=1 -DPYTHON_VERSION=39 .`?

Thank you!

miguelyermo-heidelberg, 2021-06-07:
Indeed, It should be as easy as that. In the end, the best would be install libpython3-dev, that should install the latest version of the library, and then get the version automatically in order to build PyHelios. What worries me the most right know is that the installation was done choosing the exact versions of the libraries you are using. More tests will be done with the latest versions of those libraries (proj, gdal and boost). I'll keep you informed.

miguelyermo-heidelberg, 2021-06-15:
I've added all the necessary steps to compile Helios and PyHelios in the BUILDME.md file, as long as a Dockerfile to build an image with Helios already compiled.

closed 2021-06-15 by miguelyermo-heidelberg