3dgeo-heidelberg/helios #46: Version 1.0.8 generates different output

aelmiger, 2021-06-14:
The newest update 1.0.8 outputs completely different point clouds.
Perhaps I am missing some changes to the API, but there are suddenly less scan lines.

This also happens with the basic test survey.

`
helios.exe data/surveys/demo/tls_arbaro_demo.xml
`

lwiniwar, 2021-06-14:
Thanks for your observation.
As you might have seen from the [release notes](https://github.com/3dgeo-heidelberg/helios/releases/tag/v1.0.8), an issue with rotating polygon mirrors, such as the one used in RIEGL TLS Sensors was fixed with v1.0.8. For more details, I would like to refer you to #39, addressing this issue.

Just for completeness, I would like to point out that there should not be *less scan lines*, but *fewer points per scan line*, i.e. a larger angular increment between consecutive pulses. This is clearly visible in the point cloud (here, ground in the `tls_arbaro_demo`), where blue is v<=1.0.6 and green is v >=1.0.8:
[image]

The old version generated ~687k, the new one ~338k first return points for the first leg in `tls_arbaro_demo`. With a horizontal angle of 125 deg as defined in the survey file, and a rotation speed of 10 deg/s, this means that the scanner was running for 12.5 seconds. With a effective measurement rate of 42 kHz (100 kHz PRR, see http://www.riegl.com/uploads/tx_pxpriegldownloads/10_DataSheet_VZ-400_2017-06-14.pdf), this would lead to 500k effective pulses, of which some never hit the ground (go into the sky). This already indicates that 700k pulses would be too many.

If you have reason to believe that the new implementation is incorrect, please let us know.

closed 2021-06-14 by lwiniwar

aelmiger, 2021-06-16:
Thank you for pointing me in the right direction.

Currently I am calculating my sensor paramters based on the equations listed in the wiki:
[Wiki](https://github.com/3dgeo-heidelberg/helios/wiki/Scanners#vertical-resolution)

Do the equations for calculating the vertical and horizontal resolution still hold?

lwiniwar, 2021-06-16:
I think you have to muliply the `pulseFreq_hz` for the vertical spacing by ( `scanAngleMax_deg` / `scanAngleEffectiveMax_deg`).

I will check and update this accordingly in the wiki. The formula for the horizontal spacing is unaffected.