3dgeo-heidelberg/helios #58: Something wrong in Rotation class

jxo, 2021-07-30:
Axis-Angle to Rotation error:
 double halfAngle = -0.5 * angle; // why negative here?

Rotation::Rotation(RotationOrder order,	double alpha1, double alpha2, double alpha3) {
    Rotation r1 = Rotation(order.getA1(), alpha1);
    Rotation r2 = Rotation(order.getA2(), alpha2);
    Rotation r3 = Rotation(order.getA3(), alpha3);
    Rotation composed = r1.applyTo(r2.applyTo(r3)); // wrong order here?
    q0 = composed.q0;
    q1 = composed.q1;
    q2 = composed.q2;
    q3 = composed.q3;
}

lwiniwar, 2021-07-30:
Hi @jxo !
In the docstring to the method you are referring to, the reasoning is explained:

https://github.com/3dgeo-heidelberg/helios/blob/e1a91893ad8e019fa9694e338cc5551f3e990856/src/maths/Rotation.cpp#L46-L81

Notably, 
(a) the sine/cosine of the negative half-angle have to be taken, so we multiply it by -1 beforehand,
(b) the order of rotations corresponds to a rotation of vectors in a *fixed frame*, i.e. the axes are not rotated, but the object is. This is different from the standard view in navigation and computer graphics communities, where the object is kept fixed but the axes are rotated.

Please also see the doxygen documentation: 
https://htmlpreview.github.io/?https://github.com/3dgeo-heidelberg/helios/blob/main/doc/html/d5/dcb/classRotation.html#a6f0092b2c6341076e44764c0f82b7c17

Greetings,
Lukas

jxo, 2021-07-30:
why to choose this form？This is the conjugate of the origin quaternion in math. glm::quat may be better to understand

albertoesmp, 2021-08-02:
Hi @jxo

I agree with you this form can be a little counter-intuitive. It was implemented this way in the old java version and we decided to port the code as it was. I don't know why it was originally coded like that. Unfortunately, we are focused on implementing some new features which will require modifying other core components in the near future. In consequence, there are no plans on changing this soon. Notwithstanding, if it is too confusing for you, please let us know what and how you want to do so we can figure out a way to make your life easier. Maybe we could provide some interface based on the use cases you propose so rotation implementation becomes more transparent to the user

Sorry for the inconveniences and thank you for your feedback, we really appreciate it.

Kind regards,
Alberto

jxo, 2021-08-06:
@albertoesmp 

Thank you for your reply!
I get it.

May be I can help to migrate  old code from  java version.

albertoesmp, 2021-08-06:
Having help from an expert user like you, which understands the software at code level, would be really appreciated. I was wondering if you could try to pseudo-code some adapter-like interfaces that you would consider comfortable and easy to understand while useful. We haven't build any advanced user experience around helios yet and we are almost having no code-level feedback. Thank you a lot for your interest and comments.

jxo, 2021-08-09:
@albertoesmp 
more clearer interface may be the operator*() definition for Rotation and glm:dvec3，left multiply operator may be clearer than applyTo interface.
just like,
R=R1*R2*R3
V=R*V1

albertoesmp, 2021-08-09:
> @albertoesmp
> more clearer interface may be the operator*() definition for Rotation and glm:dvec3，left multiply operator may be clearer than applyTo interface.
> just like,
> R=R1*R2*R3
> V=R*V1

Great! That is something we can easily implement. Thank you a lot for your advice which will be reflected in future updates.

closed 2021-10-04 by lwiniwar