3dgeo-heidelberg/helios #70: Loading multiple files to the scene with random settings

yurithefury, 2021-08-17:
Hi,

I noticed that I can load multiple .obj files with the exact same settings using the "efilepath" key:
`<param type="string" key="efilepath" value="data/sceneparts/toyblocks/.*.obj" />`

Is there also a straightforward way to assign random "scale" and "offset" parameters (within certain limits of course) to each of the loaded .obj files?

Thank you in advance,
Yuri

yurithefury, 2021-08-18:
Resolved using `lxml`

closed 2021-08-18 by yurithefury

lwiniwar, 2021-08-18:
Hi @yurithefury,
perfectly done - there is no *built-in* way to do that, but I would also have used any xml library and a python script or similar.
I also guess that the number of options to maintain some sort of general-purpose usability that we would need to (a) provide and (b) document and explain would be quite immense.