3dgeo-heidelberg/helios #79: Boost should not force static linking

copyme, 2021-09-20:
There should be an option as it is in CGAL for controlling whether boost should use static or dynamic libraries. Only some Linux distributions offer static libraries for boost and most do not do that.

albertoesmp, 2021-09-27:
First, thank you for opening this issue. I think you are pointing to something that we should really improve for Linux users.

I've tried to solves this by updating our CMake files, so now it is possible to use `-DBOOST_DYNAMIC_LIBS=1` to use boost dynamic libraries when compiling Helios++.  Unfortunately, I couldn't test this on my system yet so I'm not sure if the fix does the trick or if the problem persists. I would really appreciate if you could try this new flag and tell us if it is working or not. In case it is not working, please provide us with both the CMake console output and the log file at _/.../helios/cmake-build-release/CMakeFiles/CMakeOutput.log_ so we can figure out what is going on.

Thank you again for your collaboration!

miguelyermo-heidelberg, 2021-10-20:
I've tried the proposed solution and I am able to compile helios against both static and dynamic libraries. I would like to do more tests in the distribution you used, @copyme, if you provide me that information. Thank you!

closed 2021-11-22 by lwiniwar