3dgeo-heidelberg/helios #90: New flag -gpsStartTime

han16nah, 2021-10-08:
**Problem**: 
We may have multiple sensors mounted on the same platform, e.g., in the case of a dual waveform system: http://www.riegl.com/uploads/tx_pxpriegldownloads/RIEGL_VQ-1560i-DW_Datasheet_2019-09-02.pdf
A (valid) way to simulate this in HELIOS++ is to define the different scanners in the scanner XML file and run multiple separate simulations with the same trajectory and platform, but exchanging the scanner. However, the simulated point measurements and trajectories will have different GpsTime values.

**Idea**:
Implement a flag `-gpsStartTime`, that allows to define the same start time for multiple simulation runs. This way, GpsTime will be matching in simulated point clouds, as desired for a multi-sensor/multi-channel system.

albertoesmp, 2022-01-21:
The --gpsStartTime flag has been added (https://github.com/3dgeo-heidelberg/helios/commit/e1ee08107fb623ce7470f58d7f1014c1102acc2e).
It is documented both, at helios --help and at README.md.
It can also be specified for pyhelios through SimulationBuilder (`setFixedGpsTime` function).

By default it is "", which means using current time (as was done before).
It can be specified as "x", where x is an integer representing a POSIX timestamp.
It can be specified as "YYYY-MM-DD hh:mm:ss", matching the exact format. So, for example, day 7 must be given as 07.
Notice, however, that to prevent problems with GPS week the usual transformation is applied even to user given fixed GPS times. It is, subtracting the duration between (1980-01-06 - 1970-01-01) and taking milliseconds since start of GPS week.

Please, confirm this implementation works as expected before closing the issue.

lwiniwar, 2022-02-02:
@albertoesmp thanks for implementing this. It seems like something is wrong with the parsing of the date when given as `YYY-MM-DD hh:mm:ss`:

[image]

Here, I calculated the unix timestamp with python:
```
    now = datetime.datetime.now()
    unixtime = datetime.datetime.timestamp(now)
```
which is 6 seconds off from the direct start (that accounts ~ for scene loading and KDT building), but the string (`stringtime = now.strftime('%Y-%m-%d %H:%M:%S')`) produces negative GPSTimes, which should not happen.

Furthermore, using e.g. the timestamp `758853922` in the python bindings and via the command line leads to different results.

When fixed, the tests in the `dev-gpsStartTimeTests`-branch should pass, see:
https://github.com/3dgeo-heidelberg/helios/blob/dev-gpsStartTimeTests/pytests/test_gpsStartTimeFlag.py

han16nah, 2022-02-02:
to add on this: I tried building two different simulations (using different scanners/survey XMLs) with the `pyhelios` `SimulationBuilder` with the same `gpsStartTime` given as `"YYYY-MM-DD hh:mm:ss"` and then executed them in a loop. The output point clouds did not have matching GPS times.. In the command line, however, specifying the `--gpsStartTime` in the format `"YYYY-MM-DD hh:mm:ss"` for the two surveys produced point clouds with matching GPS Times.

han16nah, 2022-02-05:
After a5c8f73f49966aaac28957f666cbf7e7afb28cc5, this looks good to me now.

closed 2023-01-04 by han16nah