3dgeo-heidelberg/helios #100: module 'pyhelios' has no attribute 'loggingDefault'

evagap, 2021-11-06:
[image]

Hello!

After running the instructions for pyhelios, I get the following error:

`module 'pyhelios' has no attribute 'loggingDefault'`

Could you please let me know how to resolve it?
Thank you!

han16nah, 2021-11-07:
Hi @evagap 
can you send the code that raised this error? From the warning, I assume you forgot the parenthesis. Make sure you use `pyhelios.loggingDefault()` and not `pyhelios.loggingDefault`
If this does not solve your issue, please provide the python code and the HELIOS++ version you are using so we can have a look at our source code. 
Best,
Hannah

evagap, 2021-11-07:
Hi Hannah, the code that raised this error is `helios.py`

I haven't made any changes to the code. I use the latest version 1.0.9
[Dockerfile log.pdf](https://github.com/3dgeo-heidelberg/helios/files/7493035/Dockerfile.log.pdf)

I also tried installing HELIOS++ with the python bindings using the Dockerfile and I get the attached errors.

Thank you for your help!
Best,
Eva

closed 2021-11-22 by lwiniwar