3dgeo-heidelberg/helios #110: Simulate 3D tree models in batch

Elephant-C, 2021-11-21:
Hi there,

I plan to simulate a hectare forest plot, which contains around 500 3D models.
Can I import all these 3D models one time? 

looking forward to your reply
Many thanks
Yujie

han16nah, 2021-11-22:
Hi @Elephant-C 
yes, you can import all these 3D models. However, the success and runtime of the simulation will depend on the **complexity of your models** (i.e., number of facets) and your **available RAM**. Depending on the type of simulation (platform, resolution, etc.) and your requirements, you might try to work with **low-poly** 3D tree models.

If the different 3D models do not require different coordinate transformations, you can easily configure your scene XML file to load all of the models using the `efilepath` parameter (see the [Loaders](https://github.com/3dgeo-heidelberg/helios/wiki/Scene#loaders) section in our Wiki). Just place them all in one folder (e.g. called `trees`) in `data/sceneparts` and then load them with the `efilepath` option using regular expressions. 

An example:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<document>
    <scene id="my_forest_plot" name="my_forest_plot">
        <part>
            <filter type="objloader">
                <param type="string" key="efilepath" value="data/sceneparts/trees/.*.obj"/>
            </filter>
        </part>
    </scene>
</document>
```
(You probably want to add a DTM or a [groundplane](https://github.com/3dgeo-heidelberg/helios/tree/dev/data/sceneparts/basic/groundplane) to the scene as well).

I hope this helps. If more questions come up, just contact us. Good luck!

han16nah, 2021-11-22:
We also have a section in our Wiki with hints on [simplifying geometries](https://github.com/3dgeo-heidelberg/helios/wiki/Dealing-with-large-scenes#simplify-your-geometries).