3dgeo-heidelberg/helios #111: ModuleNotFoundError: No module named 'pyhelios'

damontallen, 2021-11-21:
After following the instructions [here](https://github.com/3dgeo-heidelberg/helios#-running-pyhelios) the command,
    
    python pyhelios_demo/helios.py data/surveys/toyblocks/als_toyblocks.xml

simply yielded 

    Traceback (most recent call last):
      File "pyhelios_demo/helios.py", line 15, in <module>
        import pyhelios
    ModuleNotFoundError: No module named 'pyhelios'

And I was unable to locate `pyhelios` within the pyhelios_env directory either.

lwiniwar, 2021-11-22:
Hi Damon,

have you downloaded and unzipped precompiled HELIOS++ binaries from [here](https://github.com/3dgeo-heidelberg/helios/releases/tag/v1.0.9)? Then you can point the path in `helios.py` to the location of these binaries 
(https://github.com/3dgeo-heidelberg/helios/blob/c534ece808a760abb98c3a68a29249acdc30f19a/pyhelios_demo/helios.py#L9). If your current working directory is the HELIOS++ root directory at the time of executing the script, it should find the precompiled `pyhelios` module in the `run` directory - look for `pyhelios.pyd` if you are on Windows, and `pyhelios.so` if you are on Linux.

Best,
Lukas

damontallen, 2021-11-23:
I have now downloaded the precompiled binary, and now from within the HELIOS++ root directory I get a new error.

    $ python pyhelios_demo/helios.py data/surveys/toyblocks/als_toyblocks.xml --lasOutput
    Traceback (most recent call last):
      File "pyhelios_demo/helios.py", line 15, in <module>
        import pyhelios
    ImportError: libhelios.so: cannot open shared object file: No such file or directory

lwiniwar, 2021-11-24:
Hi Damon,
which python version are you working with? You might want to try the pre-release 1.1.0-Versions available here (scroll down to Artifacts and pick the right one for your python version):
https://github.com/3dgeo-heidelberg/helios/actions/runs/1494738430
Then, extract these files into the `run` directory and try running the script again.

lwiniwar, 2021-11-24:
It might also be necessary to add the `run` directory to the `LD_LIBRARY_PATH` environment variable:
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/lwiniwar/helios/run/` (adapt to your path)

damontallen, 2021-11-24:
~/helios$ python --version
    Python 3.8.0
    ~/helios$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/damon/helios/run/
    ~/helios$ python pyhelios_demo/helios.py data/surveys/toyblocks/als_toyblocks.xml --lasOutput
    Traceback (most recent call last):
      File "pyhelios_demo/helios.py", line 15, in <module>
        import pyhelios
    ImportError: libpython3.8.so.1.0: cannot open shared object file: No such file or directory
    ~/helios$ cd run
    ~/helios/run$ ls
    helios  libarmadillo.so.10  libgdal.so.29  libhelios.so  libproj.so.22  LICENSES.md  pyhelios.so

lwiniwar, 2021-11-25:
Hi Damon, 
we'll have a look and try to get that working. In the meantime, you could try to use the Docker image as expained in Issue #100 (see last comment there). I'll let you know as soon as we have any updates!

damontallen, 2021-11-25:
root@a45011dd446d:/home/helios# python3 pyhelios_demo/helios.py data/surveys/toyblocks/als_toyblocks.xml --lasOutput
INFO - 2021-11-25 23:15:46,699 - font_manager - generated new fontManager
xmlDocFilename: als_toyblocks.xml
xmlDocFilePath: data/surveys/toyblocks
Segmentation fault (core dumped)

lwiniwar, 2021-11-29:
Hi Damon, 
sorry for the delay. We had some troubles getting the Python bindings working again. We have also introduced python testing to our pipleline now, so this shouldn't happen again.

We are planning a new release in the next couple of days, so either feel free to wait for that or download the current dev artifacts from the latest build, matching your python version, or run the Docker container again (clearing the cache):
https://github.com/3dgeo-heidelberg/helios/actions/runs/1517550322
I (running Ubuntu 20.04 on WSL) additionally had to add symbolic links for 
- `libhdf5_serial.so.100`
- `libnetcdf.so.13`
and had to install BLAS, so:
1) `sudo apt install libopenblas0 gdal-bin`
2) `sudo apt install mlocate` (optional if you already have `locate` installed)
3) `locate libhdf5_serial`, here it found `libhdf5_serial.so.103`
4) `cd helios/run`
5) `ln -s /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.103 libhdf5_serial.so.100` (the path from before, and the version that helios requested)
6) repeat for `libnetcdf` (for me it was `ln -s /usr/lib/x86_64-linux-gnu/libnetcdf.so.15 libnetcdf.so.13`)
7) run pyhelios

We have noticed `pyhelios` to crash when the XML was malformed, sometimes it's a good idea to test the validity with the executable first (or with the XML checker, see #102)

Hoping this works for you, fingers crossed!
Lukas

damontallen, 2021-12-01:
I rebuild the Docker container but, though it ran for quite a bit longer, it still resulted in an error.

    Pulse frequency set to 1200000
    Scan angle set to 20
    outputPath=output/output/Survey Playback/toyblocks_als/2021-12-01_03-50-42/points/leg005_points.las
    
    Waypoint reached!
    stepCount = 3000010
    Main thread simulation loop finished in 78 sec.
    Waiting for completion of pulse computation tasks...
    Pulse computation tasks finished in 78 sec.
    [Open3D WARNING] GLFW Error: The GLFW library is not initialized
    [Open3D WARNING] GLFW Error: The GLFW library is not initialized
    [Open3D WARNING] GLFW Error: The GLFW library is not initialized
    Segmentation fault (core dumped)
    root@cf79be82aaa9:/home/helios#

lwiniwar, 2021-12-01:
Hi Damon,
good to hear that we are progressing. From your output I can see that the simulation ran through and your result point clouds should have been written - can you confirm? The rest might have to do with the visualisation through docker, and we will investigate this.

damontallen, 2021-12-03:
This is what I found.

    root@f5af1a435688:/home/helios/output/Survey Playback/toyblocks_als/2021-12-03_00-44-27/points# ls -ls
    total 91888
    26356 -rw-r--r-- 1 root root 26984997 Dec  3 00:44 leg000_points.las
        4 -rw-r--r-- 1 root root      611 Dec  3 00:44 leg000_trajectory.txt
        4 -rw-r--r-- 1 root root     1197 Dec  3 00:44 leg001_points.las
        0 -rw-r--r-- 1 root root        0 Dec  3 00:44 leg001_trajectory.txt
    39296 -rw-r--r-- 1 root root 40236975 Dec  3 00:45 leg002_points.las
        4 -rw-r--r-- 1 root root      591 Dec  3 00:45 leg002_trajectory.txt
        4 -rw-r--r-- 1 root root     1197 Dec  3 00:45 leg003_points.las
        0 -rw-r--r-- 1 root root        0 Dec  3 00:45 leg003_trajectory.txt
    26216 -rw-r--r-- 1 root root 26843031 Dec  3 00:45 leg004_points.las
        0 -rw-r--r-- 1 root root        0 Dec  3 00:45 leg004_trajectory.txt
        4 -rw-r--r-- 1 root root     1197 Dec  3 00:45 leg005_points.las
    root@f5af1a435688:/home/helios/output/Survey Playback/toyblocks_als/2021-12-03_00-44-27/points#

lwiniwar, 2021-12-03:
Hi Damon, this looks like expected, the simulation results are all there. My guess is that the remaining `Segmentation fault` really only comes from the visualisation. Is this something you can work with as-is for now or do you require the 3D viewer?

damontallen, 2021-12-03:
I managed to use a git repo to get the point data out, so if you could point me to the documentation for rendering the .las files in Python I'll be good.

lwiniwar, 2021-12-09:
Hi Damon,
a couple of point cloud visualisation libaries for python exist, see e.g.
https://stackoverflow.com/questions/50965673/python-display-3d-point-cloud
To read the .las-Files, I suggest `laspy`:
https://laspy.readthedocs.io/en/latest/

closed 2021-12-09 by lwiniwar