3dgeo-heidelberg/helios #142: Cherry-picking settings between scanner and leg XML

albertoesmp, 2022-01-04:
Scanner settings read at _XmlAssetsLoader::createScannerSettingsFromXml_ are overwritten by leg settings read at _XmlSurveyLoader::createLegFromXml_. The cherry-picking between read scanner settings and each leg settings could be done inside the `while(legNodes != nullptr){ ... }` loop at _XmlSurveyLoader::createSurveyFromXml_.

Cherry-picking strategy must take by default all values from scanner and overwrite only those which were specifically given for the leg. This cherry-picking strategy must be applied for each leg, so a different overwriting for each leg might be possible.

albertoesmp, 2022-01-20:
This issue has been implemented in commit https://github.com/3dgeo-heidelberg/helios/commit/51e32d83916c2a69bdcb6e6243a5e6ba2eea684d merged with https://github.com/3dgeo-heidelberg/helios/commit/d93f8c6e935f24cad90d53d0b304326c9a603e80

If no bugs are found, the issue can be closed.

han16nah, 2022-01-21:
closes #28

closed 2022-01-21 by han16nah