3dgeo-heidelberg/helios #144: Maximum range parameter

vivienzahs, 2022-01-11:
For some applications, users might want to avoid simulating measurement points with very large ranges to the scanner (e.g. to avoid measurements with high positional uncertainties and footprints).
If we want to exclude measurements exceeding a certain range, we would currently have to filter them out after the simulation.

I think it would be a great feature to have the option of setting a maximum range parameter (`rangeMax_m`) in the scanner settings of the survey.xml so it can already be considered for the simulation.

If no rangeMax_m is set by the user, it could be set to a default value corresponding to what is defined as maximum range in the datasheet for the respective scanner.

han16nah, 2022-01-21:
Working as expected. It's already documented in the Wiki and I will add it to one of the demo surveys (together with the new `stripId` #143).

han16nah, 2022-02-02:
The `rangeMax` attribute still has to be added to pyhelios 

[image]

han16nah, 2022-02-05:
`rangeMax` is now accessible from pyhelios as well.

closed 2022-02-05 by han16nah