3dgeo-heidelberg/helios #170: Data-Analytics compilation mode

albertoesmp, 2022-02-04:
**Problem**
Certain debug operations, implementation of new features, understanding and maintenance of current functionalities, insight analysis and performance profiling with respect to different variables could benefit a lot from getting internal HELIOS++ data. At the moment, it is up to the dev to inject code modifications to extract certain measurements which are currently not exported.

**Solution**
Add a new compilation mode to HELIOS++ so, when compiled this way, it exports data of interest for deep analysis of what is going on behind the simulation. At the moment this is available for certain components such as the BUDDING_METRICS mode. But a more wide spectrum compilation mode for general data will be more useful. It is recommended to implement this as a compilation mode so data analytics related code will not be inserted in normal mode, thus avoiding messing up with performance.

**Caveat**
This compilation mode must not be confused with typical profiling. For such needs already existing tools such as flame graphs, callgrind, massif, gprof, etc. already exist. It must be understood as a way to facilitate analysis of HELIOS++ internals, not as a general purpose for memory profiling nor computational profiling.

han16nah, 2023-12-29:
On Windows, there is an issue with the file separators, i.e., data-analytic files are saved in the HELIOS++ root folder with names like `helios_pulse_records92subray_sim.csv` instead of in a `helios_pulse_records` subfolder.

- [ ] Fix separator for directories so it works on Window

han16nah, 2023-12-29:
- [ ] If it is not too complicated, add an option to store the outputs in subfolders named after the `survey_name` + timestamp to avoid overwriting when running multiple surveys consecutively