3dgeo-heidelberg/helios #174: Non linear legs

albertoesmp, 2022-02-08:
**Problem**
At the moment, HELIOS++ supports platform trajectory specification through linear legs. In consequence, it is not possible to specify a curved trajectory without splitting it into multiple small legs.

**Solution**
Implement leg trajectory specification through multiple points sorted over time. Thus, giving a set of (t, x, y, z) that describe a curve, the leg can be automatically built and handled by HELIOS++

**Tasks**

- [ ] Interpolation from points over time
- [ ] Support non linear legs with constant velocity
- [ ] Compute frontier points for non linear legs with non constant velocity
- [ ] Support non linear legs with acceleration/deceleration

lwiniwar, 2022-02-10:
**Subset of given trajectory**
To separate individual flight strips, it would be good to have the option to define a filter when loading the trajectory. Usage example (`tStart` and `tEnd` are optional):

```
		<leg>
			<platformSettings trajectory="traj01.trj" tStart="413814.3" tEnd="413982.1" template="platform1"/>
			<scannerSettings template="scanner1" />
		</leg>
		<leg>
			<platformSettings trajectory="traj01.trj" tStart="417123.5" tEnd="419000"  template="platform1"/>
			<scannerSettings template="scanner1" />
		</leg>
```
Other optional parameters can give the column ids, similar to the `DetailedVoxels`:
```
<platformSettings trajectory="traj01.trj" tStart="413814.3" tEnd="413982.1">
<param type="int" key="tIndex" value="3" />
<param type="int" key="xIndex" value="0" />
<param type="int" key="yIndex" value="1" />
<param type="int" key="zIndex" value="2" />
<param type="int" key="rollIndex" value="4" />
<param type="int" key="pitchIndex" value="5" />
<param type="int" key="yawIndex" value="6" />
</platformSettings>
```

han16nah, 2022-05-05:
Currently, the trajectory file is expected as a comma separated file. We should have an option `separator` which allows specifying different separators, like for the `xyzloader` of the scene XML.

han16nah, 2023-12-29:
Closing this issue, if any extensions are needed or bugs are found related to non-linear legs, a new issue shall be created.

closed 2023-12-29 by han16nah