3dgeo-heidelberg/helios #209: Output LAZ files remain locked after simulation

mwimmer3005, 2022-04-20:
After finishing the simulation, it is not permitted to change the resulting LAZ files. The following python code leads to a permission error (assuming a simulation object has previously been built):

# [...]
sim.start()
output = sim.join()
fn = output.filepath
sim = None
os.replace(fn, new_fn)

In order to directly put the LAZ files to a more systematic
filename-/folder-structure, it would be convenient to rename/move them after the simulation is finished.

mwimmer3005, 2022-04-29:
Thank you so much for your answer @bhoefle-3dgeo .

My Helios version is 1.1.0, OS Windows 10. I have of course previously double-checked the output file paths, copying the file does work.

However, maybe your hint of the file being blocked is a possible solution. I talked to some colleagues who suggested that files are sometimes locked remarkably long on our network drives. During some more tests, I found out that even after running a different Helios-simulation, the outputs from the previous simulation may still be blocked if the time span is less than about 1 minute.

This makes it somewhat likely that the problem comes from our infrastructure. If this proves true, I'm sorry for bothering you.

bhoefle-3dgeo, 2022-04-29:
This sounds like a valid explanation. Similar issue encounter when working on cloud drives - during sync the files can be locked. Working locally for processing might a good solution. 

Any feedback is welcome and helps us to understand user problems much better. Thanks for pointing to this issue, which other users might also have.

closed 2022-04-29 by bhoefle-3dgeo