3dgeo-heidelberg/helios #219: Helios++ v1.1.0 fails to load/process scenes that can be processed by Helios++ v1.0.9

ChanningYing, 2022-05-03:
Hi,

As the title said, what is the reason and how can I bypass the problem when using v1.1.0. I am using v1.1.0 rather than v1.0.9 because the former one is much faster than the latter.  

Many thanks,
Enoch

bhoefle-3dgeo, 2022-05-03:
Do you mean the XML files of the scene or the binary kdtree files? The binary kdtree files that you built with 1.0.9 are not compatible due to major changes and improvements in the kdtree of HELIOS++ 1.1.0 ([https://github.com/3dgeo-heidelberg/helios/releases/tag/v1.1.0](https://github.com/3dgeo-heidelberg/helios/releases/tag/v1.1.0)). In this case the straightforward solution is to build the binary scene files once and then can reuse it for later simulations. 

Also few parameters in the XML definitions have changed. So far we have no systematic documentation of those parameter changes. Easiest is to check the examples scenes (and their history), e.g. [https://github.com/3dgeo-heidelberg/helios/commits/dev/data/scenes/demo](https://github.com/3dgeo-heidelberg/helios/commits/dev/data/scenes/demo).

Alternatively, please provide the error messages and - if possible - also (parts) of the XML files. This helps us to identify the problem.

ChanningYing, 2022-05-03:
For the v1.1.0, I failed at the exact step of locading xml files of a building floor scene, i.e. cannot creating a .scene file, which should be before creating kdtree?

the error message, the xml files of survey and scene are attached.

It is weird that helios files of other floors can be successfully processed by v1.1.0. These files are prepared by the same software. They should not have difference in syntactic.  

[data.zip](https://github.com/3dgeo-heidelberg/helios/files/8609298/data.zip)

ChanningYing, 2022-05-03:
Thanks so much for the prompt reply!

I located and checked the obj file where the program stops. I did not find any issues. I attached the full set of data here. Hopefully you can help find the problems. 

I actually scanned several buildings. And often there are several storeys of each building that cannot be processed by v1.1.0 but can by v1.0.9. So I am wondering there my exist bugs in v1.1.0. 

[fullset data.zip](https://github.com/3dgeo-heidelberg/helios/files/8610465/fullset.data.zip)

ChanningYing, 2022-05-03:
but v1.0.9 is too slow for big building scenes.

bhoefle-3dgeo, 2022-05-03:
A really great dataset for challenging HELIOS++ with such a large scene (wrt input files). 

By loading the almost 5000 obj files into a 3D software (I used CloudCompare), we found out that two of the obj-files are corrupt and cannot be loaded. We assume this causes the program abort. In the earlier version, this problem might have been circumvented by skipping such files. We will improve error messages regarding damaged input files and will avoid stopping the program for the next release.

The screenshots of the problematic files are enclosed. Either fix them or remove them from or comment in the scene XML (see scene.xml attached to this post). We could run the simulation when removing the who corrupt files from the scene. Don't forget to `--rebuildScene` if you already have a .scene file with the same name.

We would be happy to get feedback if this solved your problem. Then can close this issue.

[[Merged]_F6.zip](https://github.com/3dgeo-heidelberg/helios/files/8612574/Merged._F6.zip)
[image]
![corruptfile2](https://user-images.githubusercontent.com/53827759/166511314-b60d837d-0342-4512-a036-b0102fce0f50.PNG
[image]

ChanningYing, 2022-05-04:
Thanks for the debugging. This does solve my problem. And I find the problem that cause the malformed obj files.

Yes, it would be great for Helios++ to output more meaningful error messages.

I will close this issue. Thanks again.

closed 2022-05-04 by ChanningYing