3dgeo-heidelberg/helios #230: Generated point cloud is cropped at limits of geotiff

pglira, 2022-05-10:
**The problem**
Generated point cloud is cropped at limits of geotiff.

**Actual behavior**
If z_max is the maximum height value of the geotiff file, all generated points have a height < z_max.

**Expected behavior**
No cropping of generated point clouds at z_max.

**Your environment (please complete the following information):**
 - OS: Ubuntu 20.04/docker
 - HELIOS++/pyhelios version: 1.1.0

**Screenshots**

[image]

albertoesmp, 2022-05-10:
First, thank you a lot for reporting this issue. We're solving it as soon as possible.

It would be really helpful to have some extra data to reproduce this bug. Is it possible for you to send us the scene and survey related XML files, as well as the scene parts that you are using? If you don't want to make them public, it is also possible to send them to us by email.

pglira, 2022-05-10:
Thanks for the quick reply and for your help.

All relevant files are contained in this zip archive: 
[helios_data.zip](https://github.com/3dgeo-heidelberg/helios/files/8661845/helios_data.zip)

BTW, this is a screenshot of the point cloud together with the trajectory:
[image]

albertoesmp, 2022-05-11:
The problem was due to the `onGround` method being applied after computing the CRS bounding box (the original one, before translating to the simulation's local coordinate system). It should have been solved at [this commit](https://github.com/3dgeo-heidelberg/helios/commit/de51a4a64598eabf6dee7cafbdf5b70419658ed6). Thus, compiling the current HELIOS++ implementation in the [dev branch](https://github.com/3dgeo-heidelberg/helios) should be enough.

Alternatively, if you don't want to update your code yet, it might be possible to solve this with a simple trick. Take the highest scene part in the scene, in your case the OBJ with the greater z coordinate in the scene space. Instead of using the `onGround` parameter for this tree, calculate its translation by manually specifying the adequate translation vector. This will lead to the correct bounding box. The other trees can be automatically placed with the `onGround` parameter without problems.

I attach an screenshot showing the output I get with the new version.

[image]

Thank you again for reporting this. I hope the solution works for you. Otherwise, please let me know.

pglira, 2022-05-18:
I can confirm that the issue is fixed in the developer branch. Thank you so much!

[image]

closed 2022-05-18 by pglira