3dgeo-heidelberg/helios #233: Export the return point loacation in the waveform

xjy11, 2022-05-16:
Hi,

In the python script provided by helios, the discrete point and fullwave data can be converted to the las 1.4 file and the external waveform packet while the **return_point_wave_location** is set to zero. So, is there any output configuration that can be set to export the point location in the waveform after simulation? Alternatively, if we knew the time bin of the point, it can simply computed with the temporal sample spacing.

Thanks a lot for your help.

lwiniwar, 2022-05-16:
Hi @xjy11,

you can edit the python code to write other values into the `Return Point Wave Location` field:

https://github.com/3dgeo-heidelberg/helios/blob/da206462676a47bab568ba73d4ea8d8c93822694/pyhelios/util/txt2las_wdp.py#L41-L42

The LAS Standard states: 

> The temporal offset in picoseconds (1e−12) from the arbitrary “anchor point” to the location within
the waveform packet for this Point Record.

An approach would be:

```
las.return_point_wave_location = list(range(xyz.shape[0])) * binSize_ns * 1e3
```
where the binSize is defined in your survey XML file.

See more information on LAS waveforms here: https://github.com/ASPRSorg/LAS/wiki/Waveform-Data-Packet-Descriptors-Explained

xjy11, 2022-05-17:
Hi @
Thank you for replying promptly.
I see the simple example to get the return point wave location. According to what I understand, Helios gets discrete points from the fullwave output array via a window method that detects the maximum value. When a bin has the highest value around winSize it is counted as a valid maximum, and the corresponding discrete points are calculated. So, how should I go about obtaining this bin value? Since multi returns may correspond to the same waveform and their time offsets differ in the waveform.

lwiniwar, 2022-05-19:
Hi @xjy11,
thank you for your patience. While the fullwave output of HELIOS++ does not contain the exact point in time attributed to each point, it can easily be calculated from the points coordinates and the trajectory location for that point itself. Consider a snippet like this:

```
# calculate location of return point along beam fwf:
return_point_wave_locations = []
c = 3e8 * 1e-12  # speed of light in m/ps
for point_id, fwf_id in enumerate(pts[:, 7]):
    XYZ_point = xyz[point_id, :]
    XYZ_origin = np.array(origins[fwf_id]) - np.array([100,100,0])
    beamVector = np.array(wfs[fwf_id][2:5])  # must be normalized
    beamVector /= np.linalg.norm(beamVector)
    # nanmean will ensure that even if one component of the beamVector is zero, we get a valid result
    deltaT = np.nanmean((XYZ_point - XYZ_origin) / (c * beamVector)) - wfs[fwf_id][0] * 1e3  # ns to ps
    return_point_wave_locations.append(deltaT)

las.return_point_wave_location = return_point_wave_locations
```

Note the ` - np.array([100,100,0])` - this is the reduction point in the simulation. @albertoesmp, can we change the output in the fullwave to write the non-reduced coordinates for the beam origin instead of the reduced ones? Then this would not be required in the future.

albertoesmp, 2022-05-20:
Hi @xjy11 and @lwiniwar ,

The coordinates of the beam origin are now translated as the output point cloud coordinates. This update is available in the dev branch after this commit: https://github.com/3dgeo-heidelberg/helios/commit/6f3529fe2a44f77e6cd92be2ff65f7f6d8306fd9

Please let me know if there are more problems related to this issue.

closed 2023-12-29 by han16nah