3dgeo-heidelberg/helios #240: pyhelios hanging (?)

celine-mercier, 2022-06-07:
Hello,

I'm trying to play with pyhelios, to eventually be able to create and test trajectories dynamically.

I'm starting with a simple example using trajectories, scenes and code snippets from existing examples, but I can't get it to work. When I run the script, it just uses one core for a seemingly infinite time without any progress.

I'm on Ubuntu 20.04.4 LTS, with an i9-10940X CPU and 128GB or RAM.

What am I doing wrong?

**Script:**

```
# Imports
import pyhelios

if __name__ == "__main__":

    pyhelios.loggingVerbose2()
    pyhelios.loggingDefault()

    pyhelios.setDefaultRandomnessGeneratorSeed("123")

    default_survey_path = 'data/surveys/default_survey.xml'

    # Default survey not containing any legs
    survey = """
    <?xml version="1.0" encoding="UTF-8"?>
    <document>
        <survey name="test" scene="data/scenes/toyblocks/toyblocks_scene.xml#toyblocks_scene" platform="data/platforms.xml#quadcopter" scanner="data/scanners_als.xml#riegl_vux-1uav">
        </survey>
    </document>
    """

    with open(default_survey_path, "w") as f:
        f.write(survey)

    simBuilder = pyhelios.SimulationBuilder(
            default_survey_path,
            'assets/',
            'output/'
        )
    simBuilder.setCallbackFrequency(10)
    simBuilder.setLasOutput(True)
    simBuilder.setFinalOutput(True)

    simB = simBuilder.build()

    # Add legs
    waypoints = [
     [ -50., -50.],
     [ 50., -50.],
     [ 70., -70.],
     [-70., -70.],
     [-70.,  -50.],
     [ 70.,  -50.],
     [ 70.,    0.],
     [-70.,    0.],
     [-70.,   50.],
     [ 70.,   50.],
     [ 70.,  70.],
     [-70.,  70.]]
    altitude = 20
    speed = 7
    shift = simB.sim.getScene().getShift()
    for j, wp in enumerate(waypoints):
        leg = simB.sim.newLeg(j)
        leg.getPlatformSettings().x = wp[0]-shift.x
        leg.getPlatformSettings().y = wp[1]-shift.y
        leg.getPlatformSettings().z = altitude-shift.z
        leg.getPlatformSettings().movePerSec = speed
        leg.getPlatformSettings().onGround = False
        if j % 2 != 0:
            leg.getScannerSettings().active = False

    # Print legs
    for i in range(simB.sim.getNumLegs()):
        leg = simB.sim.getLeg(i)
        print(f'Leg {i}\tposition = '
            f'{leg.getPlatformSettings().x+shift.x},'
            f'{leg.getPlatformSettings().y+shift.y},'
            f'{leg.getPlatformSettings().z+shift.z}\t'
            f'active = {leg.getScannerSettings().active}')

    simB.start()

    output = simB.join()

    # Create instances of vector classes by accessing 'measurements' and 'trajectories' attributes of output wrapper.
    measurements = output.measurements
    trajectories = output.trajectories

    # Get amount of points in trajectory and amount of measurements by accessing length of measurement and trajectory vectors.
    print('Number of measurements : {n}'.format(n=len(measurements)))
    print('Number of points in trajectory : {n}'.format(n=len(trajectories)))
```

**Output:**

```
SimulationBuilder is building simulation ...
xmlDocFilename: default_survey.xml
xmlDocFilePath: data/surveys
xmlDocFilename: scanners_als.xml
xmlDocFilePath: data
Using default value for attribute 'averagePower_w' : 4
Using default value for attribute 'beamQualityFactor' : 1
Using default value for attribute 'opticalEfficiency' : 0.99
Using default value for attribute 'receiverDiameter_m' : 0.15
Using default value for attribute 'atmosphericVisibility_km' : 23
Using default value for attribute 'wavelength_nm' : 1064
Scanner: riegl_vux-1uav Power: 4.000000 W Divergence: 0.500000 mrad Wavelength: 1064 nm Visibility: 23.000000 km
Using default value for attribute 'maxNOR' : 0
Number of subsampling rays: 19
Using default value for attribute 'headRotatePerSecMax_deg' : 0
Using default value for attribute 'rangeMax_m' : 1.79769e+308
xmlDocFilename: platforms.xml
xmlDocFilePath: data
Number of subsampling rays: 19
Using default value for attribute 'numRuns' : 1
Using default value for attribute 'simSpeed' : 1
Loading Scene...
Reading serial scene wrapper object from data/scenes/toyblocks/toyblocks_scene.scene ...
Building KD-Grove... 
KDTree (num. primitives 1110) :
	Max. # primitives in leaf: 34
	Min. # primitives in leaf: 1
	Max. depth reached: 29
	KDTree axis-aligned surface area: 51657.7
	Interior nodes: 3916
	Leaf nodes: 3470
	Total tree cost: 6.50873
KDGrove stats:
	Number of trees: 1
	Number of static trees: 1
	Number of dynamic trees: 0
	Statistics (min, max, total, mean, stdev):
		Building time: (0.0100, 0.0100, 0.0100, 0.0100, 0.0000)
		Tree primitives: (1110, 1110, 1110, 1110.0000, 0.0000)
		Max primitives in leaf: (34, 34, 34, 34.0000, 0.0000)
		Min primitives in leaf: (1, 1, 1, 1.0000, 0.0000)
		Maximum depth: (29, 29, 29, 29.0000, 0.0000)
		Axis-aligned surface area: (51657.7208, 51657.7208, 51657.7208, 51657.7208, 0.0000)
		Number of interior nodes: (3916, 3916, 3916, 3916.0000, 0.0000)
		Number of leaf nodes: (3470, 3470, 3470, 3470.0000, 0.0000)
		Tree cost: (6.5087, 6.5087, 6.5087, 6.5087, 0.0000)

KDG built in 0.01s
Scene loaded in 0.026s
Reading Spectral Library...
SimulationBuilder built simulation in 0.0678533969912678 seconds
Leg 0	position = -50.0,-50.0,20.0	active = True
Leg 1	position = 50.0,-50.0,20.0	active = False
Leg 2	position = 70.0,-70.0,20.0	active = True
Leg 3	position = -70.0,-70.0,20.0	active = False
Leg 4	position = -70.0,-50.0,20.0	active = True
Leg 5	position = 70.0,-50.0,20.0	active = False
Leg 6	position = 70.0,0.0,20.0	active = True
Leg 7	position = -70.0,0.0,20.0	active = False
Leg 8	position = -70.0,50.0,20.0	active = True
Leg 9	position = 70.0,50.0,20.0	active = False
Leg 10	position = 70.0,70.0,20.0	active = True
Leg 11	position = -70.0,70.0,20.0	active = False
output/Survey Playback/test/2022-06-08_10-49-44/
Simulation: Scanner changed!
Starting leg 0

Pulse frequency set to 0
Scan angle set to 180
Applying settings for PolygonMirrorBeamDeflector...Vertical angle min/max nan/nan
 -- verticalAngleMin not set, using the value of -165 degrees
 -- verticalAngleMax not set, using the value of 165 degrees
Leg0 waypoints:
	Origin: (0, 20, 20.2339)
	Target: (100, 20, 20.2339)
	Next: (120, 0, 20.2339)

outputPath=output/Survey Playback/test/2022-06-08_10-49-44/points/leg-01_points.las
```

lwiniwar, 2022-06-08:
Hi @celine-mercier!
Thank you for your question and the code example. Using the Verbose logging option, I could find that the pulse repetition rate of the scanner was set to 0 (internal default) - resulting in no simulation being carried out. By adding

```
        leg.getScannerSettings().pulseFreq = 100_000
```
to the loop where you construct each leg, the simulation started running and reporting progress.

However, I also needed to add
```
        leg.getScannerSettings().scanFreq = 50
```
to get a meaningful output (and not just points below the trajectory).

The multicopter platform still suffers from problems where single legs do not end (#22), which is also the case in your simulation. However, the workaround presented in this issue does not provide a solution here. Instead, I suggest you use the `copter_linearpath` platform which mimics a forward-tilted drone but does not support curves and/or acceleration. We are working on a more rigorous implementation of the multicopter platform.

Finally, the legs are currently overwriting each other as they all have the same ID (and writing 0 byte files in the end). This is a deeper-rooted issue that cannot be solved on `python`-side alone. @albertoesmp, I guess we need to export `leg.getStrip()` to python and allow overwriting the `stripID` there? And can we create a default incrementing counter when legs are generated with `sim.newLeg(int id)`?

So, if you keep the `multicopter` platform, the simulation does not end, but you can access and look at the output files (at least if you remove the `simBuilder.setLasOutput(True)`) for now. We'll report once this issue is fixed.

Cheers,
Lukas

lwiniwar, 2022-06-08:
@celine-mercier that sounds like a viable workaround for now. Sorry for your inconveniences with that!

lwiniwar, 2022-06-10:
Hi @celine-mercier, we have good news.
In current builds of the `dev` branch (which you can download from the "Actions" tab when you are logged in to github - scroll down to "Artifacts", e.g. here (for Linux) https://github.com/3dgeo-heidelberg/helios/actions/runs/2474747830) @albertoesmp implemented strip handling. Just download and extract/overwrite the version corresponding to your python installation into the `/run` folder.
You can then change the loop generating the legs to:

```
    for j, wp in enumerate(waypoints):
        leg = simB.sim.newLeg(j)
        strip = simB.sim.newScanningStrip(f"strip_{j}")
        leg.getPlatformSettings().x = wp[0]-shift.x
        leg.getPlatformSettings().y = wp[1]-shift.y
        leg.getPlatformSettings().z = altitude-shift.z
        leg.getPlatformSettings().movePerSec = speed
        leg.getPlatformSettings().onGround = False
        leg.getScannerSettings().pulseFreq = 100_000
        leg.getScannerSettings().scanFreq = 50
        leg.serialId = j
        if j % 2 != 0:
            leg.getScannerSettings().active = False
        simB.sim.assocLegWithScanningStrip(leg, strip)
```

Let us know if that works for you!

Cheers,
Lukas

celine-mercier, 2022-06-12:
Hi Lukas, yay! Thanks a lot for implementing this. 
I overwrote the files in the run folder with the files from [Linux-helios-pybinds-3.8](https://github.com/3dgeo-heidelberg/helios/suites/6878798983/artifacts/266306243), but it looks like the build is missing something:

```
In [1]: import pyhelios                                                                                                                                                      
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-3a55a116722f> in <module>
----> 1 import pyhelios

~/workspace/NS-TIP/helios_software/helios-plusplus-lin/pyhelios/__init__.py in <module>
----> 1 from ._pyhelios_relay import *
      2 
      3 from .output_handling import outputToNumpy
      4 from .coordinates_utils import \
      5     cartesianToSpherical, sphericalToCartesian

~/workspace/NS-TIP/helios_software/helios-plusplus-lin/pyhelios/_pyhelios_relay.py in <module>
      7 # append run folder to sys path before import
      8 sys.path.append(str(Path(__file__).parent.parent / "run"))
----> 9 from _pyhelios import *

ImportError: /home/scion.local/mercierc/workspace/NS-TIP/helios_software/helios-plusplus-lin/run/_pyhelios.so: undefined symbol: _ZNK3Leg10getPyStripEv
```

What do you think? Cheers

celine-mercier, 2022-06-28:
Hello, sorry for bothering you again, but I would really appreciate if you could recompile the files for me :)

lwiniwar, 2022-07-01:
Hi @celine-mercier,

thank you for your patience! I have just downloaded the latest release again (https://github.com/3dgeo-heidelberg/helios/releases/tag/v1.1.0), overwrote the three files `libhelios.so`, `_pyhelios.so` and `helios` with the ones in the provided ZIP and tried running that. I can import `pyhelios` in python with no problems. I am using a fresh Ubuntu 20.04LTS install. 
Are you able to run the `helios` executable, e.g. with the demo scene `data\surveys\demo\tls_arbaro_demo.xml` ?

Cheers,
Lukas

celine-mercier, 2022-07-01:
Hi Lukas, 

Thanks a lot for looking into it. 

I can run the helios executable, it runs fine with the demo scene.

Unfortunately I still can't import pyhelios with a fresh install. I tried running inside and outside the conda environment, but I get the same undefined symbol error.

What python version are you using? I might try with other versions to see if that fixes the issue.

```
(pyhelios_env) 14:47/helios-plusplus-lin_1$ which ipython
/home/scion.local/mercierc/miniconda3/envs/pyhelios_env/bin/ipython
(pyhelios_env) 14:48/helios-plusplus-lin_1$ ipython
Python 3.8.13 | packaged by conda-forge | (default, Mar 25 2022, 06:04:10) 
Type 'copyright', 'credits' or 'license' for more information
IPython 8.4.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import pyhelios
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
Input In [1], in <cell line: 1>()
----> 1 import pyhelios

File ~/workspace/NS-TIP/helios_pyhelios/helios-plusplus-lin_1/pyhelios/__init__.py:1, in <module>
----> 1 from ._pyhelios_relay import *
      3 from .output_handling import outputToNumpy
      4 from .coordinates_utils import \
      5     cartesianToSpherical, sphericalToCartesian

File ~/workspace/NS-TIP/helios_pyhelios/helios-plusplus-lin_1/pyhelios/_pyhelios_relay.py:9, in <module>
      7 # append run folder to sys path before import
      8 sys.path.append(str(Path(__file__).parent.parent / "run"))
----> 9 from _pyhelios import *

ImportError: /home/scion.local/mercierc/workspace/NS-TIP/helios_pyhelios/helios-plusplus-lin_1/run/_pyhelios.so: undefined symbol: _ZNK3Leg10getPyStripEv
```

celine-mercier, 2022-07-01:
Update: I am getting the same error with Python 3.9 and 3.7

lwiniwar, 2022-07-04:
(solved off-site)

Always make sure to add the path of the `run` directory to the `LD_LIBRARY_PATH` variable (when running on Linux).

e.g.

```
export LD_LIBRARY_PATH=/path/to/helios/run/:$LD_LIBRARY_PATH
/path/to/helios/run/helios <path to survey file>
```

closed 2022-07-04 by lwiniwar