3dgeo-heidelberg/helios #245: pyhelios: empty trajectories when creating legs

celine-mercier, 2022-07-06:
Hello, sorry, it's me again :)

Now that I can add legs through pyhelios (https://github.com/3dgeo-heidelberg/helios/issues/240), I am trying to draw a simple square. The simulation runs, but I am getting empty trajectories in the output. The leg waypoints also seem wrong in the logs. Am I doing something wrong?

```
import os
import sys
from pathlib import Path

helios_path = str(Path.cwd().parent)
sys.path.append(helios_path)
os.chdir(helios_path)

import pyhelios

pyhelios.loggingVerbose2()
pyhelios.loggingDefault()

pyhelios.setDefaultRandomnessGeneratorSeed("123")

default_survey_path = 'data/surveys/default_survey.xml'

# default survey not containing any legs
survey = """
<?xml version="1.0" encoding="UTF-8"?>
<document>
    <survey name="test" scene="data/scenes/toyblocks/toyblocks_scene.xml#toyblocks_scene" platform="data/platforms.xml#copter_linearpath" scanner="data/scanners_als.xml#riegl_vux-1uav">
    </survey>
</document>
"""

with open(default_survey_path, "w") as f:
    f.write(survey)

simBuilder = pyhelios.SimulationBuilder(
        default_survey_path,
        'assets/',
        'output/'
    )
simBuilder.setCallbackFrequency(10)
simBuilder.setLasOutput(True)
simBuilder.setFinalOutput(True)

simB = simBuilder.build()

#Add legs
waypoints = [
 [ 50., -50.],
 [ -50., -50.],
 [ -50., 50.],
 [ 50., 50.],
 [ 50.,  -50.]]
altitude = 20
speed = 7
shift = simB.sim.getScene().getShift()
for j, wp in enumerate(waypoints):
    leg = simB.sim.newLeg(j)
    strip = simB.sim.newScanningStrip(f"strip_{j}")
    leg.getPlatformSettings().x = wp[0]-shift.x
    leg.getPlatformSettings().y = wp[1]-shift.y
    leg.getPlatformSettings().z = altitude-shift.z
    leg.getPlatformSettings().movePerSec = speed
    leg.getPlatformSettings().onGround = False
    leg.getScannerSettings().pulseFreq = 100_000
    leg.getScannerSettings().scanFreq = 50
    leg.serialId = j
    if j % 2 != 0:
        leg.getScannerSettings().active = False
    simB.sim.assocLegWithScanningStrip(leg, strip)

# Print legs
shift = simB.sim.getScene().getShift()
for i in range(simB.sim.getNumLegs()):
    leg = simB.sim.getLeg(i)
    print(f'Leg {i}\tposition = '
        f'{leg.getPlatformSettings().x+shift.x},'
        f'{leg.getPlatformSettings().y+shift.y},'
        f'{leg.getPlatformSettings().z+shift.z}\t'
        f'active = {leg.getScannerSettings().active}')

simB.start()

output = simB.join()

# Read output
measurements = output.measurements
trajectories = output.trajectories
print('Number of measurements : {n}'.format(n=len(measurements)))
print('Number of points in trajectory : {n}'.format(n=len(trajectories)))
```

The output:

```
/home/scion.local/mercierc/workspace/NS-TIP/helios_pyhelios/helios-plusplus-lin_1
SimulationBuilder is building simulation ...
SimulationBuilder built simulation in 0.017281091000768356 secondsxmlDocFilename: default_survey.xml
xmlDocFilePath: data/surveys
xmlDocFilename: scanners_als.xml
xmlDocFilePath: data
Using default value for attribute 'averagePower_w' : 4
Using default value for attribute 'beamQualityFactor' : 1
Using default value for attribute 'opticalEfficiency' : 0.99
Using default value for attribute 'receiverDiameter_m' : 0.15
Using default value for attribute 'atmosphericVisibility_km' : 23
Using default value for attribute 'wavelength_nm' : 1064
Scanner: riegl_vux-1uav Power: 4.000000 W Divergence: 0.500000 mrad Wavelength: 1064 nm Visibility: 23.000000 km
Using default value for attribute 'maxNOR' : 0
Number of subsampling rays: 19
Using default value for attribute 'headRotatePerSecMax_deg' : 0
Using default value for attribute 'rangeMax_m' : 1.79769e+308
xmlDocFilename: platforms.xml
xmlDocFilePath: data
Number of subsampling rays: 19
Using default value for attribute 'numRuns' : 1
Using default value for attribute 'simSpeed' : 1
Loading Scene...
Reading serial scene wrapper object from data
/scenes/toyblocks/toyblocks_scene.scene ...
Building KD-Grove... 
KDTree (num. primitives 1110) :
	Max. # primitives in leaf: 34
	Min. # primitives in leaf: 1
	Max. depth reached: 29
	KDTree axis-aligned surface area: 51657.7
	Interior nodes: 3926
	Leaf nodes: 3480
	Total tree cost: 6.50873
KDGrove stats:
	Number of trees: 1
	Number of static trees: 1
	Number of dynamic trees: 0
	Statistics (min, max, total, mean, stdev):
		Building time: (0.0050, 0.0050, 0.0050, 0.0050, 0.0000)
		Tree primitives: (1110, 1110, 1110, 1110.0000, 0.0000)
		Max primitives in leaf: (34, 34, 34, 34.0000, 0.0000)
		Min primitives in leaf: (1, 1, 1, 1.0000, 0.0000)
		Maximum depth: (29, 29, 29, 29.0000, 0.0000)
		Axis-aligned surface area: (51657.7208, 51657.7208, 51657.7208, 51657.7208, 0.0000)
		Number of interior nodes: (3926, 3926, 3926, 3926.0000, 0.0000)
		Number of leaf nodes: (3480, 3480, 3480, 3480.0000, 0.0000)
		Tree cost: (6.5087, 6.5087, 6.5087, 6.5087, 0.0000)

KDG built in 0.005s
Scene loaded in 0.013s
Reading Spectral Library...

Leg 0	position = 50.0,-50.0,20.0	active = True
Leg 1	position = -50.0,-50.0,20.0	active = False
Leg 2	position = -50.0,50.0,20.0	active = True
Leg 3	position = 50.0,50.0,20.0	active = False
Leg 4	position = 50.0,-50.0,20.0	active = True

Output directory: "output//test/2022-07-06_13-35-09/"
Simulation: Scanner changed!
Starting leg 0

Pulse frequency set to 100000
Scan angle set to 180
Applying settings for PolygonMirrorBeamDeflector...
Vertical angle min/max nan/nan
 -- verticalAngleMin not set, using the value of -165 degrees
 -- verticalAngleMax not set, using the value of 165 degrees
Leg0 waypoints:
	Origin: (100, 20, 20.2339)
	Target: (0, 20, 20.2339)
	Next: (0, 120, 20.2339)

Survey 0.25%	Elapsed 00 00:00:00 Remaining 00 00:00:15
Leg1/4 1.00%	Elapsed 00 00:00:00 Remaining 00 00:00:03
Survey 0.50%	Elapsed 00 00:00:00 Remaining 00 00:00:12
[...]
Survey 24.75%	Elapsed 00 00:00:01 Remaining 00 00:00:05
Leg1/4 99.00%	Elapsed 00 00:00:01 Remaining 00 00:00:00
Waypoint reached!
Starting leg 1

Pulse frequency set to 100000
Scan angle set to 180
Applying settings for PolygonMirrorBeamDeflector...
Vertical angle min/max nan/nan
 -- verticalAngleMin not set, using the value of -165 degrees
 -- verticalAngleMax not set, using the value of 165 degrees
Iterative mode was used for manual leg initialization because default one failed for MovingPlatform
Leg1 waypoints:
	Origin: (0, 20, 20.2339)
	Target: (0, 120, 20.2339)
	Next: (100, 120, 20.2339)

Survey 25.25%	Elapsed 00 00:00:01 Remaining 00 00:00:05
Leg2/4 1.00%	Elapsed 00 00:00:00 Remaining 00 00:00:00
Survey 25.50%	Elapsed 00 00:00:01 Remaining 00 00:00:05
[...]
Survey 49.75%	Elapsed 00 00:00:01 Remaining 00 00:00:01
Leg2/4 99.00%	Elapsed 00 00:00:00 Remaining 00 00:00:00
Waypoint reached!
Starting leg 2

Pulse frequency set to 100000
Scan angle set to 180
Applying settings for PolygonMirrorBeamDeflector...
Vertical angle min/max nan/nan
 -- verticalAngleMin not set, using the value of -165 degrees
 -- verticalAngleMax not set, using the value of 165 degrees
Iterative mode was used for manual leg initialization because default one failed for MovingPlatform
Leg2 waypoints:
	Origin: (0, 120, 20.2339)
	Target: (100, 120, 20.2339)
	Next: (100, 20, 20.2339)

Survey 50.25%	Elapsed 00 00:00:01 Remaining 00 00:00:01
Leg3/4 1.00%	Elapsed 00 00:00:00 Remaining 00 00:00:02
Survey 50.50%	Elapsed 00 00:00:01 Remaining 00 00:00:01
[...]
Survey 74.75%	Elapsed 00 00:00:03 Remaining 00 00:00:01
Leg3/4 99.00%	Elapsed 00 00:00:01 Remaining 00 00:00:00
Waypoint reached!
Starting leg 3

Pulse frequency set to 100000
Scan angle set to 180
Applying settings for PolygonMirrorBeamDeflector...
Vertical angle min/max nan/nan
 -- verticalAngleMin not set, using the value of -165 degrees
 -- verticalAngleMax not set, using the value of 165 degrees
Iterative mode was used for manual leg initialization because default one failed for MovingPlatform
Leg3 waypoints:
	Origin: (100, 120, 20.2339)
	Target: (100, 20, 20.2339)
	Next: (100, 20, 20.2339)

Survey 75.25%	Elapsed 00 00:00:03 Remaining 00 00:00:01
Leg4/4 1.00%	Elapsed 00 00:00:00 Remaining 00 00:00:00
Survey 75.50%	Elapsed 00 00:00:03 Remaining 00 00:00:01
[...]
Survey 99.75%	Elapsed 00 00:00:03 Remaining 00 00:00:00
Leg4/4 99.00%	Elapsed 00 00:00:00 Remaining 00 00:00:00
Waypoint reached!
Starting leg 4

Pulse frequency set to 100000
Scan angle set to 180
Applying settings for PolygonMirrorBeamDeflector...
Vertical angle min/max nan/nan
 -- verticalAngleMin not set, using the value of -165 degrees
 -- verticalAngleMax not set, using the value of 165 degrees
Waypoint reached!
Elapsed simulation steps = 5714293
Elapsed virtual time = 57.1429 sec.
Main thread simulation loop finished in 3.85998 sec.
Waiting for completion of pulse computation tasks...
Pulse computation tasks finished in 3.86017 sec.
output = simB.join()
output = simB.join()

Number of measurements : 955497
Number of points in trajectory : 0
```

han16nah, 2022-07-07:
Dear @celine-mercier 
1. please try adding `leg.getScannerSettings().trajectoryTimeInterval = 0.001` (you can also specify another small number > 0) in the loop where you go through the waypoints and set scanner and platform settings.

2. `"The leg waypoints also seem wrong in the logs"` 
--> I am pasting a section from the Wiki below 

[Python bindings 🐍 Details](https://github.com/3dgeo-heidelberg/helios/wiki/Python-bindings-%F0%9F%90%8D-Details)
---
When loading a survey, a shift is applied to the scene and to each leg. We can obtain this shift:

```python
scene = simB.sim.getScene()
shift = scene.getShift()
print(f'Shift = {shift.x},{shift.y},{shift.z}')
```

Example output for data/toyblocks/als_toyblocks.xml:

```console
Shift = -50.0,-70.0,-0.233912
```

Using a for-loop, we can get all leg positions. Note that we add the shift to obtain the coordinates as specified in the XML-file:

```python
for i in range(simB.sim.getNumLegs()):
    leg = simB.sim.getLeg(i)
    print(f'Leg {i}\tposition = '
          f'{leg.getPlatformSettings().x+shift.x},'
          f'{leg.getPlatformSettings().y+shift.y},'
          f'{leg.getPlatformSettings().z+shift.z}\t'
          f'active = {leg.getScannerSettings().active}')
```

Example output for data/toyblocks/als_toyblocks.xml:

```console
Leg 0	position = -30.0,-50.0,100.0	active = True
Leg 1	position = 70.0,-50.0,100.0	active = False
Leg 2	position = 70.0,0.0,100.0	active = True
Leg 3	position = -30.0,0.0,100.0	active = False
Leg 4	position = -30.0,50.0,100.0	active = True
Leg 5	position = 70.0,50.0,100.0	active = False
```

celine-mercier, 2022-07-10:
Hello, yay, adding the suggested line fixed it, thanks a lot. What does this `trajectoryTimeInterval` attribute correspond to? Should I worry about its value?

I also have one last (for now :)) unrelated question, in the `trajectory_plot_demo.ipynb` example, is there a reason why the scanner is turned off for every other leg (`if j % 2 != 0: leg.getScannerSettings().active = False`)?

Thanks a lot for the help, cheers

lwiniwar, 2022-07-11:
Hi @celine-mercier,
the `trajectoryTimeInterval` value represents the interval between two lines written to the trajectory file - or the sampling density of the trajectory, if you wish.

About your other question: The goal in this notebook is to create parallel flight lines, therefore the scanner is disabled for the legs where the platform moves from the end of the first flight line to the start of the second one.

Does that make it clearer to you?
Cheers,
Lukas

celine-mercier, 2022-07-11:
Hi Lukas,

Yes thank you, this is very helpful. 

We have one last question for now: is there an ETA for quadcopter support (we talked about it in the other issue https://github.com/3dgeo-heidelberg/helios/issues/240)? Should we just keep an eye on updates?

Cheers
Celine

lwiniwar, 2022-07-12:
Hi @celine-mercier,
thanks for the feedback.

We are investigating the quadrocopter issue over the summer and are hoping for a relatively easy solution. If this should not be feasible, we will suggest to use external tools (flight simulators) to create a trajectory and then import this into HELIOS++ using the support for interpolated trajectories (see https://github.com/3dgeo-heidelberg/helios/wiki/Survey#scanner-settings-and-platform-settings for an example). If you can work with this workaround, great, otherwise I would say check back around September to know if we will support this feature or drop it.

Feel free to open a new issue if you run into any other problems/questions!

Cheers,
Lukas

closed 2022-07-12 by lwiniwar

celine-mercier, 2022-07-12:
OK, thanks again!