3dgeo-heidelberg/helios #254: Scene part partially loaded ?

NanBlanc, 2022-08-02:
**The problem**
Some object parts seems to be ignored in the collision with the laser beams, resulting in incomplete objects in point clouds.

**How to reproduce**
1. Recompiling lastest code from git
2. Use a simple scene : Tiff as ground (25x25m), one .obj model for tree axis and one .obj for leafs (8.7, 6.2 and 6.5 Mo respectively). 
3. Do ALS scan.

I can join all the files if needed.

**Your environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - HELIOS++/pyhelios version (or Git revision) that exhibits the issue : Version 1.1.0
 - Last HELIOS++/pyhelios version that did not exhibit the issue (if applicable): Version 1.1.0 (no --version arguement, so hard to tell but I downloaded my old version on July 8 2022)
 - Further system details (if relevant): 64 RAM 

**Screenshots (Optional)**
Expected :
[image]

Actual :
[image]

**Ideas for possible solutions (Optional)**
When not using the latest compilation, my example works fine, which means recent modifications may have introduced this error. Funny things that can help to pin point the source : 
When running the simulation with not-the-last-compilation on the same scene, then redoing it with the lastest version, but without --rebuildScene, everything is fine. Which means it has something to do with the scene building

albertoesmp, 2022-08-02:
Thank you for reporting this. It looks very interesting that the bug was not present in July 08. It is a really nice hint. I'm in the middle of a lot of work but I'll try to solve this as soon as possible.

albertoesmp, 2022-08-03:
Hello again, @NanBlanc .

I've been checking the latest updates but could not find any obvious modification on the scene loading procedure after the 8th of July. I would like to use your survey to debug this issue, so I can start by reproducing it exactly to be on the safe side. Would you mind sharing your survey, scene, and scene parts (tiff file included) data with us? If you are not comfortable with publishing them here at GitHub you can send them privately to us through our email [helios@uni-heidelberg.de](mailto:helios@uni-heidelberg.de)

bhoefle-3dgeo, 2022-08-03:
It might also be good to check the parameters of each version (i.e. have some default parameters changed between the used versions). @albertoesmp, we should also consider writing an extra nicely formatted "log" file for all the parameters used for a simulation in a proper format that allows quick comparison, including all parameters (also default and user set ones). 

e.g. --logFileParameters (?)

albertoesmp, 2022-08-03:
> It might also be good to check the parameters of each version (i.e. have some default parameters changed between the used versions). @albertoesmp, we should also consider writing an extra nicely [...]

I think we have a debugging tool that does something quite similar to what you propose. See https://github.com/3dgeo-heidelberg/helios/blob/dev/doc/debug/DEBUG.md for more information, more concretely the "JSON Report" subsection. With this we can generate JSON files storing the internal state (e.g. survey, scanner and platform variables). Then using the python script at scripts/debug/hda_diff_report.py the JSON files are compared and any difference between them is explicitly pointed by the script output. I can do this once I have the survey to debug this issue. I think it is a very useful tool.

NanBlanc, 2022-08-03:
I just found out that your precompiled version for ubuntu 20.04 https://github.com/3dgeo-heidelberg/helios/releases/download/v1.1.0/helios-plusplus-lin.tar.gz of the 15 march shows the same problem

albertoesmp, 2022-08-04:
Hello again @NanBlanc 

I have some updates on this. I tried the survey with the current development version of HELIOS++ and the output looks good to me. Please, see the screenshot below and correct me if I'm wrong.
[image]

If I'm right with my previous statement, then I think that the issue might be related to this one: https://github.com/3dgeo-heidelberg/helios/issues/230 In fact, you are also using the `onGround` parameter to place your tree automatically at ground level, which was causing the problem with previous versions.

If I'm right, the good news is that the issue is already solved in the latest development version of the software. You have different options to check this. You could try to compile it manually from the development branch. Alternatively, you could check the artifacts that are generated through Github CI workflow (scroll down a little to see them).

Artifacts for Linux: https://github.com/3dgeo-heidelberg/helios/actions/runs/2796520666
Artifacts for Windows: https://github.com/3dgeo-heidelberg/helios/actions/runs/2796520667

Please, let us know your results if you try with the newer version.

albertoesmp, 2022-08-04:
> I didn't paid attention i was -b 1.1.0 when git cloning, so i indeed didn't used the last dev version. Thanks for the help, It was the same problem as #230. Sorry for the useless bug report.

Don't worry! Thank you for using HELIOS++

I'm closing this issue because it is solved.

Please, let us know if we can help you with anything else.

closed 2022-08-04 by albertoesmp