3dgeo-heidelberg/helios #257: Leg skip (Teleportation powers)

NanBlanc, 2022-08-04:
**Is your feature request related to a problem? Please describe.**
No 

**Describe the solution you'd like**
A leg flag to ignore a leg, just to be able to "teleport" to the start of the next point, without any scan of trajectory recording. (maybe there is a way of doing it that i missed in the wiki ?)

**Describe alternatives you've considered**
I tried to set `trajectoryTimeInterval_s=0.0` but this only removes the trajectory, and does not ignore the scanning. Then i just dumped all unwanted legs into the same "bin cloud". And finnally I raised the `movePerSec_m` to light speed, so it feels like teleportation, but it seems kind of cheating.

**Additional context**
What i want :
[image]

What i have to do : 
[image]

lwiniwar, 2022-08-04:
Hi @NanBlanc,

have you tried setting `active="False"` in the scanner settings of your survey leg?
https://github.com/3dgeo-heidelberg/helios/wiki/Survey#scannersettings--within-the-leg-tag

See e.g. the example here:

https://github.com/3dgeo-heidelberg/helios/blob/3666f67e141f76fefb6b424b67be8257b0907dd9/data/surveys/toyblocks/als_toyblocks.xml#L7-L18

Cheers,
Lukas

NanBlanc, 2022-08-05:
This is pretty much what I needed, thanks for pointing that out, I wasn't able to figure it out alone from the wiki.

closed 2022-08-05 by lwiniwar

NanBlanc, 2022-08-10:
I just figured out that this survey :

            <leg>
			<platformSettings x="-240" y="-10" z="1200.000" onGround="false" movePerSec_m="74.59" />
			<scannerSettings template="set" trajectoryTimeInterval_s="0.05" />
		</leg>
		
		<leg stripId="dump">
			<platformSettings x="-240" y="30" z="1200.000" onGround="false" movePerSec_m="10000000000000000" />
			<scannerSettings template="set" active="false" />
		</leg>
		
		<leg>
			<platformSettings x="-115" y="-10" z="1200.000" onGround="false" movePerSec_m="74.59" />
			<scannerSettings template="set" trajectoryTimeInterval_s="0.05" />
		</leg>
		<leg stripId="dump">
			<platformSettings x="-115" y="30" z="1200.000" onGround="false" movePerSec_m="10000000000000000" />
			<scannerSettings template="set" active="false" />
		</leg>
outputs, on the first leg a correct point cloud, but keep the dumb speed for all legs, so i will only have point on the first leg, and none on the other that still are activated.

lwiniwar, 2022-08-10:
I would assume that with that speed the survey is considered finished after your second leg, as there are no more beams left to simulate for the rest of the path. The simulation frequency (including location updates) is defined as the pulse repetition rate of the scanner, so using 3,000,000 times the speed of light as the platform speed does not work with this definition.

Is there a specific reason why you need to have it that high if the scanner is inactive?

NanBlanc, 2022-08-10:
not at all, i just didn't removed it while using the `active=false` and didn't expect it to have an influence on the simulation, so I just wanted to let you know.
by removing it, everything is fine.

lwiniwar, 2022-08-10:
Thanks for reporting! Given that highly unrealistic speed, I think we will consider this a non-issue, though.