3dgeo-heidelberg/helios #275: HELIOS writes LAS 1.0 instead of 1.4 if `lasOutput` is set

han16nah, 2022-10-30:
**The problem**
It is stated that HELIOS++ writes LAS 1.4 by default if `--lasOutput` is specified (unless `--las10` is explicitly set). However, HELIOS++ writes LAS 1.0. I checked with `lasinfo` by LAStools.

Snippet of the output:
```
reporting all LAS header entries:
  file signature:             'LASF'
  file source ID:             0
  global_encoding:            0
  project ID GUID data 1-4:   00000000-0000-0000-0000-000000000000
  version major.minor:        1.0
  system identifier:          ''
  generating software:        'HELIOS++V1.1.1'
  file creation day/year:     302/2022
```
 

**How to reproduce**

1. Run any survey with the current HELIOS++ version and use the `lasOutput` flag, e.g.: `run\helios data\surveys\toyblocks\als_toyblocks.xml --lasOutput`
2. Check the header of the output files, e.g. with lastools: `lasinfo -i output\toyblocks_als\2022-10-30_20-37-15\leg000_points.las` --> Reports `version major.minor:        1.0`

**Expected behavior**
HELIOS++ writes LAS 1.4 if `--lasOutput` is set (and `--las10` is not).

**Actual behavior**
HELIOS++ always writes LAS 1.0 if `--lasOutput` is set, even if `--las10`  is not set.

**Environment:**
 - OS: Windows 10
 - HELIOS++/pyhelios version (or Git revision) that exhibits the issue: 1.1.1

**Additional context**
Maybe the problem was introduced with the File Management System. **When we still wrote the output into the "Survey Playback" folder, H++ still correctly wrote LAS 1.4** (I checked with old output files I have locally).

albertoesmp, 2022-11-17:
This bug should be solved in this PR: https://github.com/3dgeo-heidelberg/helios/pull/284

Please, let me know if it persists.

han16nah, 2022-11-22:
@albertoesmp Now LAS 1.4 is written by default, however, the `gps_time` is missing in the LAS files

albertoesmp, 2022-11-24:
> @albertoesmp Now LAS 1.4 is written by default, however, the `gps_time` is missing in the LAS files

LAS 1.4 has been updated from point format 0 to point format 6 to include mandatory GPS time (https://github.com/3dgeo-heidelberg/helios/commit/106191c00b41074462d536acb2e9d670797153f2).

Please, let me know if further updates are required.

closed 2022-12-14 by han16nah