3dgeo-heidelberg/helios #276: Object instance correspondance

chenzhaiyu, 2022-11-07:
Hi! in virtual surveying on Wavefront OBJ files, is there a way to associate the simulated point clouds with corresponding OBJ models? The use case is to simulate a scene of buildings while being able to link to the original OBJ models given per-building point clouds one by one. It seems the `hitObjectId` attribute is useable for separating the buildings (CloudCompare screenshot attached), however, the one-by-one mapping is unknown. Any hints? Thanks in advance. 

[image]

chenzhaiyu, 2022-11-07:
To help with explaining, this is an example of my scene XML:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<document>
  <scene id="helsinki_scene" name="HelsinkiScene">
    <part id="0">
      <filter type="objloader">
        <param type="string" key="filepath" value="data/helsinki\BID_0019d516-d07e-4315-957c-e4c0335d95e2.obj"/>
      </filter>
      <filter type="scale">
        <param type="double" key="scale" value="100"/>
      </filter>
      <filter type="translate">
        <param type="vec3" key="offset" value="0;0;20.0"/>
      </filter>
    </part>
    <part id="1">
      <filter type="objloader">
        <param type="string" key="filepath" value="data/helsinki\BID_002accf5-775f-4ab4-a9fb-266fe452848e.obj"/>
      </filter>
      <filter type="scale">
        <param type="double" key="scale" value="100"/>
      </filter>
      <filter type="translate">
        <param type="vec3" key="offset" value="0;0;20.0"/>
      </filter>
    </part>
    <part id="2">
      <filter type="objloader">
        <param type="string" key="filepath" value="data/helsinki\BID_00ac506c-24b6-4483-ae1d-49477b1df631.obj"/>
      </filter>
      <filter type="scale">
        <param type="double" key="scale" value="100"/>
      </filter>
      <filter type="translate">
        <param type="vec3" key="offset" value="0;0;20.0"/>
      </filter>
    </part>
    ......
  </scene>
</document>
```

What I would need is for each building in the above screenshot the filename of the corresponding OBJ model, e.g., `BID_00ac506c-24b6-4483-ae1d-49477b1df631.ob`.

lwiniwar, 2022-11-07:
Hi Chen,
thank you for your question! With your XML as defined above, everything enclosed in `<part id="0">` will get the `hitObjectId` set to `0`. So in your example,

BID_0019d516-d07e-4315-957c-e4c0335d95e2.obj -> ID 0
BID_002accf5-775f-4ab4-a9fb-266fe452848e.obj -> ID 1
BID_00ac506c-24b6-4483-ae1d-49477b1df631.obj -> ID 2
...

I suppose this is the one-to-one mapping you are looking for? If you generate the scene XML automatically, I guess you could just export that mapping when you write the XML, or alternatively use an XML parser in post-processing to read it from the scene XML.

Finally, if you do not define `part id`s, the `hitObjectId` will start with the first `objloader` object at zero and then increase as in the order in the file.

I hope this clarifies!

Cheers,
Lukas

chenzhaiyu, 2022-11-07:
Hi Lukas, thank you very much for your reply! Can't believe I didn't catch the apparent linking between `hitObjectId` and the `<part id>`. This is super clear to me now :)

closed 2022-11-07 by chenzhaiyu