3dgeo-heidelberg/helios #277: `scanAngle_deg` in survey XML has no effect.

chenzhaiyu, 2022-11-09:
**The problem**

The `scanAngle_deg` value in survey XML has no effect.

**How to reproduce**
1. Specify scanner and platform respectively as `scanners_als.xml#riegl_vq-880g` and `platforms.xml#sr22`.
2. Assign any value other than 20 (e.g., 0 in the example below) to `scanAngle_deg` parameter in the `<scannerSettings [...]>` tag or in the `<scannerSettings [...]/>` within the `<leg>` tag.
3. Start surveying.

- survey XML:
```xml
<?xml version="1.0"?>
<document>
	<platformSettings id="platform1" movePerSec_m="20" />
    <scannerSettings active="true" id="scanner1" pulseFreq_hz="550000" scanAngle_deg="0" scanFreq_hz="160" trajectoryTimeInterval_s="0.01"/>
    <survey name="helsinki_als" platform="conf/platforms.xml#sr22" scanner="conf/scanners_als.xml#riegl_vq-880g" scene="conf/helsinki_scene.xml#helsinki_scene">
	<!-- platform: plane, deflector: conic -->
		<leg stripId="mission">
			<platformSettings x="-100.0" y="-200.0" z="400.0" onGround="false" template="platform1"/>
			<scannerSettings template="scanner1" scanAngle_deg="0" />
		</leg>
                <!-- more legs following -->
	</survey>
</document>
```

**Expected behavior**
Scan angle set to the specified value (0 in the above example).

**Actual behavior**
```
...
Starting leg 0

Pulse frequency set to 550000
Scan angle set to 20
...
```

**Your environment (please complete the following information):**
 - OS: Windows 11
 - HELIOS++/pyhelios version: 1.1.0

**Screenshots**
Scan angle looks indeed non-0.
[image]

lwiniwar, 2022-11-09:
Hi @chenzhaiyu,

the RIEGL VQ-880G is a conical ("Palmer-") scanner, which does not have a selectable scanAngle. In consequence, only the `maxScanAngle_deg` defined in the scanner XML is used as half-opening angle of the cone.

If you use different scanners, such as the Leica ALS50 or the RIEGL VQ-780i, you should be able to set the scan angle as desired.

Also note Issue #141 on the topic.

I'm leaving this issue open as the documentation is clearly lacking here. Sorry for that!

han16nah, 2023-12-29:
Documentation has been added to the [Scanner](https://github.com/3dgeo-heidelberg/helios/wiki/Scanners) and [Survey](https://github.com/3dgeo-heidelberg/helios/wiki/Survey) pages.

closed 2023-12-29 by han16nah