3dgeo-heidelberg/helios #283: Question on simulation of TLS panoramic scanners

simBauer, 2022-11-17:
Hello,

How is the support for TLS panoramic laserscanners (vertically rotating mirror with horizontal rotation, e.g. Leica RTC360, Trimble TX6)? The figure by Medic at [https://www.researchgate.net/figure/Perfect-panoramic-terrestrial-laser-scanner-geometry_fig1_316990821](https://www.researchgate.net/figure/Perfect-panoramic-terrestrial-laser-scanner-geometry_fig1_316990821) illustrates the principle.

I assumed that I can simulate this type of scanner with the rotating deflector type.

In an older issue #141 you already explained that the parameters `scanAngleMax_deg` and `scanAngleEffectiveMax_deg` correspond to rotating polygonal mirrors. So for the panoramic scanner I would set these to `scanAngleMax_deg = scanAngleEffectiveMax_deg = 360` (similar to the settings of the profile laserscanner RIEGL VQ-450). 
However, panoramic laserscanners have a limited field of view due to the mounting which results in an conical “empty space” underneath the scanner. So far I realized this by adjusting the parameters for the field of view (`verticalAngleMin_deg=-60` and `verticalAngleMax_deg= 240` for a field of view of 300°) and “cutting out” the cone underneath the scanner. 

Is there a more convenient way to set this property directly for the scanner?

Simone

lwiniwar, 2022-11-18:
Hi @simBauer ,

yes, for a laser scanner of the type you linked, the `scanAngleMax_deg` should be set to 360, and equal to `scanAngleEffectiveMax_deg` (to mimic a `rotating` polygon with 0 sides). During the time the mirror points 'downwards' (i.e., into the mounting), no other points can be recorded, so time continues running. So the way you implemented this with the "filter" critera of `verticalAngleMin_deg` and `verticalAngleMax_deg` are correct.

If you are not doing it that way anyway, note that you can set these for the `survey` once in a `scannerSettings` tag before the `legs`:

```xml
<document>
        <scannerSettings active="true" id="scanner1" verticalAngleMin_deg="-60" verticalAngleMax_deg="240" />
        ...
        <survey name="..." platform="..." scanner="..." scene="...">
		<leg>
			<platformSettings x="-30.0" y="-50.0" />
			<scannerSettings template="scanner1" />
		</leg>
                ...
	</survey>
</document>
```

So no - this is not possible to set in the scanner XML, but only in the survey XML (but you can set it once there for all legs). I hope this helps!

Cheers,
Lukas

simBauer, 2022-11-21:
Hello Lukas,
Sounds good. Thanks for your support.
Cheers,
Simone

closed 2022-11-21 by simBauer