3dgeo-heidelberg/helios #288: detailedvoxels transmittive mode vs scaled

kmik, 2022-11-19:
**The problem**
Doing some preliminary testing of helios for forest inventory related research. A simulation of TLS scan to detailedvoxels scene works in scaled mode and not in transmittive mode. 

**How to reproduce**
I voxelized a UAV LiDAR scan from a 30m x 30m forest plot with AMAPvox 1.6.4. The file structure is similar to that what is used in the example, please see a portion of the .vox file in the attachements.

I run helios:

> helios.exe data\surveys\test\test.xml --lasOutput

If you uncomment <<"param type="string" key="intersectionMode" value="scaled" >>" in the test_voxels.xml (test_voxels.txt in the attachements) file everything seems to work. Otherwise (I believe it is the "transmittive" mode when no param for either "scaled" or "fixed" is given) the simulation finishes but the point cloud is empty.

- survey XML: [test.txt](https://github.com/3dgeo-heidelberg/helios/files/10047033/test.txt)
- scene XML: [test_voxels.txt](https://github.com/3dgeo-heidelberg/helios/files/10047037/test_voxels.txt)
- Portion of the .vox file: [vox_file.txt](https://github.com/3dgeo-heidelberg/helios/files/10047046/vox_file.txt)
- full .vox file: [27_vox_1_6.zip](https://github.com/3dgeo-heidelberg/helios/files/10047091/27_vox_1_6.zip)
- log file "scaled" mode: [helios_2022-11-19_15-37-22.log](https://github.com/3dgeo-heidelberg/helios/files/10047106/helios_2022-11-19_15-37-22.log)
- log file "transmittive" mode:  [helios_2022-11-19_15-38-54.log](https://github.com/3dgeo-heidelberg/helios/files/10047111/helios_2022-11-19_15-38-54.log)
- resulting point clouds: [points.zip](https://github.com/3dgeo-heidelberg/helios/files/10047113/points.zip)

**Expected behavior**
Would expect "transmittive" and "scaled" modes both to produce output.

**Actual behavior**
Only "scaled" mode produces point cloud. "transmittive" mode finishes the simulation without errors, but the point cloud is empty. I must be missing something, but cannot figure out what it is. 

**Your environment (please complete the following information):**
 - OS: windows 10
 - HELIOS++/pyhelios version 1.1.0
 -  32GB RAM

han16nah, 2022-11-19:
Dear Mikko,
that's interesting, thank you very much for reporting. I can reproduce the error and cannot find out what exactly is wrong.
Your `.vox` file looks totally valid. Only "scaled" mode works, both "fixed" and "transmittive" result in an empty point cloud.

I've ruled out a few things by trying to
- remove NaNs
- translate the .vox file (changing min and max corner) as well as the scan position to local coordinates

But the issue persists.

I furthermore tested some of my own surveys using the transmittive mode to make sure that we did not introduce a general bug in the latest version(s) but those still work well.

Interestingly, there is even an empty point cloud produced if I add another scenepart with the `objloader` (in my tests a simple ground plane). The ground plane is not recorded which also speaks against the case that the rays simply go through all transmittive voxels with the respective parametrization.

We will investigate this further next week, thanks again for reporting.

han16nah, 2022-11-25:
Dear @kmik 
we investigated this issue and found the cause. 
As a quick workaround, can you use the "Skip empty voxel" option in AMAPVox (`Output` tab -> `Output format and precision`) and use the new .vox file in your survey?
Do you now get a non-empty point cloud when running your simulations also in the default `transmittive` mode?
[image]

han16nah, 2022-12-14:
@kmik We think this issue was solved with 96443f608b6f1cd99d1cc0a2580698b6f5f4da54 and will close it.
Let us know if you encounter further problems

closed 2022-12-14 by han16nah