3dgeo-heidelberg/helios #295: Extend Gallery

han16nah, 2022-12-14:
**The Problem**
At the moment, it is not straightforward to find easily find HELIOS++ demos using certain features. However, the easiest way to get into HELIOS++ (and any other software, honestly) is by looking at examples (cf. [matplotlib gallery](https://matplotlib.org/2.0.2/gallery.html)).

**The Solution/Idea**

We want to create a "Gallery" with demos of HELIOS++ simulations in Jupyter Notebooks.

*Preliminary work* 
In #294, such a gallery was already started.
- [ ] Create a "gallery" in the README. It's a markdown table with two columns. For each demo, there should be a thumbnail and a link to nbviewer.
- [ ] We already have some example notebooks. Here, only a link to nbviewer had to be created and added to the gallery table and a suitable thumbnail has to be created.
- [ ] Some examples do not (yet) exist as Jupyter Notebook. Here, a template is needed. This was created using the [arbaro demo](https://github.com/3dgeo-heidelberg/helios/blob/65a2819ee385fd88b12f7d6eb4eed772e660a5d4/example_notebooks/3_arbaro_TLS_demo.ipynb). The Notebook simply displays the XML files and contains some explanations in text but no (pyhelios) modifications are made. In the end, the survey is executed and the outputs displayed. For displaying XML files, [utility functions]((https://github.com/3dgeo-heidelberg/helios/blob/65a2819ee385fd88b12f7d6eb4eed772e660a5d4/pyhelios/util/xmldisplayer.py)) were written 

*Open Tasks*
- [ ] Using the [arbaro demo](https://github.com/3dgeo-heidelberg/helios/blob/65a2819ee385fd88b12f7d6eb4eed772e660a5d4/example_notebooks/3_arbaro_TLS_demo.ipynb) template, create further Notebooks for the following surveys and add them to the Gallery in the README file:
  - [data/surveys/demo/als_hd_demo_tiff.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/demo/als_hd_demo_tiff.xml)
  - [data/surveys/demo/mls_wheat_demo.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/demo/mls_wheat_demo.xml)
  - [data/surveys/demo/light_als_toyblocks_multiscanner.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/demo/light_als_toyblocks_multiscanner.xml)
  -  [data/surveys/toyblocks/mls_toyblocks.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/toyblocks/mls_toyblocks.xml)
  - [data/surveys/toyblocks/uls_toyblocks_survey_scene_combo.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/toyblocks/uls_toyblocks_survey_scene_combo.xml)
  - [data/surveys/toyblocks/livox_test.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/toyblocks/livox_test.xml)
  - [data/surveys/toyblocks/uls_toyblocks_livox.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/toyblocks/uls_toyblocks_livox.xml)
  - [data/surveys/voxels/als_detailedVoxels_mode_comparison.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/voxels/als_detailedVoxels_mode_comparison.xml) and [data/surveys/voxels/uls_detailedVoxels_mode_comparison.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/voxels/uls_detailedVoxels_mode_comparison.xml) (may be combined in one notebook)
  - one of [data/surveys/voxels/tls_sphere_xyzloader.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/voxels/tls_sphere_xyzloader.xml) and [data/surveys/voxels/tls_sphere_xyzloader_normals.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/voxels/tls_sphere_xyzloader_normals.xml)
  -  [data/surveys/toyblocks/als_toyblocks_stripid.xml](https://github.com/3dgeo-heidelberg/helios/blob/dev/data/surveys/toyblocks/als_toyblocks_stripid.xml) (potentially modify first)
- [ ] Create thumbnail for each demo: Some ideas:
  - use the Python logo when it's a proper pyhelios demo, other wise, just have the HELIOS++ Logo in there
  - show previews of the simulated point clouds in the thumbnails.
  - use some common themes, e.g., for the platforms. If it's airborne, have the same airplane in there, if it's TLS, display a tripod etc.
  - Once we have them all, we may think of an order/structure (e.g., by platform, by topic or by complexity) and suitable numbering.

lwiniwar, 2023-01-08:
When using the (non-interactive) nbviewer, outputs have to be saved in the `ipynb` files to be displayed (and this is arguably the main point of the notebooks). We should therefore ensure we actually _run_ the notebooks before committing the files to github.
Apparently, I failed to do so for at least https://nbviewer.org/github/3dgeo-heidelberg/helios/blob/dev/example_notebooks/1-getting-started.ipynb. 

@SinaZumst , please check all the notebooks if the outputs are included and if not, re-run them (a clean "restart kernel & run all" should be sufficient), then save & commit the changes. Thanks!

han16nah, 2023-01-09:
> When using the (non-interactive) nbviewer, outputs have to be saved in the `ipynb` files to be displayed (and this is arguably the main point of the notebooks). We should therefore ensure we [...]

Good point. Could've also been me, I've been used to strip outputs from notebooks when comitting, because it keeps them cleaner in the git history, but you're right: In this case, we want users to be able to see the outputs so the notebooks are informative.

SinaZumst, 2023-01-09:
Okay! I will check that. Would it be best to open a new branch for each notebook or can i use one branch for all?

SinaZumst, 2023-01-09:
I do have some problems with the table formatting: How can i achieve that all thumbnails have the same extent?
I just saw that the formating in the dev branch does not look good at all, it probably screwed up when the 5_hd_demo branch was merged.
I tried to fix it in the new 5_hd_demo branch but i don't think this is how it should look like.

han16nah, 2023-01-10:
> I do have some problems with the table formatting: How can i achieve that all thumbnails have the same extent? I just saw that the formating in the dev branch does not look good at all, it probably [...]

It looks okay for me. You could make sure that your PNG images also have the same size. To keep it like in the first examples, they should be 394 x 315 pixels.

SinaZumst, 2023-01-10:
> 
Okay i will adapt the size later on the 5_hd_demo branch

lwiniwar, 2023-01-19:
[ ] please also update the wiki here: https://github.com/3dgeo-heidelberg/helios/wiki/First-steps#learn-by-example

han16nah, 2023-01-26:
- [ ] We might also think about running the surveys with the -q parameter to avoid those very long logging outputs in the notebook. This can be done when re-running all notebooks once the gallery is final

closed 2023-02-10 by han16nah