3dgeo-heidelberg/helios #301: Accuracy of angles and distance

simBauer, 2023-01-04:
The suggestion is to model angular and distance-dependent accuracy.

- The angular accuracy could be implemented in the way you are currently modelling the distance accuracy. Did I understand it correctly from the Wiki that you are using the `accuracy_m` as standard deviation for adding random noise from a normal distribution to the measurements?
- Another feature might be distance-dependent accuracy. It is usually given in ppm (parts per million). For example, if an accuracy of 1 mm + 10 ppm is given in the datasheet, 10 mm/km = 10 $\cdot10^{-6}$ m/m proportional to the distance would be added to the constant 0.001 m `accuracy_m` for each point.

lwiniwar, 2023-01-08:
Thank you for this suggestion @simBauer !
1) yes, after a hit is detected and the corresponding range is calculated, the point's range coordinate is updated with an error value drawn from a normal distribution (0, `accuracy_m`) before being converted to cartesian coordinates. Note that the error is only applied in the _point cloud_ output, but not in the _fullwave_ output.

2) We can easily extend this to be distance-dependent. I suggest a new element `accuracy_distance_dependent_ppm` in the `ScannerSettings` with a default value of 0.

Then, we can update
https://github.com/3dgeo-heidelberg/helios/blob/0787b1ae44492d45b6bb11ad3428c9632c8cfd6e/src/scanner/detector/AbstractPulseRunnable.cpp#L72

to include a distance-dependent term.

Looking at 
https://github.com/3dgeo-heidelberg/helios/blob/0787b1ae44492d45b6bb11ad3428c9632c8cfd6e/src/scanner/Scanner.cpp#L376-L385 
it might be easier to change this to a standard (mean=0,std.dev.=1) distribution though and then multiply by the magnitude later, i.e.

`distance += rg.normalDistributionNext() * (getDetector()->cfg_device_accuracy_m + distance * getDetector()->cfg_device_accuracy_distance_dependent_ppm * 1e-6)` 

As we work with Standard deviations already, simple addition and multiplication should suffice (as opposed to working with variances), but feel free to verify this.

3) Angular uncertainties may be a bit more tricky in the context of HELIOS++'s handling of different scanner types. For instance, while a TLS has two angular quantities (yaw and elevation/zenith distance), an ALS only has one (scan angle). In the current implementation of `AbstractPulseRunnable`, we do not have a distinction for these different types (which are, in HELIOS, defined by the platform being static or dynamic, correct?). @albertoesmp do you have a suggestion on how we could include angular uncertainties, e.g. by defining `accuracy_scanangle_deg` and `accuracy_head_rotation_deg` or similar?

albertoesmp, 2023-01-09:
Hi @lwiniwar and @simBauer, and thanks for the interesting discussion/proposal.

First, I agree with Lukas's suggestion on distance-dependent accuracy. It should be easy to implement this by updating the `AbstractPulseRunnable::applyMeasurementError` method.

Regarding the angular uncertainties, at the moment, I can only think about attitude noise applied from platform parameters. More concretely, I'm thinking about the `Scanner::handleSimStepNoise` method (see snippet below).
```cpp
// Apply noise to beam attitude
if(!platformNoiseDisabled){
    if(platform->attitudeXNoiseSource != nullptr) { // Add attitude X noise
        absoluteBeamAttitude = absoluteBeamAttitude.applyTo(Rotation(
            Directions::right, platform->attitudeXNoiseSource->next()
        ));
    }
    if(platform->attitudeYNoiseSource != nullptr) { // Add attitude Y noise
        absoluteBeamAttitude = absoluteBeamAttitude.applyTo(Rotation(
            Directions::forward, platform->attitudeYNoiseSource->next()
        ));
    }
    if(platform->attitudeZNoiseSource != nullptr) {// Add attitude Z noise
        absoluteBeamAttitude = absoluteBeamAttitude.applyTo(Rotation(
            Directions::up, platform->attitudeZNoiseSource->next()
        ));
    }
}
```

If we need to extend this to add scanner-level angular uncertainty, we can add a 3D vector of angular uncertainties in ScannerSettings. Then, we can consider only the first component when there is only one angular quantity and the following components if there are more angular quantities.

lwiniwar, 2023-01-09:
Hi @albertoesmp, thank you for the suggestion for an implementation.

When you say `we can consider only the first component when there is only one angular quantity`, do you mean to consider only the first component in the case that there is only one setting in the `scannerSettings`? I.e., the decision for when to apply a second angular quantity stems from what is given in the XML, is that correct?

Cheers,
Lukas

albertoesmp, 2023-01-10:
Hi again, @lwiniwar , and thanks for the discussion

I propose implementing an XML attribute like `angularNoise="0.1, 0.2, 0.3"` (maybe we could find a better name). If the particular scanner has only one angular quantity (e.g., angle wrt horizontal plane), then 0.2 and 0.3 will be ignored, and 0.1 will be considered. If the scanner has no angular quantities, nothing is considered. If it has two angular quantities, 0.1 will be considered for the first one and 0.2 for the second one, but 0.3 will be ignored. By default, it is `angularNoise="0, 0, 0"` (no noise at all).

My point for this suggestion is that we have different scanner models (e.g., different deflection strategies). Thus, each case would require handling its angular noise sources. Each deflector can be seen as a function that computes the `emitterRelativeAttitude` at a given simulation step. Thus, each deflector will handle angular noise on its way (e.g., `PolygonMirroBeamDeflector` calculates the attitude from the last beam angle, `OscillatingMirrorBeamDeflector` works similarly but with accurate simulation of accelerate/decelerate stages, `RisleyBeamDeflector` simulates both horizontal and vertical field of views from two different rotation speeds).

Answering the question, "the decision for when to apply a second angular quantity stems from what is given in the XML, is that correct?" I'd say yes; the decision stems from what is given in the XML. If you specify a Scanner that requires only one angular quantity, it doesn't matter what you define as the second and third components of the angular uncertainties vector. The number of angles used by the model depends on the complete XML scanner specification.

Maybe I'm making this more complicated/exhaustive than needed. My proposal assumes that the noise applied to the platform's attitude is insufficient, and we must go deeper. My main concern is to avoid having one different set of angular noise parameters per deflection strategy.

simBauer, 2023-01-11:
Thank you for the great ideas. I actually do not know if the angular accuracy is applicable to all scanner types, expecially for ALS (I have no experience in that field). It is important for TLS rotating scanners. Sorry, I should have included that in the issue.

simBauer, 2023-01-19:
Hello @albertoesmp and @lwiniwar,
Thank you for the great ideas. I actually do not know if the angular accuracy is applicable to all scanner types, expecially for ALS (I have no experience in that field). It is important for TLS rotating mirror scanners. Sorry, I should have included that in the issue. Maybe it could be possible to allow the setting of angular accuracy only for the "rotating" scanner type?

My idea was rather to apply the noise directly to the polar coordinates in the scanner coordinate system (to the vertical and horizontal angle, for illustration see figure at https://www.researchgate.net/figure/a-Local-Cartesian-coordinate-system-of-the-scanner-with-a-respect-to-the-main_fig2_316990821) before converting to a ray direction vector. Did I get it right that including the angular noise in the platform noise would result in a 3D rotation of the platform coordinate system? I do not think that this will yield the same results.
Is it possible to apply noise directly to the angular polar coordinates? Couldn't it be included in the doSimStep() functions?
https://github.com/3dgeo-heidelberg/helios/blob/32c353befa8d3ef48a60c94ad8c46cd6ac22ba23/src/scanner/beamDeflector/PolygonMirrorBeamDeflector.cpp#L89-L102
https://github.com/3dgeo-heidelberg/helios/blob/32c353befa8d3ef48a60c94ad8c46cd6ac22ba23/src/scanner/ScannerHead.cpp#L21-L28
Best,
Simone

lwiniwar, 2023-01-19:
Hi @simBauer, @albertoesmp !

The code snippets you sent out refer to the increase of the rotation angle at each step. I do not think that the angular uncertainty refers to that, but rather to an uncertainty in the _measurement_ of said angle. This re-opens a discussion we had a while ago when talking about random platform (position) noise:

Do we want to 
(a) cast the ray without any errors, do the intersection, get the polar coordinates, and then apply a random noise to these coordinates in the exported point cloud, or do we 
(b) want to add the noise before casting the ray, subsequently casting it into a (slightly) different direction, but _record_ the original direction before the ray is cast.

(b) would be to add it at the location @simBauer pointed out, but it's also the one that comes with unwanted side-effects. Consider a scanner with just angular uncertainty (ranging uncertainty set to 0), facing an edge/corner (of a building, ...). The resulting distance measurement will vary by a large amount even with just a small angular uncertainty, especially when using an infinitesimal beam (just 1 subray). 

@albertoesmp, using `platformAttitudeNoise` would be like (b), but recording the updated direction, right? So it is more a "shaking of the scanner" resulting in a different point cloud than an actual measurement error.

Therefore, I advocate for (a). Carry out a "perfect" simulation, but add noise on the measured direction(s).

Having said that, I think implementing the `verticalAngleError` just for `rotating` optics and a `horizontalAngleError` for platforms with a `headRotation > 0` makes sense. This concept should be extendable to other scanners at some point (e.g., the conical scanners). 

Cheers,
Lukas

simBauer, 2023-01-23:
Hello @lwiniwar @albertoesmp,
You're right. Option (a) would model an error at the detection while (b) would rather refer to an error at the deflection. 
So, probably it would be easiest to leave the modelling of the random errors to the "processing" by the user? It would avoid the differentiation of scanner types. The workflow would be then: do a simulation of the perfect point cloud with Helios++, access the results with pyhelios, transform to scanner coordinate system, re-calculate polar coordinates in the required format, add noise at that level, convert to cartesian, transform to world coordinate system and save the adjusted point cloud / analyse / ... 
Best, 
Simone

lwiniwar, 2023-01-23:
Hi @simBauer,
yes, the `pyhelios` concept allows you to do exactly that. You can 
1. access the point cloud within `pyhelios` (or load it from XYZ or LAS if you prefer that), 
2. convert it to polar coordinates (note that you need to subtract the scanner location first)
3. add random, uncorrelated noise in the three dimension
4. convert it back to cartesian and finally
5. write the output

For the conversion, you can use the helper functions `pyhelios.cartesianToSpherical` and `pyhelios.sphericalToCartesian` as shown here:

https://github.com/3dgeo-heidelberg/helios/blob/e2e04c0d3cd77f99492ec8bec5ae6aafe61af4dc/example_scripts/pysimulation_multiscanner_demo.py#L119-L120

The random noise can be added e.g. using numpy:

```
import numpy as np

cart = pyhelios.cartesianToSpherical(xyz - scanPos)

noise_range = np.random.normal(0, sigma_dist, shape=(spher.shape[0], 1))
noise_elev = np.random.normal(0, sigma_elev, shape=(spher.shape[0], 1))
noise_azim = np.random.normal(0, sigma_azim, shape=(spher.shape[0], 1))

spher[:, 0] += noise_range
spher[:, 1] += noise_elev
spher[:, 2] += noise_azim

xyz_new = pyhelios.sphericalToCartesian(spher) + scanPos
```

Note that the angles are in radians, so you need to convert your standard deviations to radians as well. Also, `np.random.normal` takes *standard deviations*, not *variances* as input.

Cheers,
Lukas

albertoesmp, 2023-01-27:
Hello again, @lwiniwar  , @simBauer 

Sorry for taking so much time to address this, but I wanted to extend the code with some tools to facilitate testing and defining univariate expressions. Thus, I implemented a univariate expression tree in HELIOS++ that is fully integrated since this commit:  https://github.com/3dgeo-heidelberg/helios/commit/d1fd77095f59c69d47fb7fad0dcc90d895ae073c

Consequently, it is now possible to define univariate models for HELIOS++ directly from the XML files. The first use case we have is related to the distance error. Please, let me define the device accuracy as $d_{\mathrm{acc}}$, such that $\mathcal{N}$ is a value taken from our normal distribution $\mathcal{N}(\mu=0, \sigma=d_{\mathrm{acc}})$. Until now, a distance with error $y$ was calculated for any given distance $x$ as shown below:

$$
y = x + \mathcal{N}
$$

We can see this as a particular case of the equation below where $f(x)=1$.

$$
y = x + \mathcal{N} f(x)
$$

We can now define this $f(x)$ through an XML attribute `distanceMeasurementError` in the `<detectorSettings>` element using a string formula. The string format will be commented on later. For now, let us say we want to ignore the device's accuracy with our expression. Thus, we could simply divide by the device accuracy to have a normal distribution with mean $\mu=0$ and standard deviation $\sigma=1$, such that:

$$
y = x + \mathcal{N} \frac{1}{d_{acc}}
$$

I'm going to assume we are using the current `leica_als50-ii` scanner specification, which has a device accuracy $d_{\mathrm{acc}} = 0.05$. Thus, in the XML, I would specify `distanceMeasurementError="1/0.05"` or even better ``distanceMeasurementError="20"``. The simpler the expression, the fewer the computations (the expression tree will not be automatically optimized, at least not now).

If we want to use distance-dependent error expressed in parts per million, we could simply write an expression following the previous proposal by @lwiniwar . 

$$
y = x + \mathcal{N} \left({
    d_{\mathrm{acc}} + x d_{\mathrm{ppm}} 10^{-6}
}\right)
$$ 

Now, we divide $\mathcal{N}$ by $d_{\mathrm{acc}}$ because we want to work with a standard normal distribution. Thus, the expression becomes:

$$
y = x + \mathcal{N} \left({
    1 + x \frac{d_{\mathrm{ppm}}}{d_{\mathrm{acc}}} 10^{-6}
}\right)
$$

Let us say we want $d_{\mathrm{ppm}}=30$. But then, $d_{\mathrm{acc}}^{-1} d_{\mathrm{ppm}} 10^{-6} = 0.0006$. Consequently, we should specify `distanceMeasurementError="1 + t*0.0006"`. We could always use `distanceMeasurementError="1 + t * 30/0.05 * 10^(-6)"`. It will work, but it will compute more operations. Remember, the simpler, the better.

Here we can see a histogram-based comparison between the behavior of HELIOS++ when `distanceMeasurementError` is not specified (case A) versus when `distanceMeasurementError="20"` (case B):
[image]

Below is a histogram-based comparison between the behavior of HELIOS++ when `distanceMeasurementError` is not specified (case A) versus when `distanceMeasurementError="1 + t*0.0006"` (case B):
[image]

For both previous plots, the error measurement always assumes $x=1000$.

In a more general way, it is now possible to do XML specifications for HELIOS++ using univariate expressions. The following symbols are available with their standard meaning. You can use them to build any expression you need and give that as a string to HELIOS++.

Variable: `t`
Numbers: All numbers can be given either as absolute or negative numbers.
Named numbers: `pi`, `e`
Operators: `+`, `-`, `*`, `/`, `^`. `(`, `)`
Functions: `exp`, `ln`, `sqrt`, `abs`, `cos`, `sin`, `tan`, `acos`, `asin`, `atan`, `atan2`, `cosh`, `sinh`, `tanh`

I hope this helps to make HELIOS++ more flexible and allows us to test more things and support different behaviors without rebuilding the software.

@simBauer , @lwiniwar , do you still interested in modifying the behavior of `PolygonMirrorBeamDeflector::doSimStep()`? If so, we could discuss what expression we want to update and connect it through a univariate expression tree as done with the distance-dependent error.

lwiniwar, 2023-01-27:
Hi @albertoesmp and thank you for the implementation of this great feature! I'm sure we will be able to make more usecases for it down the (development) road!

A few comments:
1) I do not understand why you multiply the randomly drawn variables with `1/d_acc`. The value the user provides should be the standard deviation, therefore a larger value corresponds to a larger spread -- and a variable drawn at random from $\mathcal{N}\left(0, 1\right)$ multiplied by $\sigma$ should give the desired results?

i.e., say $X \sim \mathcal{N}(0, \sigma^2)$ and $Z = X / \sigma$, then $Z \sim \mathcal{N} (0, 1)$.

This also relates to values provided by the user being attached with a unit, i.e., $\sigma=3 mm$. So if the user sets `distanceMeasurementError="0.05"`, I expect to see a standard deviation in the measurements of 0.05, i.e., 

$d = d_0 + X$ where $X\sim \mathcal{N}(0, 0.05^2)$. 

As shown above, this can be subsitituted by $d= d_0 + \sigma \cdot Z$ where $Z \sim \mathcal{N}(0, 1) $.

Now, the same should be true for the univariate expressions, in the general case $f(\vec x )$:

$d= d_0 + f(\vec x) \cdot Z$

and in our specific case

$d= d_0 + \left( \sigma_{static} + d_0 \cdot \sigma_{ppm} \cdot 1e-6 \right) \cdot Z$,

with numbers

$d= d_0 + \left( 0.05 + d_0 \cdot 2.0 \cdot 1e-6 \right) \cdot Z$,

where I would write `distanceMeasurementError="0.05+t*2e-6"` ? This is what I would expect from a user perspective.

2) Is there a specific reason for the _distance_ parameter to be called `t`? Could we instead call it `d` as in distance? What other variables would we have access to at that point in the code - how about the local incidence angle (`lia`)? The target reflectance value `r`?

3) Do you have an estimate on the runtime cost of evaluating these functions on-the-fly? Is it a significant slowdown in HELIOS++?

Cheers,
Lukas

PS: I am assuming we are _replacing_ the current parameter `accuracy_m`. We can of course also just call the `distanceMeasurementError` `accuracy_m` instead and leave it at $f = const.$ for scanners without a distance-dependant error.

albertoesmp, 2023-01-27:
Hi @lukas,

Find my answers below (your 3. is answered in my 1.)

1. I think the confusion here comes from the assumption that we are replacing `accuracy_m`. It is not replaced. The `distanceMeasurementError` attribute is an optional one that we could easily not specify. If the user simply wants $X \sim \mathcal{N}(0, \sigma^2)$, then the classical accuracy_m=" $\sigma$ "  is the way to go. The standard deviation is always seen as defined by the `accuracy_m` parameter, and the expression scales the normal distribution defined by the `accuracy_m`. I did some benchmarking using the ALS TIFF demo, and I didn't find any noticeable performance drop on my PC (10 runs). Nevertheless, the main reason to preserve `accuracy_m` is to prevent making HELIOS++ dependent on expression trees. It is a new feature that might not be well-suited for high-performance simulations depending on the component. Thus, to be on the safe side, I think we must adhere to the red line of HELIOS++ working without any expression tree and implementing them as alternatives over a base case. If you think it would be better to automatically replace $X \sim N(0, \sigma^2)$ by $Z = X/\sigma$ when using the expression method, just let me know, and I'll update it. 

2. Yes, the reason is that I implemented a factory to build expression trees from strings. This factory parses the given string assuming certain substrings have a special meaning (e.g., "t" for the variable, "+" for addition, "pi" for the number pi, "cos" for the cosine function). Internally, the variable must be referred to by a special symbol. One could think that replacing any fixed expression with 't' would do the trick, but such a naive approach might cause problems when there are common patterns between substrings. Doing a specific parsing method for each expression to have an adequate variable symbol will make extending HELIOS++ with further expression trees more complicated and time-consuming. I think it is not a big problem because it will be used mostly like `someName="(t-1)^2"` where the `someName` will already speak about the meaning of the variable. At the moment, there is only one expression tree. But connecting new variables should be straightforward now that we have the univariate expression tree implementation.

To explain a little bit better what is going on, we now have two methods in the `AbstractPulseRunnable` class. First, the `applyMeasurementErrorDirectly` method is the classical/legacy way of computing the distance error. It will be called as usual when no `distanceMeasurementError` is given (maybe we should discuss a convenience like naming *_f any function-like attribute?). 
https://github.com/3dgeo-heidelberg/helios/blob/4be3f735cabc2d03db52e43bf769fc0d002813f4/src/scanner/detector/AbstractPulseRunnable.cpp#L93-L100

Second, the `applyMeasurementErrorFromExpr` method will be called if `distanceMeasurementError` is given.
https://github.com/3dgeo-heidelberg/helios/blob/4be3f735cabc2d03db52e43bf769fc0d002813f4/src/scanner/detector/AbstractPulseRunnable.cpp#L101-L109

One final point regarding performance. It is reasonable that some variables could be modeled using this method without a significant slowdown. However, we should not infer that this will be the case for any variable. It depends on where the variable is computed (sequential region, parallel ray-tracing, inside how many loop levels, etc.).

lwiniwar, 2023-01-27:
Hi @albertoesmp ,
thank you for the comments! This clears a lot of things up for me.

1) I think the current implementation (i.e., having to "undo" the constant error) is not user-friendly. I do get your point about "breaking things" so, I'm wondering if we could analyse the value given as `accuracy_m` and if it is numeric, use the default implementation, and if not, we try to parse it as an univariate expression. If that fails, we throw an error. I assume C++ has a proper method of checking if a string can be fully parsed as a numeric value or not.

2) I see now also why they are called _univariate_ expressions. As in my formulas above, I thought it would also be plausible to include other variables in the expression (as the reflectance and/or the incidence angle). I see that this is not possible in the current implementation. Do you think it could be feasible to include other variables - one that are clearly defined by us as developers, where we can make sure no collisions with pre-defined functions like `cos`, `sin`, etc. occur?

Cheers,
Lukas

albertoesmp, 2023-05-24:
Everything is okay from my side too. Feel free to close the issue.

closed 2023-05-26 by simBauer