3dgeo-heidelberg/helios #302: Simulation of panoramic scanners

simBauer, 2023-01-04:
Hello,

The issue concerns the simulation of TLS panoramic scanners. Examples for this scanner type would be the Leica RTC360 or the Zoller+Fröhlich Imager 5016. 

### Description of the problem
I wanted to realize a panoramic scanner with the rotating deflector type and generate a specific resolution in the scan. 
So far, a panoramic field of view of e.g. 300° can be simulated with the scanner parameters `scanAngleMax_deg`= `scanAngleEffectiveMax_deg`= 360° and the survey parameters `verticalAngleMin_deg`=-60° and `verticalAngleMax_deg`=240° (#283).

However, the setting of `scanAngleMax_deg`=360° seems not consistent with the usage of the scan angle as half of the swath. I use the formulas in the [Wiki](https://github.com/3dgeo-heidelberg/helios/wiki/Scanners#defining-horizontal-and-vertical-angular-resolution) for calculating the scan parameters `scanFreq_hz` and `headRotatePerSec_deg` for a defined resolution and `pulseFreq_hz`. According to the formulas, the scan angle value needs to be doubled, so I am actually calculating parameters for a "720° vertical rotation". I also tried to avoid that by using `scanAngleMax_deg`= `scanAngleEffectiveMax_deg`=180°. This results in an incomplete view and a cut-off pointcloud. If I calculate values for a 360° vertical rotation the scan pattern does not match the resolution. 

### Ideas for solution
The following suggestions might improve convenience for the simulation of TLS panoramic scanners.

#### Survey parameters: Resolution and scan time

Since panoramic scanners are usually operated by choosing a resolution (e.g. 3mm @ 10m), it could be used as the central survey parameter. With the corresponding scan time $t_{scan}$ for a complete panorama all relevant parameters for the simulation could be derived using similar formulas as documented in the [Wiki](https://github.com/3dgeo-heidelberg/helios/wiki/Scanners#defining-horizontal-and-vertical-angular-resolution).

- Angular resolution from $r_{res}$ [m] @ $d_{res}$ [m]:
     $\alpha_{res} = \arctan(r_{res}/d_{res})$

- Horizontal rotational speed $\omega_{head}$ [°/s] (`headRotatePerSec_deg`):
  For panoramic scanners a head rotation of 180° results in a full scan. 
     $\omega_{head} = 180°/t_{scan}$
    
- Frequency of the vertical rotation $f_{scan}$ [Hz] (`scanFreq_hz`):
    $f_{scan} = \omega_{head} / \alpha_{res}$  
    
- Pulse frequency (`pulseFreq_hz`):
    The pulse frequency would be used as measurement frequency $f_{meas}$ [Hz] (the frequency at which points are recorded). This simplification should allow to ignore differing distance measurement principles and quality level settings for panoramic scanners. 
    $f_{meas} = f_{scan} \cdot 360° / \alpha_{res}$
    

If time values aren’t available, they could be replaced by setting a realistic vertical rotation frequency (`scanFreq_hz`) or measurement frequency (`pulseFreq_hz`).

#### Scanner parameter: Vertical field of view

Another convienent feature might be the setting of the vertical field of view as a scanner parameter. The field of view is limited by the mounting of the scanner and results in an empty conical area underneath the scanner. 

Looking forward to hearing from you.
Best,
Simone

lwiniwar, 2023-01-08:
Hi @simBauer and thank you for this input!

Quick question: is it common to set the vertical resolution via spacing and range (e.g., 3mm at 10m) and the horizontal one using a completely different measure (i.e., time for 360 deg scan)? From our RIEGL TLS scanners we usually set a vertical and horizontal resolution directly (i.e., 0.017 deg spacing in horizontal direction and 0.03 deg in vertical; or alternatively as a spacing and range, for example).

Following up, if you wanted to use the panoramic scanner, but only scan half of the panorama (i.e., horizontal field of view of 180 deg), and you set the time to, say, 60 seconds, would that be the time for the full panorama, or for the 180 degrees?

Cheers,
Lukas

simBauer, 2023-01-09:
Hello @lwiniwar,
I think there was a misunderstanding. The user sets a resolution that applies to both horziontal and vertical. The scan time is then given for the specified settings (some scanners allow an additional quality setting). My idea was to use the scan time for calculating the parameters `headRotatePerSec_deg`, `scanFreq_hz` and `pulseFreq_hz` in case one has no values for those. 
The time corresponds to a full 360° scan. However, this only requires a head rotation of 180° with the panoramic scanner. (For explanation and detail about 2-face-scans, see https://www.youtube.com/watch?v=1HZH06bZk5A&list=PL1yg2aKW1HSO0mGrh1WP8Jsn4LoQG-CyE, starting at 8:49)
I hope this was clearer.
Have you already thought about how the panoramic scanner could be included in Helios++? Do you think it is possible to integrate it into the rotating deflector? 
Best,
Simone

albertoesmp, 2023-01-09:
I wonder whether to implement a `TwoFacePanoramicScanner` (or similar) class or suggest the `MultiScanner` implementation from the next release. Can't we model a two-face scanner as a multi-scanner of two 180º scanning devices operating simultaneously at the same frequency in opposite directions? What do you think, @lwiniwar?

lwiniwar, 2023-01-09:
Hi @simBauer,
thanks for the clarification.

1) I see what you mean now. So the real scanner provides you with the scan time for a given vertical and horizontal resolution [and potentially a quality setting], and you would like to use that as input for the simulation. I see two issues with that:
a) the scan time will be rounded to the nearest second, and when going back to `pulseFreq_Hz`, the value will not be integers. However, the pulseFrequency needs to be an integer in the current implementation. Rounding will lead to some - potentially unwanted - effects and errors.
b) we want users to be able to use the simulator without having access to the real scanner. While it is easy for you to set e.g. 0.017 deg for both resolutions and grab the scan time from the display, there is no way for me to do so, meaning I can just guesstimate and have no way of knowing if the scanner is actually capable of doing that.

On the other hand, the `pulseFreq_hz` is a sensor property, which can be found in the datasheet (e.g. https://media.faro.com/-/media/Project/FARO/FARO/FARO/Resources/2_TECH-SHEET/TechSheet_Focus_Laser_Scanner/Focus-Core/5976_TechSheet_FocusCore_AEC_ENG.pdf?rev=0c5a385287f8420e9c97157fdc5e51e4 - here it says 'up to 0.5MPts/s', so 500,000 Hz). This might be limited by the vertical FoV (see previous discussions like #141 ) but once set, will stay the same for the scanner; at least when keeping the same quality setting. Converting from points per second to actual pulse frequency is a simple matter of ratios, i.e., if of 360 deg points are only recorded in 300 deg, this means we need to consider more points: 500,000 * 360/300 = 600,000 Hz. Just from the datasheet, I cannot tell if we need to consider this or if they have already converted it, but looking at real scan's point count should give you that information.

For Z+F, the datasheet (e.g. https://zf-usa.com/wp-content/uploads/2021/06/ZF-IMAGER-5016_Datasheet-E_compr.pdf) gives a "data acquisition rate" in Pixels per second, which should be the same.

With the `pulseFreq_hz` given from the datasheet, and a _vertical_ resolution given from the user, the `scanFreq_Hz` can be calculated. This is what I propose we include in HELIOS++ in this update.
With a given _horizontal_ resolution, the `headRotatePerSec_deg` can be calculated. I also propose we include this in HELIOS++.
@albertoesmp these should be settings in the survey `leg`'s `platformSettings`.

To summarize: Instead of `scanFreq_Hz` and  `headRotatePerSec_deg`, the user provides the more intuitive `verticalResolution_deg` and `horizontalResolution_deg`. When implementing a new scanner, you have to look up or calculate the `pulseFreq_Hz` once, but this will stay the same for subsequent simulations after that.

2) Simulation of two-face scans is no issue, I thought #283 solved that issue? I am unsure why we would need a separate class for that. I might also have been mistaken in that issue that you cannot set the `verticalAngleMin` and `Max` in the scanner settings, that should actually work. Here's an example with a _theoretical_ panoramic scanner with the following properties:

| Parameter                  | Value                 |
|----------------------------|-----------------------|
| Vertical effective FoV               | 300 deg (-60 to +240 from horizon) |
| Vertical theoretical FoV               | 360 deg |
| Effective measurement rate | 500,000 pts/sec       |
| Pulse repetition rate      | 600,000 pts/sec       |

Then, I want a scan with angular resolutions of `0.2 deg` in both horizontal and vertical direction. **For now**, I need to calculate scan speed and rotation speed manually, giving me:

> 0.2 deg vertical over 300 deg = 1500 points, multiplied by 360deg/300deg gives a total of 1800 theoretical points per line
> with a prr of 600 kHz, this means one line takes 0.003 seconds
> inverting gives 333.33333 lines per second, which is the `scanFreq_Hz`

and

> 0.2 deg horizontal means that after 1800 points I have to have rotated by 0.2 deg
> so 0.2 deg per 0.003 s = 66.6666667 deg/s `headRotatePerSecond_deg`

**--> these calculations would be solved by implementing (1) **

I can then write a demo scene and survey:

In the first leg, we do a 360deg two-face scan, which is effectively a rotation about 180 deg. In the second leg, we do the full 360 deg rotation, resulting in twice the point density.

```xml
<?xml version="1.0" encoding="UTF-8"?>
<document>
    <survey name="box_tls" scene="data/scenes/box_scene.xml#box_scene" platform="data/platforms.xml#tripod" scanner="data/scanners_tls.xml#panoscanner">
		<!-- platform: tripod, deflector: rotating -->
		<leg>
            <platformSettings  x="0" y="0" z="-1.7" onGround="false"/>       
			<scannerSettings  active="true" pulseFreq_hz="600000" scanFreq_hz="333.333333" headRotatePerSec_deg="66.666667" headRotateStart_deg="0.0" headRotateStop_deg="180.0"/>
        </leg>
		<leg>
            <platformSettings   x="0" y="0" z="-1.7" onGround="false"/>
			<scannerSettings  active="true" pulseFreq_hz="600000" scanFreq_hz="333.333333" headRotatePerSec_deg="66.666667" headRotateStart_deg="0.0" headRotateStop_deg="360.0"/>
        </leg>
    </survey>
</document>
```

The scene loads the toyblocks cube, which is placed somewhat annoyingly, so we translate it to be centered around (0,0,0).

```xml
<?xml version="1.0" encoding="UTF-8"?>
<document>
    <scene id="box_scene" name="box_scene">
        <part>
[... cut ...] nding, with a rotation about 360 degrees, you get a double scan.

I looked at the results in CloudCompare. The green ball is the Scanner.

First leg, colored and animated by GPStime:
[image]

Second leg, colored and animated by GPStime:
[image]

And for the angular resolution:
Vertical
[image]

Horizontal
[image]

So if we implement (1), i.e., the conversion from angular resolution to scan frequency and pulse repetition rate, I would think this solves your issues. Is that correct?

Cheers,
Lukas

han16nah, 2023-01-10:
Nice explanation Lukas!
I agree that for (1), implementing `verticalResolution_deg` and `horizontalResolution_deg` is the way to go.

PS: 
In case that helps for (1), here is a small Python tool as well as an Excel tool for calculating the rotation speed/scan frequency from horizontal/vertical resolution. This is what we sometimes worked with up until now.
[tls_scan_settings.zip](https://github.com/3dgeo-heidelberg/helios/files/10380597/tls_scan_settings.zip)
[rot_speed_from_spacing.xlsx](https://github.com/3dgeo-heidelberg/helios/files/10380602/rot_speed_from_spacing.xlsx)

lwiniwar, 2023-01-10:
Hi all,
okay, I thought there was a better way to do that. In fact, there is:

- We can rotate the default beam attitude from pointing horizontally to pointing to zenith by supplying a rotation in the `scanner` XML:
```xml
	<scanner 	id 							= "panoscanner"
				name              			= "PANO"
				accuracy_m					= "0.005"
			 	beamDivergence_rad 			= "0.0003"
				headRotatePerSecMax_deg		= "500"
				
				optics            			= "rotating"
				pulseFreqs_Hz				= "100000,300000"
				pulseLength_ns     			= "5"
				rangeMin_m					= "0.01"
				scanAngleMax_deg   			= "180"
				scanAngleEffectiveMax_deg   = "150"
				scanFreqMin_Hz				= "0"
				scanFreqMax_Hz				= "12000"
				verticalAngleMin_deg		= "-60"
				verticalAngleMax_deg		= "240"
				>
		
		<FWFSettings beamSampleQuality="3"/>	
		<beamOrigin x="0" y="0" z="0.2">
			<rot axis="y" angle_deg="0" />
			<rot axis="z" angle_deg="0" />
			<rot axis="x" angle_deg="90" />
		</beamOrigin>
		<headRotateAxis x="0" y="0" z="1"/>
		
	</scanner>
```

Then, the FoV is symmetric with respect to the origin, (i.e., -150 deg to + 150 deg) and the simulation can be carried out without setting `verticalAngleMin/Max_deg`:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<document>
    <survey name="box_tls" scene="data/scenes/box_scene.xml#box_scene" platform="data/platforms.xml#tripod" scanner="data/scanners_tls.xml#panoscanner">
		<!-- platform: tripod, deflector: rotating -->
		<leg>
            <platformSettings  x="0" y="0" z="-1.7" onGround="false"/>       
			<scannerSettings  active="true" pulseFreq_hz="600000" scanFreq_hz="333.333333" headRotatePerSec_deg="66.666667" headRotateStart_deg="0.0" headRotateStop_deg="180.0"/>
        </leg>
		<leg>
            <platformSettings   x="0" y="0" z="-1.7" onGround="false"/>
			<scannerSettings  active="true" pulseFreq_hz="600000" scanFreq_hz="333.333333" headRotatePerSec_deg="66.666667" headRotateStart_deg="0.0" headRotateStop_deg="360.0"/>
        </leg>
    </survey>
</document>
```

This gives me the desired point clouds with the gap at the bottom and the resolution you asked for. Maybe this is a better solution than to further complicate the `lastPulseLeftDevice` function. With appropriate choices of the beam origin, I see no need for adaptation.

So this leaves (1) to be implemented. Sorry for not thinking of that earlier.

albertoesmp, 2023-01-11:
Hi again, @simBauer @lwiniwar and @han16nah 

I committed an update to support this new parametrization; find it here: https://github.com/3dgeo-heidelberg/helios/commit/4a4c068cf7a575810100ba7641ae05515fe0c4be

The associated PR is already closed; see: https://github.com/3dgeo-heidelberg/helios/pull/305

Thus, anyone working with the dev branch should be able to use the new XML survey specification. Following is an example based on @lwiniwar  suggestion. First, the classical specification.
```xml
        <!-- Specification using pulseFreq_hz and headRotatePerSec_deg -->
	<leg>
            <platformSettings  x="0" y="0" z="-1.7" onGround="false"/>       
            <scannerSettings  active="true" pulseFreq_hz="600000" scanFreq_hz="333.333333" headRotatePerSec_deg="66.666667" headRotateStart_deg="0.0" headRotateStop_deg="180.0" verticalAngleMin_deg="-60" verticalAngleMax_deg="240"/>
       </leg>
       <leg>
            <platformSettings   x="0" y="0" z="-1.7" onGround="false"/>
            <scannerSettings  active="true" pulseFreq_hz="600000" scanFreq_hz="333.333333" headRotatePerSec_deg="66.666667" headRotateStart_deg="0.0" headRotateStop_deg="360.0" verticalAngleMin_deg="-60" verticalAngleMax_deg="240"/>
       </leg>
```

Which should be equivalent to the new specification:
```xml
        <leg>
            <platformSettings  x="0" y="0" z="-1.7" onGround="false"/>       
            <scannerSettings  active="true" pulseFreq_hz="600000" verticalResolution_deg="0.2" horizontalResolution_deg="0.2" headRotateStart_deg="0.0" headRotateStop_deg="180.0" verticalAngleMin_deg="-60" verticalAngleMax_deg="240"/>
        </leg>
        <leg>
            <platformSettings   x="0" y="0" z="-1.7" onGround="false"/>
            <scannerSettings  active="true" pulseFreq_hz="600000" verticalResolution_deg="0.4" horizontalResolution_deg="0.2" headRotateStart_deg="0.0" headRotateStop_deg="360.0" verticalAngleMin_deg="-60" verticalAngleMax_deg="240"/>
        </leg>
```

Note that I updated the scanner definition to be as shown below. This update was done because the XSD we are using to validate the XML defines that `verticalAngleMin_deg` and `verticalAngleMax_deg` must be specified inside `<scannerSettings>` and not inside `<scanner>`. I detected this thanks to the new XML pytests added by @han16nah  (https://github.com/3dgeo-heidelberg/helios/blob/dev/pytests/test_demo_xmls.py)
```xml
	<!-- ##### BEGIN Panoramic scanner ##### -->	
	<scanner 	id 							= "panoscanner"
				name              			= "PANO"
				accuracy_m					= "0.005"
			 	beamDivergence_rad 			= "0.0003"
				headRotatePerSecMax_deg		= "500"
				optics            			= "rotating"
				pulseFreqs_Hz				= "100000,300000"
				pulseLength_ns     			= "5"
				rangeMin_m					= "0.01"
				scanAngleMax_deg   			= "180"
				scanAngleEffectiveMax_deg   = "150"
				scanFreqMin_Hz				= "0"
				scanFreqMax_Hz				= "12000"
	>
		<FWFSettings beamSampleQuality="3"/>	
		<beamOrigin x="0" y="0" z="0.2">
			<rot axis="y" angle_deg="0" />
			<rot axis="z" angle_deg="0" />
			<rot axis="x" angle_deg="90" />
		</beamOrigin>
		<headRotateAxis x="0" y="0" z="1"/>
	</scanner>
	<!-- ##### END Panoramic scanner ##### -->
```

The corresponding [survey](https://github.com/3dgeo-heidelberg/helios/blob/4a4c068cf7a575810100ba7641ae05515fe0c4be/data/surveys/toyblocks/panorama_tls_toybox.xml) and [scene](https://github.com/3dgeo-heidelberg/helios/blob/4a4c068cf7a575810100ba7641ae05515fe0c4be/data/scenes/toyblocks/toybox_scene.xml) files have been uploaded. The [scanners_tls](https://github.com/3dgeo-heidelberg/helios/blob/4a4c068cf7a575810100ba7641ae05515fe0c4be/data/scanners_tls.xml) XML file has been updated too.

Apart from the typical boilerplate from parsing and verification, the core of the update lies in the `ScannerSettings::fitToResolution()` method reproduced below:
```cpp
    void fitToResolution(){
        double const scanAngle_rad = (this->scanAngle_rad != 0.0) ?
            this->scanAngle_rad : headRotateStop_rad-headRotateStart_rad;
        scanFreq_Hz = (verticalResolution_rad * pulseFreq_Hz) /
            (2*scanAngle_rad);
        headRotatePerSec_rad = horizontalResolution_rad * scanFreq_Hz;
    }
```

First, the `scanAngle_rad` attribute is considered. If it is zero, the scan angle $\alpha$ will be assumed to be defined by the difference between the start and stop frontiers of the scanner head rotation. But then, the scanning frequency $f_s$ can be estimated from given pulse frequency $f_p$ and vertical resolution $V_{\mathrm{res}}$ as follows:

$$
f_s = \frac{V_{\mathrm{res}} f_p}{2 \alpha}
$$

Analogously, the head rotation per second can be derived from the given horizontal resolution $H_{\mathrm{res}}$ such that:

$$
H_{\mathrm{rps}} = H_{\mathrm{res}} f_s
$$

Please, let me know if the solution is not working as expected or if further updates are required.

lwiniwar, 2023-01-11:
Thank you, @albertoesmp . I have updated the Wiki pages for Scanner and Survey to reflect these changes.

lwiniwar, 2023-01-11:
Hi @albertoesmp ,

I think we still need to fix some things with the implementation. 
When I adapt the `tls_arbaro_demo` survey, to a target `verticalResolution_deg`=`horizontalResolution_deg`=`0.2`, and then measure in the point cloud, the vertical resolution is somewhere at 0.24 deg instead of 0.2. The horizontal resolution is 0.2 as requested.

Here's the output:
```
Scanner settings have been fitted from verticalResolution_rad = 0.00349066, horizontalResolution_rad = 0.00349066,
        pulseFreq_Hz = 100000, scanAngle_rad = 1.74533, headRotationLength_rad = 2.18166
Consequently, the old scanFreq_Hz = 0, and headRotatePerSec_rad = 0 have been updated to
        the new scanFreqHz = 100, and headRotatePerSec_rad = 0.349066
```
According to the MS Excel sheet, the solution should be scanFreq_Hz=84 and headRotatePerSec_rad=0.293215, then the target resolution is also correct.

I think the main reason for this is again the discrepancy between _effective measurement rate_ and the _pulse repetition rate_. The factor to covert between them can be calculated as `scanAngleMax_deg` \ `scanAngleEffectiveMax_deg`.

simBauer, 2023-03-03:
Hello @albertoesmp @han16nah @lwiniwar
I just tried the resolution parameters and the other settings you proposed for the panoramic scanner. It works fine for me. Thanks for the implementation!
Best,
Simone

closed 2023-03-03 by simBauer