3dgeo-heidelberg/helios #312: Definition scanAngleMax_deg and scanAngleEffectiveMax_deg

zoeschindler, 2023-01-18:
Hello, I just started with HELIOS++ and I want to add the RIEGL VZ-400i to the TLS scanner file. So I just copied the data of the RIEGL VZ-400 and started modifying. But there, I stumbled across the "scanAngleMax_deg" and the "scanAngleEffectiveMax_deg". I don't really understand what they both mean and where the data is from? I thought the  "scanAngleMax_deg" might be the vertical field of view of the scanner, but I checked the datasheets of several TLS RIEGL scanners and could not find these numbers there. For example, the RIEGL VZ-400 has an opening angle of 100 degree, not 120 degree, from +60 to -40. For the "scanAngleEffectiveMax_deg", I have no idea what it is describing. Can someone help me understand where to get these parameters from? Thanks!

bhoefle-3dgeo, 2023-01-18:
Have you checked the respective Wiki page in particular the figures? https://github.com/3dgeo-heidelberg/helios/wiki/Scanners

If this is not helping, we can give more hints and extend the Wiki page accordingly. 

Feel also free to contact us directly via email ([helios@uni-heidelberg.de](mailto:helios@uni-heidelberg.de)) in case you need a more advanced support for your project.

han16nah, 2023-01-18:
Hi @zoeschindler 

Thank you very much for the good question! 🙂 

The `scanAngleEffectiveMax_deg` is often needed to properly simulate rotating mirror deflectors like the RIEGL VZ-400/RIEGL VZ-400i. If you look at the [datasheet](http://www.riegl.com/uploads/tx_pxpriegldownloads/RIEGL_VZ-400i_Datasheet_2022-09-27.pdf) on page 10, you will see the laser pulse repetition rate (PRR) and below an effective measurement rate, which is lower. For the VZ400i, the PRR is 2.4 times the efffective measurement rate. We'll come back to this in a second.

Our HELIOS++ `scanAngle` refers to the half angle. According to the datasheet, the total scan angle is 100°, so we would set the `scanAngleEffectiveMax_deg` to 50. From the ratio of effective measurement rate to pulse repetition rate, we can now compute the actual `scanAngleMax_deg` as 50° x 2.4 = 120° (half angle).

To understand this, we have to go a bit into the mechanisms of the deflector. The VZ-400i uses a prismatic deflector, probably a triangular prism. The actual output window (which is mentioned in the datasheet with -40° to +60°/100° total) is smaller than the maximum output window (2 x 120° = 240° total, as we calculated) to avoid that the beam hits another face of the rotating deflector, which would result in scattering in not-well defined directions.

Currently, users usually have to calculate this value for the `scanAngleMax_deg` themselves if they define a new scanner with "rotating" optics from the ratio of effective measurement rate to pulse repetition rate. But as you saw, this is pretty straightforward.

Btw, this is only important for rotating mirrors. For other deflectors ("optics"), this parameter will be ignored.

We have to admit that this is not well explained in the Wiki, yet. We will take care of this, thanks again 👍 

(I learned this all from @lwiniwar, so if there are further questions (or if I got something wrong), it's probably best if he jumps in)

zoeschindler, 2023-01-19:
Hello @han16nah, thank you very much for the explanation! This helps me very much!

han16nah, 2023-02-10:
An explanation was added to the Wiki (https://github.com/3dgeo-heidelberg/helios/wiki/Scanners)

closed 2023-02-10 by han16nah