3dgeo-heidelberg/helios #345: SimulationBuilder.setFinalOutput(True) cause error return.

Terrensou, 2023-03-05:
**The problem**
 I try to get simulation pointcloud data without output pointcloud file `simBuilder.setFinalOutput(True)` `simBuilder.setExportToFile(False)` . But the python code exit with return -1073741819 (0xC0000005) when it run to `join()` function.

**How to reproduce**
The code is simplified version of [I-getting-started.ipynb](https://github.com/3dgeo-heidelberg/helios/blob/dev/example_notebooks/I-getting-started.ipynb)

```
import pyhelios

simBuilder = pyhelios.SimulationBuilder(
    "data/surveys/toyblocks/als_toyblocks.xml", "assets/", "output/"
)
# simBuilder.setNumThreads(1)  # use only one thread (to ensure reproducibility)
simBuilder.setLasOutput(False)
simBuilder.setZipOutput(False)
simBuilder.setCallbackFrequency(10)  # Run with callback
simBuilder.setFinalOutput(True)  # Return output at join
simBuilder.setExportToFile(False)  # Disable export point cloud to file
simBuilder.setRebuildScene(True)

sim = simBuilder.build()

sim.start()
# Create instance of PyHeliosOutputWrapper class using sim.join().
# Contains attributes 'measurements' and 'trajectories' which are Python wrappers
# of classes that contain the output vectors.
output = pyhelios.outputToNumpy(sim.join())
```

**Expected behavior**
I want to get the pointclouds without create output files.

**Actual behavior**
The code interrupt with return -1073741819 (0xC0000005).

**Your environment (please complete the following information):**
 - OS: Windows 11 22621.1265
 - HELIOS++/pyhelios version 1.2.0

**Screenshots (Optional)**
[image]

han16nah, 2023-03-05:
Hi @Terrensou 
Thanks for reporting. I can confirm this issue.
`sim.join()` only seems to work when we provide `True` to `simBuilder.setExportToFile()`. 
@albertoesmp Can you have a look at this?

albertoesmp, 2023-03-06:
Hi @Terrensou and @han16nah 

There is a fix available in this commit: https://github.com/3dgeo-heidelberg/helios/commit/5bcdfdb185b67edb1ee18f33897bd2ffa314b34a

Let me know if you test it. I hope it solves the issue.

Thanks for reporting!

han16nah, 2023-03-08:
Hi @albertoesmp 
I recompiled HELIOS++ after your fix and tested it, but I am still getting `exit code -1073741819 (0xC0000005)` when running `sim.join()` after a simulation build with `simBuilder.setExportToFile(False)`. 😕

han16nah, 2023-03-08:
#348 solves the issue now. 👍 

Remaining task @han16nah : Write a Python test to avoid such a problem occurring again.

closed 2023-03-09 by han16nah