3dgeo-heidelberg/helios #347: How to scan at a uniform discrete angle in the vertical direction?

qixuema, 2023-03-07:
I am scanning a sphere from the center of the sphere with a TLS scanner (riegl_vz400), and the resulting point cloud is dense at the bottom and sparse at the top. Like this:

[image]

And I got the point cloud like this:

[image]

At the top, the point clouds are relatively sparse.

[image]

At the bottom, the point cloud is denser.

[image]

And surveys is:
```
<?xml version="1.0" encoding="UTF-8"?>
<document>
	<!-- Default scanner settings: -->
    <scannerSettings    id="profile1" 
                        active="true" 
                        pulseFreq_hz="100000" 
                        scanFreq_hz="120" 
                        scanAngle_deg="100" 
                        headRotatePerSec_deg="10.0"/>
                        
    <survey name="sphere_tls" 
            scene="data/scenes/sphere/sphere.xml#sphere" 
            platform="data/platforms.xml#tripod" 
            scanner="data/scanners_tls.xml#riegl_vz400">
        <FWFSettings binSize_ns="0.2" beamSampleQuality="3" />
        <leg>
            <platformSettings   x="0" 
                                y="0" 
                                onGround="true" />
            <scannerSettings    template="profile1" 
                                verticalAngleMin_deg="-20.0" 
                                verticalAngleMax_deg="100.0" 
                                headRotateStart_deg="0" 
                                headRotateStop_deg="3" 
                                trajectoryTimeInterval_s="1.0"/>
        </leg>
    </survey>
</document>
```

If I want the discretization of the upper and lower point clouds to be the same, what should I do?

lwiniwar, 2023-03-07:
Hi @qixuema!
Thank you for your question! I don't have your scene file or the .obj of the sphere, but from what it looks like to me, the scanner is just much closer to the bottom of the sphere than to the top -- which results in more points at the bottom than at the top. Maybe you can try the following:

```
<platformSettings   x="0" 
                    y="0" 
                    z="-1.7"
                    onGround="false" />

```

I set `z` to `-1.7` because the tripod adds 1.5 m and the Riegl VZ series add another 0.2 m -- so the scanner origin should then be exactly at (0,0,0). If that's the center of your sphere, you should be getting more what you are looking for.

Cheers,
Lukas

qixuema, 2023-03-08:
Wow, it worked!
Thank you so much. After I followed your tips to set it up, I got a perfect scan result -- symmetrical top and bottom.
I probably understand how it works.
Finally thank you very much again!
@lwiniwar 
[image]

closed 2023-03-08 by qixuema