3dgeo-heidelberg/helios #350: How to set different pitch angles for multiple TLS sensors in the same scene?

qixuema, 2023-03-13:
I want to set the pitch angle of multiple TLS sensors to different values in the same scene, such as some facing up and some facing down. 
I looked at the wiki and found that the direction of the sensor can be rotated by setting the rotation angle of the beamOrigin. However, this method would make all the sensors in the scene have the same pitch angle. 
So, I would like to ask, how can I set different pitch angles for different TLS sensors?

albertoesmp, 2023-03-13:
Hi @qixuema ,
I wonder if this might be a use case for a new feature included in the last release: [Multi-channel scanner](https://github.com/3dgeo-heidelberg/helios/wiki/Scanners#multi-channel-scanner). You could define a scanner with many channels that are all the same but with some variation. In your case, the beam origin rotation will be the only difference between channels. You can run helios with `--splitByChannel` if you want each TLS output to be written to a separate file. Otherwise, the points from different channels will be included in a single output point cloud. The example below shows how to define a TLS scanner with five different channels with an independent beam origin rotation each. 

```xml
<!-- ##### BEGIN : FIVE-CHANNELS RIEGL VZ-400 ##### -->	
	<scanner 	id 					= "riegl_vz400x5"
				name              		= "RIEGL VZ-400x5"
				accuracy_m			= "0.005"
			 	beamDivergence_rad 		= "0.0003"
				headRotatePerSecMax_deg	= "60"				
				optics            		= "rotating"
				pulseFreqs_Hz			= "100000,300000"
				pulseLength_ns     		= "5"
				rangeMin_m			= "1.5"
				scanAngleMax_deg   		= "120"
				scanAngleEffectiveMax_deg	= "50"
				scanFreqMin_Hz		= "3"
				scanFreqMax_Hz		= "120">
		
		<FWFSettings beamSampleQuality="3"/>	
		<headRotateAxis x="0" y="0" z="1"/>
		<channels>
			<channel id="0">
				<beamOrigin x="0" y="0" z="0.2">
					<rot axis="x" angle_deg="-30" />					
				</beamOrigin>
			</channel>
			<channel id="1">
				<beamOrigin x="0" y="0" z="0.2">
					<rot axis="x" angle_deg="-10" />					
				</beamOrigin>
			</channel>
			<channel id="2">
				<beamOrigin x="0" y="0" z="0.2">
					<rot axis="x" angle_deg="0" />					
				</beamOrigin>
			</channel>
			<channel id="3">
				<beamOrigin x="0" y="0" z="0.2">
					<rot axis="x" angle_deg="10" />					
				</beamOrigin>
			</channel>
			<channel id="4">
				<beamOrigin x="0" y="0" z="0.2">
					<rot axis="x" angle_deg="30" />					
				</beamOrigin>
			</channel>
		</channels>	
	</scanner>
	<!-- ##### END : FIVE-CHANNELS RIEGL VZ-400 ##### -->
```

Please, let us know if this works for you. If it doesn't, we can explore other alternatives. Maybe @lwiniwar already has some better idea to model this.

lwiniwar, 2023-03-13:
Hi @qixuema and @albertoesmp,

if I understand this correctly, you want to have multiple scanners in the same scene? @albertoesmp had a good solution, but if you don't mind running multiple surveys, you could also imagine having multiple scanners (one for vertical, one for horizontal settings) and then just running the simulation twice - one time with the vertical scan positions, one time with the horizontal ones.

Cheers,
Lukas

qixuema, 2023-03-14:
> Hi @qixuema , I wonder if this might be a use case for a new feature included in the last release: [Multi-channel scanner](https://github.com/3dgeo-heidelberg/helios/wiki/Scanners#multi-channel-scanner). You could define a scanner with many channels that are all the same but with some variation. In your case, the beam origin rotation will be the only difference between channels. You can run helios with `--splitByChannel` if you want each TLS output to be written to a separate file. Otherwise, the points from different channels will be included in a single output point cloud. The example below shows how to define a TLS scanner with five different channels with an independent beam origin rotation each.
> 
> ```
> 	
> 	<scanner 	id 					= "riegl_vz400x5"
> 				name              		= "RIEGL VZ-400x5"
> 				accuracy_m			= "0.005"
> 			 	beamDivergence_rad 		= "0.0003"
> 				headRotatePerSecMax_deg	= "60"				
> 				optics            		= "rotating"
> 				pulseFreqs_Hz			= "100000,300000"
> 				pulseLength_ns     		= "5"
> 				rangeMin_m			= "1.5"
> 				scanAngleMax_deg   		= "120"
> 				scanAngleEffectiveMax_deg	= "50"
> 				scanFreqMin_Hz		= "3"
> 				scanFreqMax_Hz		= "120">
> 		
> 		<FWFSettings beamSampleQuality="3"/>	
> 		<headRotateAxis x="0" y="0" z="1"/>
> 		<channels>
> 			<channel id="0">
> 				<beamOrigin x="0" y="0" z="0.2">
> 					<rot axis="x" angle_deg="-30" />					
> 				</beamOrigin>
> 			</channel>
> 			<channel id="1">
> 				<beamOrigin x="0" y="0" z="0.2">
> 					<rot axis="x" angle_deg="-10" />					
> 				</beamOrigin>
> 			</channel>
> 			<channel id="2">
> 				<beamOrigin x="0" y="0" z="0.2">
> 					<rot axis="x" angle_deg="0" />					
> 				</beamOrigin>
> 			</channel>
> 			<channel id="3">
> 				<beamOrigin x="0" y="0" z="0.2">
> 					<rot axis="x" angle_deg="10" />					
> 				</beamOrigin>
> 			</channel>
> 			<channel id="4">
> 				<beamOrigin x="0" y="0" z="0.2">
> 					<rot axis="x" angle_deg="30" />					
> 				</beamOrigin>
> 			</channel>
> 		</channels>	
> 	</scanner>
> 	
> ```
> 
> Please, let us know if this works for you. If it doesn't, we can explore other alternatives. Maybe @lwiniwar already has some better idea to model this.

Hi, @albertoesmp 

It looks very cool!

But I'm sorry, I would like to know how to set the channel for each scanner in the leg of the survey after setting the parameters for the scanner.

lwiniwar, 2023-03-14:
Hi @qixuema,

I believe @albertoesmp 's suggestion is to run all channels on all scan positions and then (by using `--splitByChannel`) selecting the point cloud output you want for that respective position. Say you have two channels and four scan positions (`leg`s) and you want to have channel 1 ("horizontal") for scan positions 1 and 4, and channel 2 ("vertical") for positions 2 and 3, you just delete the point cloud files for leg000channel001, leg001channel000, leg002channel000, and leg003channel001 (or similar).

@albertoesmp please correct me if that was not what you intended. We could also think about allowing `active=True` to be a list of IDs in the future?

Cheers,
Lukas

qixuema, 2023-03-14:
Hi, @albertoesmp @lwiniwar 

Thank you for your timely and patient response to my previous inquiry. After reviewing your suggestions, I think I have a better understanding of the helios and I now realize that I can achieve the desired results by using sensors with different pitch angle parameters.

While this approach may be a bit more cumbersome, I am willing to explore it further.

Once again, I greatly appreciate your help and expertise in this matter.

Cheers,
Xueqi

The following figure shows the results of scanning with multiple sensors.

[image]

[image]

qixuema, 2023-03-14:
There seems to be no more suitable method at the moment. I decided to use the multiple-scanner form.
So, let's turn off this issue for now.

closed 2023-03-14 by qixuema