3dgeo-heidelberg/helios #351: `Install on Linux with PyHelios support` -> Problem about installing GDAL

qixuema, 2023-03-15:
Hi,

I tried to install PyHelios on ubuntu 20.04. When I executed the following command, I encountered the problem.

GDAL
```
cd proj && ./configure --prefix=$PWD/../gdal/projlib && make && make install
```

[image]

```
Making all in src
make[1]: Entering directory '/studio/project/helios/lib/proj/src'
make  all-am
make[2]: Entering directory '/studio/project/helios/lib/proj/src'
/bin/bash ../libtool  --tag=CXX   --mode=link g++ -Wall -Wextra -Winit-self -Wunused-parameter -Wformat -Werror=format-security -Wno-format-nonliteral -Wlogical-op -Wshadow -Werror=vla -Wdate-time -Wnull-dereference -Wduplicated-cond -Wextra-semi -Wfloat-conversion -Wmissing-declarations -Wnon-virtual-dtor -Weffc++ -Woverloaded-virtual -Wdeprecated -Wzero-as-null-pointer-constant  -g -O2 -fvisibility=hidden   -o proj apps/proj.o apps/emess.o apps/utils.o libproj.la -lpthread -ldl -lm
libtool: link: g++ -Wall -Wextra -Winit-self -Wunused-parameter -Wformat -Werror=format-security -Wno-format-nonliteral -Wlogical-op -Wshadow -Werror=vla -Wdate-time -Wnull-dereference -Wduplicated-cond -Wextra-semi -Wfloat-conversion -Wmissing-declarations -Wnon-virtual-dtor -Weffc++ -Woverloaded-virtual -Wdeprecated -Wzero-as-null-pointer-constant -g -O2 -fvisibility=hidden -o .libs/proj apps/proj.o apps/emess.o apps/utils.o  ./.libs/libproj.so -L/opt/mambaforge-pypy3/envs/pyhelios/lib -lsqlite3 -lcurl /usr/lib/x86_64-linux-gnu/libsqlite3.so -ltiff /usr/lib/x86_64-linux-gnu/libcurl.so -lpthread -ldl -lm -pthread -Wl,-rpath -Wl,/studio/project/helios/lib/proj/../gdal/projlib/lib
/usr/bin/ld: ./.libs/libproj.so: undefined reference to `curl_easy_getinfo@CURL_OPENSSL_4'
/usr/bin/ld: ./.libs/libproj.so: undefined reference to `curl_easy_perform@CURL_OPENSSL_4'
/usr/bin/ld: ./.libs/libproj.so: undefined reference to `curl_easy_setopt@CURL_OPENSSL_4'
/usr/bin/ld: ./.libs/libproj.so: undefined reference to `curl_easy_init@CURL_OPENSSL_4'
/usr/bin/ld: ./.libs/libproj.so: undefined reference to `curl_easy_cleanup@CURL_OPENSSL_4'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1610: proj] Error 1
make[2]: Leaving directory '/studio/project/helios/lib/proj/src'
make[1]: *** [Makefile:1067: all] Error 2
make[1]: Leaving directory '/studio/project/helios/lib/proj/src'
make: *** [Makefile:486: all-recursive] Error 1
(pyhelios) root@xueqi-155:/studio/project/helios/lib/proj#
```

Installation steps.
- I first configured the python environment using conda and switched to a python environment with the name pyhelios;.
- Next, I followed the steps of Install on Linux with PyHelios support step by step.
- Then, when the execution reached `cd proj && . /configure --prefix=$PWD/... /gdal/projlib && make && make install`
 there is an error.

I would like to ask you how to solve this problem?

lwiniwar, 2023-03-15:
Could you try installing the `libcurl-dev` package?

```
sudo apt install libcurl-dev
```

Cheers!

qixuema, 2023-03-15:
Hi @lwiniwar 
I try to install `libcurl-dev` like this, and got three other packages.
```
(pyhelios) root@420c4e3252ac:~/studio/project/helios/helios-dev/lib/proj# apt install libcurl-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package libcurl-dev is a virtual package provided by:
  libcurl4-openssl-dev 7.68.0-1ubuntu2.16
  libcurl4-nss-dev 7.68.0-1ubuntu2.16
  libcurl4-gnutls-dev 7.68.0-1ubuntu2.16
You should explicitly select one to install.

E: Package 'libcurl-dev' has no installation candidate
(pyhelios) root@420c4e3252ac:~/studio/project/helios/helios-dev/lib/proj#
```

To avoid installation errors, I'd like to confirm whether I should install one of these three, is that correct?

qixuema, 2023-03-15:
It still doesn't seem to work.

```
(pyhelios) root@420c4e3252ac:~/studio/project/helios/helios-dev/lib/proj# apt install libcurl4-openssl-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libcurl4-openssl-dev is already the newest version (7.68.0-1ubuntu2.16).
The following packages were automatically installed and are no longer required:
  g++-7 libgdbm5 libisl19 libperl5.26 libreadline7 libstdc++-7-dev multiarch-support perl-modules-5.26
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
(pyhelios) root@420c4e3252ac:~/studio/project/helios/helios-dev/lib/proj#
```

```
(pyhelios) root@420c4e3252ac:~/studio/project/helios/helios-dev/lib/proj# ./configure --prefix=$PWD/../gdal/projlib && make && make install
```

```
Making all in include
make[1]: Entering directory '/studio/project/helios/helios-dev/lib/proj/include'
Making all in proj
make[2]: Entering directory '/studio/project/helios/helios-dev/lib/proj/include/proj'
Making all in internal
make[3]: Entering directory '/studio/project/helios/helios-dev/lib/proj/include/proj/internal'
Making all in nlohmann
make[4]: Entering directory '/studio/project/helios/helios-dev/lib/proj/include/proj/internal/nlohmann'
make[4]: Nothing to be done for 'all'.
make[4]: Leaving directory '/studio/project/helios/helios-dev/lib/proj/include/proj/internal/nlohmann'
make[4]: Entering directory '/studio/project/helios/helios-dev/lib/proj/include/proj/internal'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/studio/project/helios/helios-dev/lib/proj/include/proj/internal'
make[3]: Leaving directory '/studio/project/helios/helios-dev/lib/proj/include/proj/internal'
make[3]: Entering directory '/studio/project/helios/helios-dev/lib/proj/include/proj'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/studio/project/helios/helios-dev/lib/proj/include/proj'
make[2]: Leaving directory '/studio/project/helios/helios-dev/lib/proj/include/proj'
make[2]: Entering directory '/studio/project/helios/helios-dev/lib/proj/include'
make[2]: Nothing to be done for 'all-am'.
make[2]: Leaving directory '/studio/project/helios/helios-dev/lib/proj/include'
make[1]: Leaving directory '/studio/project/helios/helios-dev/lib/proj/include'
Making all in src
make[1]: Entering directory '/studio/project/helios/helios-dev/lib/proj/src'
make  all-am
make[2]: Entering directory '/studio/project/helios/helios-dev/lib/proj/src'
/bin/bash ../libtool  --tag=CXX   --mode=link g++ -Wall -Wextra -Winit-self -Wunused-parameter -Wformat -Werror=format-security -Wno-format-nonliteral -Wlogical-op -Wshadow -Werror=vla -Wdate-time -Wnull-dereference -Wduplicated-cond -Wextra-semi -Wfloat-conversion -Wmissing-declarations -Wnon-virtual-dtor -Weffc++ -Woverloaded-virtual -Wdeprecated -Wzero-as-null-pointer-constant  -g -O2 -fvisibility=hidden   -o proj apps/proj.o apps/emess.o apps/utils.o libproj.la -lpthread -ldl -lm
libtool: link: g++ -Wall -Wextra -Winit-self -Wunused-parameter -Wformat -Werror=format-security -Wno-format-nonliteral -Wlogical-op -Wshadow -Werror=vla -Wdate-time -Wnull-dereference -Wduplicated-cond -Wextra-semi -Wfloat-conversion -Wmissing-declarations -Wnon-virtual-dtor -Weffc++ -Woverloaded-virtual -Wdeprecated -Wzero-as-null-pointer-constant -g -O2 -fvisibility=hidden -o .libs/proj apps/proj.o apps/emess.o apps/utils.o  ./.libs/libproj.so -L/opt/mambaforge-pypy3/envs/pyhelios/lib -lsqlite3 -lcurl /usr/lib/x86_64-linux-gnu/libsqlite3.so -ltiff /usr/lib/x86_64-linux-gnu/libcurl.so -lpthread -ldl -lm -pthread -Wl,-rpath -Wl,/root/studio/project/helios/helios-dev/lib/proj/../gdal/projlib/lib
/usr/bin/ld: ./.libs/libproj.so: undefined reference to `curl_easy_getinfo@CURL_OPENSSL_4'
/usr/bin/ld: ./.libs/libproj.so: undefined reference to `curl_easy_perform@CURL_OPENSSL_4'
/usr/bin/ld: ./.libs/libproj.so: undefined reference to `curl_easy_setopt@CURL_OPENSSL_4'
/usr/bin/ld: ./.libs/libproj.so: undefined reference to `curl_easy_init@CURL_OPENSSL_4'
/usr/bin/ld: ./.libs/libproj.so: undefined reference to `curl_easy_cleanup@CURL_OPENSSL_4'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1610: proj] Error 1
make[2]: Leaving directory '/studio/project/helios/helios-dev/lib/proj/src'
make[1]: *** [Makefile:1067: all] Error 2
make[1]: Leaving directory '/studio/project/helios/helios-dev/lib/proj/src'
make: *** [Makefile:486: all-recursive] Error 1
(pyhelios) root@420c4e3252ac:~/studio/project/helios/helios-dev/lib/proj#
```

lwiniwar, 2023-03-15:
Looks to me like you need to set the path to the correct libcurl using the options in `./configure`: https://proj.org/install.html#cmake-configure-options

Especially `CURL_INCLUDE_DIR` and `CURL_LIBRARY` would need to be set, I believe?

qixuema, 2023-03-16:
Sorry, I checked the paths for `CURL_INCLUDE_DIR` and `CURL_LIBRARY` and they are as follows:

```
(pyhelios) root@420c4e3252ac:/studio/project/helios/helios-dev/lib/proj# dpkg-query -L libcurl4-openssl-dev
/.
/usr
/usr/bin
/usr/bin/curl-config
/usr/include
/usr/include/x86_64-linux-gnu
/usr/include/x86_64-linux-gnu/curl
/usr/include/x86_64-linux-gnu/curl/curl.h
/usr/include/x86_64-linux-gnu/curl/curlver.h
/usr/include/x86_64-linux-gnu/curl/easy.h
/usr/include/x86_64-linux-gnu/curl/mprintf.h
/usr/include/x86_64-linux-gnu/curl/multi.h
/usr/include/x86_64-linux-gnu/curl/stdcheaders.h
/usr/include/x86_64-linux-gnu/curl/system.h
/usr/include/x86_64-linux-gnu/curl/typecheck-gcc.h
/usr/include/x86_64-linux-gnu/curl/urlapi.h
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libcurl.a
/usr/lib/x86_64-linux-gnu/libcurl.la
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/libcurl.pc
/usr/share
/usr/share/aclocal
/usr/share/aclocal/libcurl.m4
/usr/share/doc
/usr/share/doc/libcurl4-openssl-dev
/usr/share/doc/libcurl4-openssl-dev/copyright
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/curl-config.1.gz
/usr/lib/x86_64-linux-gnu/libcurl.so
/usr/share/doc/libcurl4-openssl-dev/NEWS.Debian.gz
/usr/share/doc/libcurl4-openssl-dev/changelog.Debian.gz
(pyhelios) root@420c4e3252ac:/studio/project/helios/helios-dev/lib/proj#
```

```
(pyhelios) root@420c4e3252ac:/studio/project/helios/helios-dev/lib/proj# find /usr/lib -name libcurl\*
/usr/lib/x86_64-linux-gnu/libcurl-gnutls.so.4.6.0
/usr/lib/x86_64-linux-gnu/libcurl-gnutls.so.3
/usr/lib/x86_64-linux-gnu/libcurl.so
/usr/lib/x86_64-linux-gnu/libcurl-gnutls.so.4
/usr/lib/x86_64-linux-gnu/libcurl.la
/usr/lib/x86_64-linux-gnu/libcurl.so.4.6.0
/usr/lib/x86_64-linux-gnu/libcurl.so.4
/usr/lib/x86_64-linux-gnu/libcurl.a
/usr/lib/x86_64-linux-gnu/pkgconfig/libcurl.pc
(pyhelios) root@420c4e3252ac:/studio/project/helios/helios-dev/lib/proj#
```

Based on the above results, I guess that the paths corresponding to `CURL_INCLUDE_DIR` and `CURL_LIBRARY` may be `/usr/include/x86_64-linux-gnu/curl` and `/usr/lib/x86_64-linux-gnu` respectively.

Then, I would like to ask, if these two paths are fine, where should I set them? Should I open the configure file under the `/proj` path and add the above two paths to it? Or is there another way to set them?

qixuema, 2023-03-16:
Hi @lwiniwar 
I know what the problem is, I created the pyhelios_env environment and installed the packages before compiling the libraries, including curl. And when I compile the "proj" in lib in the "pyhelios virtual environment", the system finds curl in the conda environment, not under "/usr". so I got an error when compiling. 

After I created the pyhelios virtual environment and before installing the relevant python extensions, I compiled the "proj" and it worked.

If others have encountered this problem, you can use `which curl` to check if the curl used is in the "/usr" directory.

closed 2023-03-16 by qixuema

lwiniwar, 2023-03-16:
Wonderful, thank you for sharing your solution!