3dgeo-heidelberg/helios #354: How can I use pyhelios in Python 3.9

Si-Xiaoming, 2023-04-07:
We now have a project which needs Python 3.9. But when we run the progrem, an error occured:"ImportError: Module use of python38.dll conflicts with this version of Python." So how can I use pyhelios in Python 3.9?

han16nah, 2023-04-07:
Dear @Si-Xiaoming 
- I assume you use one of our [releases](https://github.com/3dgeo-heidelberg/helios/releases)? Which version are you using?
- Did you replace the default _pyhelios.pyd/_pyhelios.so file (which is for Python 3.8) in the run folder with the respective file corresponding to your version? (e.g., delete the _pyhelios.pyd and rename the _pyhelios.cpython-39.pyd to _pyhelios.pyd)
- Can you search in your helios "run" folder for the python38.dll, and if it exists, delete it (or move/rename it), and try if it works then?
Best, Hannah

Si-Xiaoming, 2023-04-18:
Thank you very much! I successfully solved the problem by replacing the ".pyd" file and deleting the "python38.dll".

closed 2023-04-18 by Si-Xiaoming