3dgeo-heidelberg/helios #362: Platform scanner mount rotation has no effect on scanner

maikherbermann, 2023-07-28:
I have defined a custom static platform with a rotation in the scanner mount tag.
The rotation has no effect on the scanner. If i define a translation it does affect the position of the scanner.
To successfully rotate the scanner i have defined a rotation in the scanner definition but this means if i want to change the rotation i have to change the scanner definition. I rather want to define it via the platform settings.

I do not know if this is a bug, expected behaviour or me doing something wrong. I expect it to work that way and have not found anything mentioned in the wiki which says otherwise.

Thanks in advance for any help on this.

The following code is a minimal working example:
```xml
<?xml version="1.0" encoding="UTF-8"?>
    <document>
        <platform id="basic_tripod" name="Tripod" type="static" drag="0">
            <scannerMount x="0" y="0" z="0" rotations="local">
                <rot axis="X" angle_deg="20" />
                <rot axis="Y" angle_deg="0" />
                <rot axis="Z" angle_deg="0" />
            </scannerMount>
        </platform>

        <scanner id="custom_lidar"
		 name="Custom Lidar"
		 accuracy_m="0.05"
		 beamDivergence_rad="0.00533"
		 headRotatePerSecMax_deg="10"
		 optics="rotating"
		 pulseFreqs_Hz="100000"
		 rangeMin_m="0.8"
               	 rangeMax_m="200"
		 scanAngleMax_deg="360"
		 scanAngleEffectiveMax_deg="360"
                 maxNOR="1"/>

        <scannerSettings id="profile1"
                         active="true"
                         pulseFreq_hz="100000"
                         verticalResolution_deg="1"
                         horizontalResolution_deg="1"
                         verticalAngleMin_deg="-22.50000"
                         verticalAngleMax_deg="22.50000"
                         headRotateStart_deg="-180.00000"
                         headRotateStop_deg="180.00000"
                         wavelength_nm="850"/>
        <platformSettings   id="platform1"
                            x="0.0"
                            y="0.0"
                            z="5.0"
                            onGround="false"/>
        <survey name="test_survey"
                scene="scene"
                platform="basic_tripod"
                scanner="custom_lidar">
            <leg>
                <platformSettings template="platform1" />
                <scannerSettings  template="profile1"/>
            </leg>
        </survey>

        <scene id="scene" name="Scene">
            <sunDir x="0" y="1" z="-1" />
            <skybox azimuth_deg="275" texturesFolder="assets/textures/sky/sky6_1024" />
                <part>
                <filter type="objloader">
                    <param type="string" key="filepath" value="Ground.obj"/>
                    <param type="string" key="up" value="z" />
                    <param type="boolean" key="castShadows" value="true" />
                    <param type="boolean" key="receiveShadows" value="true" />
                    <param type="boolean" key="recomputeVertexNormals" value="true" />
                </filter>
            </part>
    </scene>
</document>
    
```

**Your environment (please complete the following information):**
 - OS:  Windows 10
 - HELIOS++/pyhelios version 1.2.0

lwiniwar, 2023-07-28:
Hi @maikherbermann,

thanks for the report. This is indeed a bug, which we fixed only recently (#360).
Please find updated binaries attached to this answer and let us know if that resolves your issue.

[Release_dev20230707.zip](https://github.com/3dgeo-heidelberg/helios/files/12193300/Release_dev20230707.zip)

Thanks!
Lukas

maikherbermann, 2023-07-28:
Thanks, nice that it is fixed. Can you maybe provide the file for python3.10 since I am using that version. I can not test it with the .exe file since I have a virus scanner which does not allow the exe to run.

lwiniwar, 2023-07-28:
Hi @maikherbermann,

here's a .pyd for Python 3.10.
[_pyhelios_py310.zip](https://github.com/3dgeo-heidelberg/helios/files/12194305/_pyhelios_py310.zip)

Cheers,
Lukas

maikherbermann, 2023-07-28:
Thank you, it works for me. Great work!

closed 2023-07-28 by maikherbermann