3dgeo-heidelberg/helios #366: Error in Documentation ("Material files" in chapter "Scene")

Hannes9, 2023-08-17:
**The problem**
The material in .mtl format in combination with the respective .obj file wasn't linked with the described code snippet in the Scene chapter regarding material files.

**How to reproduce**
Explain in detail how to reproduce the bug. You can structure the individual steps as a numbered list, like below.

1. Do virtual scanning with the above mentioned code
2. Check e.g. in CloudCompare (CC) if the material values (Scalar values#4 for e.g. the helios_classification values are the correct ones which you put in the .mtl file

If necessary, provide simulation XML files, XML code snippets or pyhelios code snippets.
- scene XML: 

```
#  <param type="string" key="mtlfile" value="data/sceneparts/arbaro/tree.mtl" />
```

**Expected behavior**
Correct helios_classification values in scalar values for example in CloudCompare.

**Actual behavior**
Still default values for the helios_classification values were displayed in CC.

**Your environment (please complete the following information):**
 - OS: Windows 10

**Ideas for possible solutions**
I had to change the code to the code mentioned above. In scene.xml the code needed to change to key="mtlfile" instead of key="matfile" to get the correct Material values.

**Can you identify the location in the source code where the problem exists?**
scene.xml -> part-> scene-> filter

han16nah, 2023-08-21:
Dear @Hannes9 
we cannot reproduce your issue. Using `"matfile"` in the XML should give the correct result. Please note that you also have to include the `"matname"`, e.g.
```XML
<param type="string" key="matfile" value="data/sceneparts/arbaro/tree.mtl" />
<param type="string" key="matname" value="leaves" />
```
Can you give us the survey files that you are using (especially the `scene.xml`)? Maybe then we can give more hints :) 

Best,
Hannah

Hannes9, 2023-08-21:
Dear Hannah,

Thanks for your response. I tried it myself with some examples again and you are right. It worked with the "matfile".

Using the "matname" would be only needed if I want to reference to only one specific material out of the .mtl file, right ? Otherwise all the materials which were linked from the .obj to the .mtl file are taken or?

Thanks in advance.

Best,
Hannes

Hannes9, 2023-08-22:
Hello Hannah.

Thanks for the explanation. I'm probably mostly loading OBJ files. So this line of code (see below) isn't even neccessary of loading .obj files for referencing to the material file, right ?:

`<param type="string" key="matfile" value="data/sceneparts/arbaro/tree.mtl" />`

Best,
Hannes

han16nah, 2023-08-22:
Hi Hannes,
exactly. for OBJ files, the **material library/libraries** is/are usually specified in the .obj file (e.g., `mtllib tree.mtl` somewhere at the beginning of the file) and the specific **material** (by name) before all faces that should use this material (e.g. `usemtl leaves`).

We are simply using the standard format definitions here, with the addition of custom HELIOS++-specific values in the material library (.mtl) file, such as `helios_classification` or `helios_spectra`.

Read more about the file formats here:
- https://en.wikipedia.org/wiki/Wavefront_.obj_file
- https://paulbourke.net/dataformats/obj/
- https://paulbourke.net/dataformats/mtl/

Hannes9, 2023-08-22:
Hi Hannah,
perfect, thanks a lot for the help.

It answered all my questions. I'll close this issue now.

closed 2023-08-22 by Hannes9

Xiaofei-Kevin-Yang, 2023-09-22:
> Hi Hannah, perfect, thanks a lot for the help.
> 
> It answered all my questions. I'll close this issue now.

Hi Hannes9,

Hope you are doing well. I encountered the same problem. Would you mind listing the solution steps to add labels to the existing scene?

Thanks,
Kevin

Xiaofei-Kevin-Yang, 2023-09-22:
Hi there,

Thanks for your excellent work. I was wondering how to add labels to the generated point clouds by HELIOS++. Could you please list the solution steps and give an example?

Thanks,
Xiaofei

Hannes9, 2023-09-22:
Hi @Xiaofei-Kevin-Yang,

Yes, thanks and I hope you as well.
What do you exactly mean by those labels ? the "helios_classification" ? I think it's depending which file format you are using for importing the scene. As I imported an .obj file, I just had to make sure that the material linking with "mtllib nameofmtlfile.mtl" is set there in the 2nd row, so it's automatically linking to it. Then, apparently now further .mtl import is needed for an .obj file in the scene.xml.

Xiaofei-Kevin-Yang, 2023-09-22:
> Hi @Xiaofei-Kevin-Yang,
>
> Yes, thanks and I hope you as well. What do you exactly mean by those labels ? the "helios_classification" ? I think it's depending which file format you are using for [...]

Hi Hannes,

Thanks for your quick response. My purpose is to generate well-annotated synthetic point cloud data from BIM. Firstly, I will convert ifc format to obj format. Then, I plan to use your algorithm to generate well-annotated synthetic point cloud data with semantic labels of each building element. Could you please provide me with an example by using your algorithm to generate well annoatetd synthetic data? 

Thanks,
Xiaofei

Hannes9, 2023-09-22:
Hi @Xiaofei-Kevin-Yang ,

You're welcome ! :)
Ah okey, sounds interesting. With my algorithm you mean Helios++ or how can I understand that ? :)

Best,
Hannes

Xiaofei-Kevin-Yang, 2023-09-22:
Hi Hannes,

I would like to generate a synthetic point cloud data that the classification labels are after the XYZ coordinates. I want to know how to automatically add classification labels.

Thanks,
Xiaofei

Xiaofei-Kevin-Yang, 2023-09-22:
Hi Hannes,

Could you please send me your email so I can share some data with you to help you better understand my purpose?

Thanks,
Xiaofei

Hannes9, 2023-09-22:
> Hi Hannes,
> 
> I would like to generate a synthetic point cloud data that the classification labels are after the XYZ coordinates. I want to know how to automatically add classification labels.
> 
> Thanks, Xiaofei

You mean between XYZ and intensity? I just used the "helios_classification" one to add it, so it's visible as the "class" in the .xyz file.

Xiaofei-Kevin-Yang, 2023-09-22:
> You mean between XYZ and intensity? I just used the "helios_classification" one to add it, so it's visible as the "class" in the .xyz file.

Yes, how can I achieve that? Could you please provide an example?

Thanks

Hannes9, 2023-09-22:
In general even for adding a new label (not sure if this is possbile reagrding the HELIOS++ .xyz output formats or assiging values for "class" or other ones a python script can be helpfull for automatically add labels in the .obj/.mtl files :)

Hannes9, 2023-09-22:
> Yes, how can I achieve that? Could you please provide an example?
>
> Thanks

As I said creating a python script for achieving that could be helpful :). I'm sorry, but for now I am not allowed probably to share my created code for this.

Xiaofei-Kevin-Yang, 2023-09-22:
> In general even for adding a new label (not sure if this is possbile reagrding the HELIOS++ .xyz output formats or assiging values for "class" or other ones a python script can be helpfull for automatically add labels in the .obj/.mtl files :)

I tried to revise the .mlt file to add helios_classification, but the generated point cloud data did not contain the classification labels. I am confused how to do that.

Xiaofei-Kevin-Yang, 2023-09-22:
> As I said creating a python script for achieving that could be helpful :). I'm sorry, but for now I am not allowed probably to share my created code for this.

I am new to the virtual laser scanning. Thanks for your patience to answer my questions. Much appreciated.

Xiaofei-Kevin-Yang, 2023-09-22:
> As I said creating a python script for achieving that could be helpful :). I'm sorry, but for now I am not allowed probably to share my created code for this.

Thanks for your reply.

Hannes9, 2023-09-22:
> I tried to revise the .mlt file to add helios_classification, but the generated point cloud data did not contain the classification labels. I am confused how to do that.

Hm okey, got it. I think I tried that with renaming the files and therefor the input files in the respective scene.xml file as well. And maybe adding this codeline which is importing the .mtl file could help as well even if it should not be necessary for the .obj file. Make sure in your obj file as well that you are referencing to the .mtl file with "mtllib".

Hannes9, 2023-09-22:
You're welcome ! :)