3dgeo-heidelberg/helios #367: Using 2 scanners in simulation

Hannes9, 2023-08-22:
**Describe the solution you'd like**
Is there a possibility in HELIOS to use 2 scanners (e.g. one horizontal with different height, and one vertical one) in one simulation ? 

**Describe alternatives you've considered**
To do 2 simulations with seperated scanners included and merging e.g. the resulted point cloud afterwards would be an alternative or ?

**Additional context**
I want to simulate this total scanning device. (see picture below)
[image]

Thanks a lot !

han16nah, 2023-08-22:
Dear @Hannes9 

yes, it is possible to define a MultiScanner as documented [here in the Wiki](https://github.com/3dgeo-heidelberg/helios/wiki/Scanners#multi-channel-scanner). So for your application, you would add a new scanner to `scanners_tls.xml`, which has two "channels". For each channel, you can define a different `beamOrigin` with the position relative to the platform) and a rotation (`rot`). 

This could look somewhat like this (assuming you are using something like the `simple_linearpath` platform):

```xml
<scanner id="navvis_vlx_3" ...> <!-- put all the scanner settings here like accuracy, wavelength, pulse frequency, max. scan angle, etc. -->
	<channels>
		<channel id="0">
			<beamOrigin x="0" y="0" z="1.9"> <!-- the one on top -->
                  		<rot axis="y" angle_deg="90"/> <!-- to scan horizontally --> 
              		</beamOrigin>
		</channel>
		<channel id="1">
			<beamOrigin x="0" y="0.3" z="1.3"> <!-- the one at the front -->
                  		<rot axis="z" angle_deg="90"/> <!-- to scan (vertically) towards the side -->
              		</beamOrigin>
		</channel>
	</channels>
</scanner>
```

(Not tested though 😉)

When executing the surveys, you can then add the option `--splitByChannel` to get one file per device or leave it out to get the merged point clouds from both devices (per leg, as always).

I hope this helps. Good luck, and just ask if you have further questions.

Best,

Hannah

Hannes9, 2023-08-22:
Dear Hannah,

Thanks for the quick response and that already helped. From my understanding one scanner can have multiple channels (see pictures from the manual of the scanner). So for my correct application I guess I need to add in total 64 channels (for both scanners) and 32 of each with different z and y values for in front and on top of a person, right ?

I am now wondering where I have to put the Vertical Angle setting for each of those channels (see table). Would those be then the `angle_deg= 19.5` (example for the 1st on top) ? I am bit confused with the coordinates & rotation angles at the moment to be honest.

Yes I want to get the merged one directly, thanks for the info.

Thanks in adcance.

[image]
[image]
[image]

han16nah, 2023-08-22:
Not better, just a bit easier to debug I guess

Hannes9, 2023-08-22:
Alright I understand. 

Perfect, thanks a lot ! :)

Hannes9, 2023-08-25:
Is there a possibility to set up the differentiation of Laser Emitter and Laser Receiver in Helios ? (see Figure 1.5 from previous message and Figure 1.6 below)

[image]

lwiniwar, 2023-08-25:
Hi @Hannes9 ,

no, this feature is not available in HELIOS++.

Cheers,
Lukas

Hannes9, 2023-08-25:
Hi Lukas,

Alright, Thanks for the fast respone ! :)

Best,
Hannes

Hannes9, 2023-08-28:
Hi @han16nah or @lwiniwar,

I tried to implement now the new Multi-Channel Multi-scanner from above (see Figure 1.5). Somehow I still get this error message shown below. I attached the whole Code in Scanners_als.xml from my new created scanner below as well. 
[image]

I was also wondeirng why those lines (see below) are marked differently (line 400-402 in code). Do maybe have an idea about that ?

[image]

	<scanner 	id 							= "hesai_XT32M2X"
				accuracy_m					= "0.01"		
				beamDivergence_rad 			= "0.0005"
				name              			= "Hesai XT32M2X 32-Channel Medium-Range Mechanical Lidar"
				optics            			= "rotating"
				pulseFreqs_Hz				= "50000,100000,200000,300000,380000,550000"
				pulseLength_ns     			= "4"
				averagePower_w				= "10"
				rangeMin_m					= "0.5"
				rangeMax_m					= "300"
				scanAngleMax_deg   			= "360"
				scanAngleEffectiveMax_deg	= "345"
				scanFreqMin_Hz				= "10"
				scanFreqMax_Hz				= "200"
				wavelength_nm				= "905"> 				

`<FWFSettings beamSampleQuality="1"/>	
		<channels>
		
			<channel id="0">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="19.5" />
				</beamOrigin>
			</channel>
            <channel id="1">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="18.2" />
				</beamOrigin>
			</channel>
            <channel id="2">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="16.9" />
				</beamOrigin>
			</channel>
            <channel id="3">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="15.6" />
				</beamOrigin>
			</channel>
            <channel id="4">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="14.3" />
				</beamOrigin>
			</channel>
            <channel id="5">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="13.0" />
				</beamOrigin>
			</channel>
            <channel id="6">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="11.7" />
				</beamOrigin>
			</channel>
            <channel id="7">
				<beamOrigin x="0" y="0.0305" z="1.9">
					<rot axis="x" angle_deg="10.4" />
				</beamOrigin>
[... 326 lines cut ...]
            <channel id="61">
				<beamOrigin x="0" y="0" z="1.335">
					<rot axis="z" angle_deg="90" />
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="18.2 />
				</beamOrigin>
			</channel>
            <channel id="62">
				<beamOrigin x="0" y="0" z="1.335">
					<rot axis="z" angle_deg="90" />
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="19.5" />
				</beamOrigin>
			</channel>
            <channel id="63">
				<beamOrigin x="0" y="0" z="1.335">
					<rot axis="z" angle_deg="90" />
					<rot axis="y" angle_deg="90" />
					<rot axis="x" angle_deg="20.8" />
				</beamOrigin>
			</channel>
			
        </channels>
	</scanner>
	 `

lwiniwar, 2023-08-28:
Hi @Hannes9 ,

you are missing a closing quotation mark (") here:
[image]

Hannes9, 2023-08-28:
Hi Lukas,

True, I'm sorry that I somehow didn't see that one.
It is working now, thanks a lot !

Hannes9, 2023-08-28:
Another topic: I have now a problem with executing the simulation along all waypoints.
In the following you can see that the simulation somehow stops after the 1st leg even more legs are provided. I didn't change anything in the survey.xml file except setting the onGround parameter and deleting the heights z values as they are no longer needed then.

[image]
Animation is just going from the first waypoint to the second waypoint as well and then it stops. 

Code of survey.xml:
`<?xml version="1.0"?>
<document>
	<platformSettings id="platform1" movePerSec_m="50" onGround="true" />
    <scannerSettings id="scanner1" active="true" verticalResolution_deg="1.3" horizontalResolution_deg="0.36" headRotateStart_deg="0" headRotateStop_deg="360" scanAngle_deg="20.15" pulseFreq_hz="100000" trajectoryTimeInterval_s="0.01"/>
	<survey name="tum_data_mls" platform="data/platforms.xml#navvis-vlx-3-on-person" scanner="data/scanners_als.xml#hesai_XT32M2X" scene="data/scenes/owndata/tum_scene/tum_scene.xml#tum_scene_DL">
	
		<leg>
			<platformSettings  template="platform1" x="2697480" y="5349130"/>
			<scannerSettings  template="scanner1"/>
		</leg>
		<leg>
			<platformSettings  template="platform1" x="2697500" y="5349250"/>
			<scannerSettings  template="scanner1"/>
		</leg>
		<leg>
			<platformSettings  template="platform1" x="2697450" y="5349250"/>
			<scannerSettings  template="scanner1"/>
		</leg>
		<leg>
			<platformSettings  template="platform1" x="2697430" y="5349190"/>
			<scannerSettings  template="scanner1"/>
		</leg>
		<leg>
			<platformSettings  template="platform1" x="2697550" y="5349170"/>
			<scannerSettings  template="scanner1"/>
		</leg>
	</survey>
</document>
`

lwiniwar, 2023-08-28:
Hi @Hannes9, please provide your platforms.xml as well.
Thanks!

Hannes9, 2023-08-28:
Hi.

Sure, the below. I have already the thought that it might happened, because for the other legs which are outside of the building, there is no z-value (height) defined then or ? -> Because of the onGround parameter.

`	<platform id="navvis-vlx-3-on-person" name="NavVis VLX 3 on person" type="linearpath">
		<scannerMount x="0" y="0" z="0">
			<rot axis="x" angle_deg="0" />	
		</scannerMount>
		
	</platform>`

han16nah, 2023-08-28:
Hi Hannes,
yes, that sounds like a likely explanation. If your platform is not at the same XY-location as your scenepart with the `helios_isGround` parameter set to true in the material library file (.mtl), then this will cause problems. So maybe expand whatever you use as "ground" or add another large ground plane or so.

Cheers,

Hannah

Hannes9, 2023-08-28:
Hi Hannah.

I am not using the `helios_isGround` paramter, but the `onGround` parameter in the platformSettings in the survey.xml. Still I will try adding another ground plane and changing the waypoints within the scenepart. Probably one of those should work.

Thansk ! :)

han16nah, 2023-08-28:
Ah sorry, I confused it with ground positioning of the scene parts, sorry

You're right. For the platform, HELIOS++ determines the ground as the lowest z-coordinate in the scene data at the xy-position of the leg. So if there is no object at that location, no platform z-value can be computed. Hope it works with the respective scene/waypoint modifications 🤞 :)

Hannes9, 2023-08-28:
Hm, it is still taking only one leg. I added a groundplane and modified the waypoints, so that they are withing the scenparts.

han16nah, 2023-08-28:
Did it work before replacing z values by the `onGround` parameter?

Hannes9, 2023-08-28:
I tried it with `onGround="false" z="470"/>` again and it is also not working.

It looks like it is not stopping at the 2nd waypoint and goes somehow to infinity maybe after the 1st waypoint (see picture below)
[image]

han16nah, 2023-08-28:
Is it correct that you want the scanner to rotate? If so, I guess you want it to rotate continuously, not just on 360° rotation?
Can you try setting the value for `headRotateStop_deg` to something much higher, to allow for many rotations of the scanner? 

You may have a look at [example notebook 12](https://nbviewer.org/github/3dgeo-heidelberg/helios/blob/dev/example_notebooks/12-multi_scanner_puck.ipynb), which uses the Puck for both static and dynamic surveys.

If you provide a `headRotateStop_deg`, the leg will be terminated once that angle is reached (cf. TLS surveys). I guess this is before reaching the next waypoint so that's why your simulation stops.

Using the rotation speed (`headRotatePerSec_deg`) and your rotation range (difference between `headRotateStop_deg` and `headRotateStart_deg`), you can calculate how many seconds the scan will take.

Hannes9, 2023-08-28:
Yes, continously. With `headRotateStart_deg` and `headRotateStop_deg` I wanted to set the FOV (horizontal) to continously 360 degrees, not only 90 degrees. So it is contiously scanning everything around 360 degrees within every leg.

I changed `headRotateStop_deg` to 36000, but the simulations looks unchanged.
I added the `headRotatePerSec_deg` with 10 as well.

I'll have a look on this notebook.

Hannes9, 2023-08-28:
This is my survey.xml file now, but it the simulation is still going beyond the 2nd waypoint.
`<?xml version="1.0"?>
<document>
	<platformSettings id="platform1" movePerSec_m="50" onGround="true"/>
    <scannerSettings id="scanner1" active="true" verticalResolution_deg="1.3" horizontalResolution_deg="0.36" headRotatePerSecMax_deg="7200" headRotatePerSec_deg="20.00" headRotateStart_deg="0" headRotateStop_deg="3600000000" scanAngle_deg="20.15" pulseFreq_hz="20000" trajectoryTimeInterval_s="0.01"/>
	<survey name="tum_data_mls" platform="data/platforms.xml#navvis-vlx-3-on-person" scanner="data/scanners_als.xml#hesai_XT32M2X" scene="data/scenes/owndata/tum_scene/tum_scene.xml#tum_scene_DL">
	
		<leg>
			<platformSettings  template="platform1" x="2697482" y="5349159"/>
			<scannerSettings  template="scanner1"/>
		</leg>
		<leg>
			<platformSettings  template="platform1" x="2697483" y="5349197"/>
			<scannerSettings  template="scanner1"/>
		</leg>
		<leg>
			<platformSettings  template="platform1" x="2697458" y="5349189"/>
			<scannerSettings  template="scanner1"/>
		</leg>
		<leg>
			<platformSettings  template="platform1" x="2697479" y="5349174"/>
			<scannerSettings  template="scanner1"/>
		</leg>
		<leg>
			<platformSettings  template="platform1" x="2697518" y="5349170"/>
			<scannerSettings  template="scanner1"/>
		</leg>
	</survey>
</document>
`

Hannes9, 2023-09-03:
I solved it with just removing the `headRotateStop_deg`. Would that mean that the FOV (horizontal) is still 360° anyways ?

closed 2024-11-28 by han16nah