3dgeo-heidelberg/helios #370: Hovermap integration

celine-mercier, 2023-08-27:
Hello,

We would love the hovermap to be added in the list of scanners. Or any idea to emulate it better than by using the best scanners with the highest settings and complicated trajectories :)

Best
Celine

han16nah, 2023-08-30:
Hi Celine,

thank you for your very interesting request. 
To bring in the additional scanner rotation of the hovermap requires either a workaround (e.g., some python script to create an appropriate `interpolated` trajectory) or a modification in the HELIOS++ core. We will have to discuss this in our next core developer meeting and will come back to you afterwards.

Best,

Hannah

celine-mercier, 2023-09-04:
Hi Hannah, 

Thanks a lot for your answer, we'll be waiting then :) Cheers

Celine