3dgeo-heidelberg/helios #372: Question regarding an external software for trajectories creation with .ifc import

Hannes9, 2023-09-12:
Hello,

Do you maybe know any proper software for creating waypoints (trajectory) by clicking on an imported .ifc file (e.g. on a surface of a slab) and the trajectory can be exported (e.g. as a CSV file) afterwards which can be used then in HELIOS++ ? 

I would need this for indoor virtual scanning purposes which avoids setting many waypoints manually in the survey XML.

Thanks a lot.

lwiniwar, 2023-09-22:
Hi @Hannes9 , 
at least the dev team does not know of such software.
Do note, however, that you can use the "interpolated" trajectory type which accepts such a CSV instead of having to place individual legs in the XML file. There is an example available here:

https://github.com/3dgeo-heidelberg/helios/blob/dev/example_notebooks/13-interpolated_trajectory.ipynb

Cheers!
Lukas

Hannes9, 2023-09-22:
Hi @lwiniwar ,

Ah okey. So basically I would have then two legs: The first one includes the start point and the second the last point of my whole path. Every points inbetween are listed in the trajectory file. And I would have only "x", "y", "z" coordiantes in the CVS/ trajectory file for the waypoints which should also work, right ? I just have to set the `xIndex`,` yIndex` and `zIndex` correctly and the time settings I can ignore.

The `moveperSec_m `and the `onGround` parameter I have to set in those legs then or ? As there is no `platformSettings` template in the Survey.xml anymore. Anyways, the` onGround` parameter won't be necessary anymore as the baseplatform will then take the trajectory file with all it points and its coordinates, right ?

lwiniwar, 2023-09-22:
Hi @Hannes9 ,
there is some documentation on the interpolated trajectories in the wiki: https://github.com/3dgeo-heidelberg/helios/wiki/Survey#interpolated-trajectories

tldr: I think you need to give a time column, and from that the speed will be inferred. Also, I would provide columns for roll, pitch, and yaw, and make sure that the yaw always points to the direction of the next point. I'm not sure what happens if you don't provide the columns but I think it will fail while parsing.

The idea of the `interpolated` platform was to provide full flexibility in the platform position and attitude at the cost of having to provide these values manually. From experience, a short python-script can easily generate these files for you.

Yes, onGround should not be required anymore (and, in fact, be ignored, if I am not mistaken).

Cheers,
Lukas

closed 2023-11-03 by lwiniwar